MichaIng/DietPi #2807: Username Changer

diveyez, 2019-05-16:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
No

#### Describe the solution you'd like:
The Ability to change default user name (dietpi) on install 

#### Describe alternatives you've considered:
Writing script myself, but met with challenges which are system breaking (groupmod maybe?)

MichaIng, 2019-05-16:
@diveyez 
Many thanks for your request.

First of all, do never rename or remove the `dietpi` user. It is used by several software titles and its group as a shared permissions group to allow e.g. media players access files created by the downloaders and such stuff.

So it's about a simple user creation implementation. Actually we have such a function in our scripts already: https://github.com/MichaIng/DietPi/blob/master/dietpi/func/dietpi-set_software#L475
But needs some review before this can be made public, e.g. accessible via `dietpi-config`.

diveyez, 2019-05-20:
Raspbian does the same thing, I hate that. I do not like having usernames out of my normal as logins. I automate and script everything and its like a wrench. Needless to say, _your Diet-pi did everything I asked it to without an OMM or Undervolt_, until gitlab-ce reconfigure hit it, *hahahaha*

__ __

> CLOSING

MichaIng, 2019-05-20:
@diveyez 
Actually having multiple users is a main driver and security concept of all operating systems. Just run `cat /etc/passwd` and you will see many lines, each of them defining an own user, nearly all of them strictly required and used by OS core functions. So `dietpi` is just the one that we use allow our software installs sharing access permissions.

The only thing that could be though through, is degrading `dietpi` from a login user to a system user. So it will not show up on e.g. deskop login prompts and you cannot use it to login into any shell. Then it should be allowed to create a new own user account on first boot of DietPi for sure.

> until gitlab-ce reconfigure hit it

What you mean by this?
_________
I reopen the issue since I like the idea to have custom user creation added to dietpi-config.

diveyez, 2019-05-20:
I installed gitlab-ce and configured it, and the os went into an overnight bathe of OMM's

MichaIng, 2019-05-20:
@diveyez 
Okay but `gitlab-ce` should not be related to the `dietpi` user at all. Did I understand correctly that you removed the `dietpi` user manually? Not sure how existing file permissions are handled in such case, I bet those are handed over to root.

Ah nope, the user ID stays in place:
```
root@VM-Stretch:~# userdel dietpi
root@VM-Stretch:~# getent passwd dietpi
root@VM-Stretch:~# l /mnt/dietpi_userdata/
total 16
drwxrwxr-x 2 1000 1000 4096 Mar 24 20:12 downloads
drwxrwxr-x 2 1000 1000 4096 Mar 24 20:12 Music
drwxrwxr-x 2 1000 1000 4096 Mar 24 20:12 Pictures
drwxrwxr-x 2 1000 1000 4096 Mar 24 20:12 Video
```

That is actually great. It means that you can simply re-create the user to fix possible occurring issues. Then, if desired as system user: `useradd -ru 1000 -g 1000 dietpi`

The alternative is to `chown` the affected files/dirs with the new user, e.g.: `chown -R diveyez:diveyez /mnt/dietpi_userdata /var/{lib,tmp}/dietpi`
That should be it. `DietPi-Software` installs will throw some errors when attempting to add users to dietpi group, but aside from that AFAIK nothing else depends on the `dietpi` user+group.

diveyez, 2019-05-21:
Software dependencies surfaced AFTER the chown 1 liner.
What I want to know is why didnt systemd say anything BEFORE we did that. Lol!

diveyez, 2019-05-21:
Timed out waiting for device dev-tty.device.
Dependency failed for Serial Getty on tty.

I had to jump (ctrl+alt+f1-2-3) and land on tty4 just to see this. This is so weird. I just updated as usual with apt and didnt do anything funky.

MichaIng, 2019-05-21:
@diveyez 
Ah, totally forgot that of course the systemd units and in case software settings need to be adjusted to use your user instead of `dietpi`. So really, I suggest you revert and re-create the dietpi user, you will otherwise just run into follow up errors that might be hard to debug when one does not know the background.

> Dependency failed for Serial Getty on tty.

Do you use any serial console? Otherwise do:
```
for i in /etc/systemd/system/getty.target.wants/serial-getty@*.service
do
systemctl disable --now ${i##*/}
done
```

However indeed the tty errors cannot be related to the `dietpi` user existence. Regardless I will reopen the issue, as I want to implement the initial suggestion: The ability to create a new login user account via dietpi-config 😉.

diveyez, 2019-05-22:
OpenMediaVault really dont like dietpi eh?

MichaIng, 2019-05-22:
@diveyez 
@FredericGuilbault also aims to do this. There is still some steps to do to bring our install and image creation scripts on a stage that derivatives are easy to create. Especially true is that we no not create our images from scratch but instead use base-images ourself (Debian- and Raspbian-based). We simply do not have the man power (and knowledge/experience) do deal with bootloader and kernel development.

Generally the image preparation script is a good start: https://github.com/MichaIng/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh
- It turns Debian/Raspbian-based images into DietPi, taking some limitations into account (at least Jessie, systemd required, in some cases certain firmware repos expected).

Then I am working on an automated image creation script: https://github.com/MichaIng/DietPi/pull/2693
- Basically to automate file system + partition shrinking + archive creation.

And finally the idea to run the above PREP_SYSTEM_FOR_DIETPI.sh from within a qemu-chroot environment: https://github.com/MichaIng/DietPi/issues/1775

All three above implemented would allow to create a ready-to-run DietPi image in one step/run from any virtualization-capable machine. This would then be a good basis to create derivatives from.

FredericGuilbault, 2019-05-22:
Yep I explored this way but it's too complicated for me ATM so I skipped. Anyway I realised that A big part of what make DietPi DietPi append at the first run not in `PREP_SYSTEM_FOR_DIETPI.sh`. 

 DietPi is not shipped in a functioning state. Most of the magic append during the first boot. PREP_SYSTEM_FOR_DIETPI is more to strip and standardise all the different OS into something ready To have Dietpi installed on it. (if my understanding is right).

---
Talking of Raspbian , I was working with `pi-gen` since xmass before comming to dietpi so I know the build process of raspbian pretty well.User `pi` is not hardcoded. Most of the referance to the user during the build process is made to the numerical UID `1000` and there is a variable somewhere where you can set the username you want. This said, there few hidden places in `pi-gen` Where they refer to `pi`directly...  But Still I think it's the best approach, UID 1000 will always be the default user regardless of it's name.

diveyez, 2019-05-23:
Finally a truly headless setup for dietpi is coming!!!

FredericGuilbault, 2019-05-23:
Post your branch. @diveyez  Im interrested in your work.

diveyez, 2019-05-23:
Well I guess now @MichaIng has some more inspiration to make a builder project for people to base their distros with dietpi! 

As dietpi is, I could not make a distribution level image.

diveyez, 2019-05-23:
@MichaIng do you have a development branch I can watch for changes to test?

MichaIng, 2019-05-23:
@diveyez 
Sure, the [dev](https://github.com/MichaIng/DietPi/tree/dev) branch it is.

diveyez, 2019-05-24:
@Michalng

I need to be your friend. Because I deploy with bash *.sh in everything I do.

This is why dietpi is the base image for the design of the Meth0d OS image.

I really need that friendship when you can spare it ;)

MichaIng, 2019-05-24:
@diveyez 
Okay for me, but I don't know if GitHub has a friendship system? Of course you can follow me and/or star/watch DietPi.

diveyez, 2019-05-25:
`friend`

diveyez, 2019-05-25:
Sorry.

diveyez, 2019-05-26:
gmail: nemofbaltimore@gmail.com
Shoot me an email, using issue comments is counter productive.

MichaIng, 2019-05-26:
@diveyez 
Still not sure if I understand what you need from me 🙈. I like to stay with DietPi-related communication on GitHub and dietpi.com forum where possible.

Also if it's about derivatives, this might be interesting for others as well, so having questions/ideas/requests documented in a GitHub discussion is beneficial?

diveyez, 2019-05-26:
Fair Enough.
Likely postponed due to personal issues and that information isn't going on here.

diveyez, 2019-05-27:
@MichaIng Brother Michalng, when can I see a tiny image to base from? It would be nice to also have a script to use to add things to that menu, as well as own functions and scripts, and such. 
I need to modify some things about dietpi which I currently cannot without doing some weird things.

The username changer thing would be nice to have before hand as well.

Be straight and honest, one love. -Richard

diveyez, 2019-06-15:
@MichaIng is there anyway to change the default dietpi login to something more exotic/personal? I really dont want to use that as a default login due to cloud hosting of my picluster's and botnet's trying to brute it.

diveyez, 2019-06-15:
I ask this of you because a new user CANNOT do things like, modify network connections from desktop, or mount an attached filesystem...

@MichaIng  I am guessing something like this *MIGHT* work?

``

#!/bin/bash
read -p "What is your desired username?" $newuname
pkill -U dietpi
usermod -l $newuname dietpi
usermod -m -d /home/$newuname dietpi
usermod -a -G dietpi,adm,dialout,cdrom,sudo,audio,video,plugdev,games,users,input,netdev,gpio,i2c,spi $newuname 
chown -R $newuname:$newuname /mnt/dietpi_userdata /var/{lib,tmp}/dietpi
echo "Rebooting in 3 seconds to complete the changes";
sleep 3
reboot

``

then

``

deluser --remove-home dietpi

``

MichaIng, 2019-06-15:
@diveyez 
The problem is that we use the dietpi user and group across many software installs as run user and to share access permissions. So the overhead of making all this variable would be very high.

As mentioned somewhere above, what I could imagine is turning dietpi into a system user, so it would not show up on login masks, if this is what annoys you currently. And then yeah we could add either/and:
- A dietpi.txt entry to create a new user on first boot
- A prompt on first login
- An option to dietpi-config

The function to create a login user is already there: https://github.com/MichaIng/DietPi/blob/dev/dietpi/func/dietpi-set_software#L569

diveyez, 2019-06-15:
It is the security concern of having a static assigned user on my system other than root that concerns me.
I will look into this after game night. =)

MichaIng, 2019-06-15:
@diveyez 
> security concern

`usermod -s $(command -v nologin) dietpi` disallows login at least. Additionally you can remove its sudo permissions: `rm /etc/sudoers.d/dietpi && gpasswd -d dietpi sudo`

ankur4u007, 2022-11-29:
@MichaIng any plans to implement a custom user creation step in dietpi.txt and make it automated ?