MichaIng/DietPi #2808: Lighttpd broken after upgrade to v6.24 (Hotfix)

maartenlangeveld, 2019-05-16:
Hi Micha/Daniel,

Have now upgraded my (restored) RPi Zero W (armv6) from v6.22 to v6.24. All went well (thanks!) except that Lighttpd seems broken. Service won't start, and it fails to reïnstall (dependency unmet libssl1.1).

Is there a solution?

Thanks

```
[FAILED] DietPi-Services | lighttpd     failed (Result: exit-code) since Thu 2019-05-16 15:44:39 CEST; 1min 51s ago
● lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2019-05-16 15:44:39 CEST; 1min 51s ago
  Process: 2029 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf (code=exited, status=255)

May 16 15:44:39 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
May 16 15:44:39 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 16 15:44:39 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 16 15:44:39 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
May 16 15:44:39 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
May 16 15:44:39 DietPi systemd[1]: Stopped Lighttpd Daemon.
May 16 15:44:39 DietPi systemd[1]: lighttpd.service: Start request repeated too quickly.
May 16 15:44:39 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 16 15:44:39 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 16 15:44:39 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
```

MichaIng, 2019-05-16:
@maartenlangeveld
Many thanks for reporting.

I recognised this with the sury.org repo (used on non-ARMv6) hence blocked the SSL packages there. Good to know this is true with Buster repo as well, so its about the libssl1.1 version (1.1.1), not about the specific package.

However since I tested with Lighttpd, this update seems to be not done automatically, right? Do you remember when this got installed?

So we need to block this for Buster branch on Stretch as well. I hope there are no follow-up dependency by this.

Will post some instructions to downgrade and block the SSL package this evening.

maartenlangeveld, 2019-05-16:
@MichaIng 

If I remember well Lighttpd it got installed when I installed O!MPD. How can I get lighttpd working again, I really need it. Have you a workaround or solution ?

Thanks!

**Another issue:**

phpmyadmin  (4:4.6.6-4) has been removed with the upgrade to v6.24

When reinstalling ( dietpi-software reinstall 90 ) it seems it installs php7.0 back again...

Would have loved to post terminal output but my terminal sessions suddenly died...

maartenlangeveld, 2019-05-16:
> Will post some instructions to downgrade and block the SSL package this evening.

Thanks, Micha !

PeterLacknase, 2019-05-16:
The old conf is preserved (on my zero w, that is). Copy /etc/lighttpd/lighttpd.conf to /etc/lighttpd/lighttpd.conf.old. After that

cp lighttpd.conf-dpkg* lighttpd.conf

and start the service. (If there is no need for special config.)

maartenlangeveld, 2019-05-16:
@PeterLacknase 
Thanks but not thanks ;-) ...your advice did not help, lighttpd sill does not start.

maartenlangeveld, 2019-05-16:
@MichaIng

Hi Micha, is there a workaround ? Or do I have to wait for v6.25 ? 
In that case I will go back to v6.22 again again... and wait for v6.25.

Thanks, Maarten

MichaIng, 2019-05-16:
@maartenlangeveld @PeterLacknase 
### Lighttpd
Reverting the old config file will not work, since the php-fpm socket has been changed to the `php7.3-fpm.sock`.

Okay now I am confused. Both Lighttpd and libssl1.1 has been upgraded to the Buster version, so they should be compatible with each other. Not sure how this can happen with priority 100 but it must be due to some dependency tail. This is why I was not happy to use the Buster repo.

Found the dependency tail for the Buster libssl1.1 version:
`php7.3-curl` >` libcurl4 (>= 7.44.0)` > `libssl1.1 (>= 1.1.1)`

Attempting to forcefully downgrade Lighttpd:
```
root@VM-Stretch:~# apt install lighttpd
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following package was automatically installed and is no longer required:
  libssl1.0.2
Use 'apt autoremove' to remove it.
The following additional packages will be installed:
  libssl1.1
Suggested packages:
  rrdtool php5-cgi apache2-utils lighttpd-doc
Recommended packages:
  spawn-fcgi
The following packages will be REMOVED:
  curl libcurl4 php7.3-curl
The following packages will be DOWNGRADED:
  libssl1.1 lighttpd
0 upgraded, 0 newly installed, 2 downgraded, 3 to remove and 36 not upgraded.
Need to get 1,638 kB of archives.
After this operation, 2,037 kB disk space will be freed.
Do you want to continue? [Y/n] n
```
- It would downgrade libssl1.1 as well, thus does not work with PHP7.3.
- Strange is that the old version should actually work with the new libssl1.1 as well:
```
Depends: init-system-helpers (>= 1.18~), libattr1 (>= 1:2.4.46-8), libbz2-1.0, libc6 (>= 2.17), libfam0, libpcre3, libssl1.1 (>= 1.1.0), zlib1g (>= 1:1.1.4), mime-support, lsb-base (>= 3.0-6)
```
- `>= 1.1.0` should include `1.1.1b`, however it does obviously not.

Okay given that Lighttpd needs to be taken from Buster, now the question is why it fails. On my test VM system it works, but might be different on Raspbian.

Please paste the following:
```
systemctl restart lighttpd
cat /var/log/lighttpd/error.log
```

MichaIng, 2019-05-16:
### phpMyAdmin
With phpMyAdmin this is indeed an issue, although easy to solve:
- phpMyAdmin depends on the PHP meta packages:
```
Depends: php, php-cli, php-mysql, php-json, php-mbstring, php-xml ...
```
- Due to priority pins those are installed from Stretch repo, thus pull in PHP7.0.
- To solve, do:
```
apt install -t buster php php-cli php-mysql php-json php-mbstring php-xml php-phpseclib php-php-gettext
apt purge *php7.0* # Check that phpMyAdmin would not be purge by this
```
- If the purge would still purge phpMyAdmin, then dependencies are different then on my VM, then please paste `apt show -a phpmyadmin`

PeterLacknase, 2019-05-16:
> @maartenlangeveld @PeterLacknase
> 
> ### Lighttpd
> Reverting the old config file will not work, since the php-fpm socket has been changed to the `php7.3-fpm.sock`.
> 
> ```

Sorry if I couldn‘t express it clear enough:

On my Zero W rpis during the lighttpd update the old file was _preserved_. Thatswhy the startup stalled. Using the updated one, that came with the new package, everything worked.

maartenlangeveld, 2019-05-16:
@MichaIng 

```
root@DietPi:/# systemctl restart lighttpd
Job for lighttpd.service failed because the control process exited with error code.
See "systemctl status lighttpd.service" and "journalctl -xe" for `details.`
```
error.log remains empty
```
root@DietPi:/# cat /var/log/lighttpd/error.log
root@DietPi:/#
```

MichaIng, 2019-05-16:
PR up to solve phpMyAdmin and hopefully Lighttpd issue as well: https://github.com/MichaIng/DietPi/pull/2809

MichaIng, 2019-05-16:
@maartenlangeveld 
Hmm, please paste:
```
lighttpd -tf /etc/lighttpd/lighttpd.conf
journalctl -u lighttpd
```

MichaIng, 2019-05-16:
@PeterLacknase 
> On my Zero W rpis during the lighttpd update the old file was preserved. Thatswhy the startup stalled. Using the updated one, that came with the new package, everything worked.

Ah, that might be possible. Okay makes sense:
- I tested with fresh install: Lighttpd was installed from Stretch first, then directly upgraded to Buster when PHP install forced it with the libssl1.1 update. So in my case the config was auto-updated by APT, which is not done when manual edit did happen.

Will check the configs for differences.

maartenlangeveld, 2019-05-16:
@MichaIng 

```
root@DietPi:~# lighttpd -tf /etc/lighttpd/lighttpd.conf
/bin/bash: /usr/share/lighttpd/create-mime.assign.pl: No such file or directory
2019-05-16 23:20:50: (configfile.c.1461) command "/usr/share/lighttpd/create-mime.assign.pl" exited non-zero: 127
2019-05-16 23:20:50: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 27 pos: 14 parser failed somehow near here: (EOL)
```

```
root@DietPi:~# journalctl -u lighttpd
-- Logs begin at Thu 2019-05-16 22:54:45 CEST, end at Thu 2019-05-16 23:21:47 CEST. --
May 16 22:57:25 DietPi systemd[1]: Starting Lighttpd Daemon...
May 16 22:57:26 DietPi lighttpd[26738]: /bin/sh: 1: /usr/share/lighttpd/create-mime.assign.pl: not found
May 16 22:57:26 DietPi lighttpd[26738]: 2019-05-16 22:57:25: (configfile.c.1461) command "/usr/share/lighttpd/create-mime.assign.pl" exited non-zero: 127
May 16 22:57:26 DietPi lighttpd[26738]: 2019-05-16 22:57:25: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 27 pos: 14 parser failed somehow near
 here: (EOL)
May 16 22:57:26 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
May 16 22:57:26 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 16 22:57:26 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 16 22:57:26 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
May 16 22:57:26 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
May 16 22:57:26 DietPi systemd[1]: Stopped Lighttpd Daemon.
May 16 22:57:26 DietPi systemd[1]: Starting Lighttpd Daemon...
May 16 22:57:27 DietPi lighttpd[26775]: /bin/sh: 1: /usr/share/lighttpd/create-mime.assign.pl: not found
May 16 22:57:27 DietPi lighttpd[26775]: 2019-05-16 22:57:26: (configfile.c.1461) command "/usr/share/lighttpd/create-mime.assign.pl" exited non-zero: 127
May 16 22:57:27 DietPi lighttpd[26775]: 2019-05-16 22:57:26: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 27 pos: 14 parser failed somehow near
 here: (EOL)
May 16 22:57:27 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
May 16 22:57:27 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 16 22:57:27 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 16 22:57:27 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
May 16 22:57:27 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
May 16 22:57:27 DietPi systemd[1]: Stopped Lighttpd Daemon.
May 16 22:57:27 DietPi systemd[1]: Starting Lighttpd Daemon...
May 16 22:57:28 DietPi lighttpd[26809]: /bin/sh: 1: /usr/share/lighttpd/create-mime.assign.pl: not found
May 16 22:57:28 DietPi lighttpd[26809]: 2019-05-16 22:57:27: (configfile.c.1461) command "/usr/share/lighttpd/create-mime.assign.pl" exited non-zero: 127
May 16 22:57:28 DietPi lighttpd[26809]: 2019-05-16 22:57:27: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 27 pos: 14 parser failed somehow near
 here: (EOL)
May 16 22:57:28 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
May 16 22:57:28 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 16 22:57:28 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 16 22:57:28 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
May 16 22:57:28 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
May 16 22:57:28 DietPi systemd[1]: Stopped Lighttpd Daemon.
May 16 22:57:28 DietPi systemd[1]: Starting Lighttpd Daemon...
May 16 22:57:29 DietPi lighttpd[26845]: /bin/sh: 1: /usr/share/lighttpd/create-mime.assign.pl: not found
May 16 22:57:29 DietPi lighttpd[26845]: 2019-05-16 22:57:28: (configfile.c.1461) command "/usr/share/lighttpd/create-mime.assign.pl" exited non-zero: 127
May 16 22:57:29 DietPi lighttpd[26845]: 2019-05-16 22:57:28: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 27 pos: 14 parser failed somehow near
 here: (EOL)
May 16 22:57:29 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
May 16 22:57:29 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 16 22:57:29 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 16 22:57:29 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
May 16 22:57:29 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
May 16 22:57:29 DietPi systemd[1]: Stopped Lighttpd Daemon.
May 16 22:57:29 DietPi systemd[1]: Starting Lighttpd Daemon...
May 16 22:57:30 DietPi lighttpd[26878]: /bin/sh: 1: /usr/share/lighttpd/create-mime.assign.pl: not found
May 16 22:57:30 DietPi lighttpd[26878]: 2019-05-16 22:57:29: (configfile.c.1461) command "/usr/share/lighttpd/create-mime.assign.pl" exited non-zero: 127
May 16 22:57:30 DietPi lighttpd[26878]: 2019-05-16 22:57:29: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 27 pos: 14 parser failed somehow near
 here: (EOL)
May 16 22:57:30 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
May 16 22:57:30 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 16 22:57:30 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 16 22:57:30 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
May 16 22:57:30 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
May 16 22:57:30 DietPi systemd[1]: Stopped Lighttpd Daemon.
[... cut ...]
May 16 23:16:53 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
May 16 23:16:53 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
May 16 23:16:53 DietPi systemd[1]: Stopped Lighttpd Daemon.
May 16 23:16:53 DietPi systemd[1]: lighttpd.service: Start request repeated too quickly.
May 16 23:16:53 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 16 23:16:53 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 16 23:16:53 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
```

MichaIng, 2019-05-16:
### Lighttpd default configs
#### Raspbian Stretch
```
server.modules = (
	"mod_access",
	"mod_alias",
	"mod_compress",
 	"mod_redirect",
)

server.document-root        = "/var/www/html"
server.upload-dirs          = ( "/var/cache/lighttpd/uploads" )
server.errorlog             = "/var/log/lighttpd/error.log"
server.pid-file             = "/var/run/lighttpd.pid"
server.username             = "www-data"
server.groupname            = "www-data"
server.port                 = 80

index-file.names            = ( "index.php", "index.html", "index.lighttpd.html" )
url.access-deny             = ( "~", ".inc" )
static-file.exclude-extensions = ( ".php", ".pl", ".fcgi" )

compress.cache-dir          = "/var/cache/lighttpd/compress/"
compress.filetype           = ( "application/javascript", "text/css", "text/html", "text/plain" )

# default listening port for IPv6 falls back to the IPv4 port
include_shell "/usr/share/lighttpd/use-ipv6.pl " + server.port
include_shell "/usr/share/lighttpd/create-mime.assign.pl"
include_shell "/usr/share/lighttpd/include-conf-enabled.pl"
```
#### Raspbian Buster
```
server.modules = (
	"mod_indexfile",
	"mod_access",
	"mod_alias",
 	"mod_redirect",
)

server.document-root        = "/var/www/html"
server.upload-dirs          = ( "/var/cache/lighttpd/uploads" )
server.errorlog             = "/var/log/lighttpd/error.log"
server.pid-file             = "/var/run/lighttpd.pid"
server.username             = "www-data"
server.groupname            = "www-data"
server.port                 = 80

# strict parsing and normalization of URL for consistency and security
# https://redmine.lighttpd.net/projects/lighttpd/wiki/Server_http-parseoptsDetails
# (might need to explicitly set "url-path-2f-decode" = "disable"
#  if a specific application is encoding URLs inside url-path)
server.http-parseopts = (
  "header-strict"           => "enable",# default
  "host-strict"             => "enable",# default
  "host-normalize"          => "enable",# default
  "url-normalize-unreserved"=> "enable",# recommended highly
  "url-normalize-required"  => "enable",# recommended
  "url-ctrls-reject"        => "enable",# recommended
  "url-path-2f-decode"      => "enable",# recommended highly (unless breaks app)
 #"url-path-2f-reject"      => "enable",
  "url-path-dotseg-remove"  => "enable",# recommended highly (unless breaks app)
 #"url-path-dotseg-reject"  => "enable",
 #"url-query-20-plus"       => "enable",# consistency in query string
)

index-file.names            = ( "index.php", "index.html" )
url.access-deny             = ( "~", ".inc" )
static-file.exclude-extensions = ( ".php", ".pl", ".fcgi" )

compress.cache-dir          = "/var/cache/lighttpd/compress/"
compress.filetype           = ( "application/javascript", "text/css", "text/html", "text/plain" )

# default listening port for IPv6 falls back to the IPv4 port
include_shell "/usr/share/lighttpd/use-ipv6.pl " + server.port
include_shell "/usr/share/lighttpd/create-mime.conf.pl"
include "/etc/lighttpd/conf-enabled/*.conf"

#server.compat-module-load   = "disable"
server.modules += (
	"mod_compress",
	"mod_dirlisting",
	"mod_staticfile",
)
```

MichaIng, 2019-05-16:
@maartenlangeveld 
> include_shell "/usr/share/lighttpd/create-mime.assign.pl"

Is there some `lighttpd.conf.ucf-new` or` *dpkg-new/ucf` file in your `/etc/lighttpd` or does the `lighttpd.conf.bak_1605XX_0` look like the Buster version above?

When you found the one, replace the existing lighttpd.conf with it and do:
```
sed -i '/^server.document-root/c\server.document-root        = "/var/www"' /etc/lighttpd/lighttpd.conf
```
- This will change the webroot to `/var/www`, @PeterLacknase you should do this as well since this is how DietPi installs expect it.

maartenlangeveld, 2019-05-16:
@MichaIng 

I get confused... both config files (stretch and buster) have
> include_shell "/usr/share/lighttpd/create-mime.assign.pl"

So what makes the difference ? which config should I take then?

Thanks Maarten

MichaIng, 2019-05-16:
@maartenlangeveld 
?? Nope definitely not. I have both here. Perhaps you found an old backup file that is from Stretch as well?
Does it include all the additional directives as above, e.g. `server.http-parseopts = (`?

Actually since everything else matches or is optional or replaceable, the only required switch is:
```
sed -i 's/create-mime\.assign\.pl/create-mime.conf.pl/g' /etc/lighttpd/lighttpd.conf
```

You can also verify that the one file exists, while the other not. Seems it was renamed:
`ls -l /usr/share/lighttpd/create-mime.conf.pl /usr/share/lighttpd/create-mime.assign.pl`

maartenlangeveld, 2019-05-16:
@MichaIng 

My apologies, you are completely right. They are different .assign. <--> .conf. I did not notice.
Will try again. Thanks !!

maartenlangeveld, 2019-05-16:
@MichaIng 

Have followed your instruction. System extremely slow. Htop shows that system continuously tries to start lighttpd, spawning new processes. Rebooting now. Takes a very very long time that I can start SSH session after reboot.

MichaIng, 2019-05-16:
@maartenlangeveld 
So the service still fails at some point.
```
lighttpd -tf /etc/lighttpd/lighttpd.conf
journalctl -u lighttpd
cat /var/log/lighttpd/error.log
```

maartenlangeveld, 2019-05-16:
@MichaIng,

I stopped lighttpd, still >95% cpu. Don't know whta's going wrong. Seems things getting even worse. Cannot restart lighttpd (failed). 

```
root@DietPi:/# lighttpd -tf /etc/lighttpd/lighttpd.conf
2019-05-17 00:22:01: (configfile.c.1321) opening configfile  /etc/lighttpd/conf-enabled/50-phpmyadmin.conf failed: No such file or directory
2019-05-17 00:22:01: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 47 pos: 15 parser failed somehow near here: (EOL)
```
```
root@DietPi:/# journalctl -u lighttpd
-- Logs begin at Thu 2019-05-16 23:54:38 CEST, end at Fri 2019-05-17 00:24:41 CEST. --
May 17 00:08:28 DietPi systemd[1]: Starting Lighttpd Daemon...
May 17 00:08:32 DietPi lighttpd[1794]: 2019-05-17 00:08:28: (configfile.c.1321) opening configfile  /etc/lighttpd/conf-enabled/50-phpmyadmin.conf failed: No suc
h file or directory
May 17 00:08:32 DietPi lighttpd[1794]: 2019-05-17 00:08:28: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 47 pos: 15 parser failed somehow near
here: (EOL)
May 17 00:08:32 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
May 17 00:08:32 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 17 00:08:32 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 17 00:08:32 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
May 17 00:08:32 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
May 17 00:08:32 DietPi systemd[1]: Stopped Lighttpd Daemon.
May 17 00:08:32 DietPi systemd[1]: Starting Lighttpd Daemon...
May 17 00:08:41 DietPi lighttpd[1932]: 2019-05-17 00:08:32: (configfile.c.1321) opening configfile  /etc/lighttpd/conf-enabled/50-phpmyadmin.conf failed: No suc
h file or directory
May 17 00:08:41 DietPi lighttpd[1932]: 2019-05-17 00:08:32: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 47 pos: 15 parser failed somehow near
here: (EOL)
May 17 00:08:41 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
May 17 00:08:41 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 17 00:08:41 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 17 00:08:41 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
May 17 00:08:42 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
May 17 00:08:42 DietPi systemd[1]: Stopped Lighttpd Daemon.
May 17 00:08:42 DietPi systemd[1]: Starting Lighttpd Daemon...
May 17 00:08:53 DietPi lighttpd[2147]: 2019-05-17 00:08:42: (configfile.c.1321) opening configfile  /etc/lighttpd/conf-enabled/50-phpmyadmin.conf failed: No suc
h file or directory
May 17 00:08:54 DietPi lighttpd[2147]: 2019-05-17 00:08:42: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 47 pos: 15 parser failed somehow near
here: (EOL)
May 17 00:08:54 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
May 17 00:08:54 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 17 00:08:54 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 17 00:08:54 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
May 17 00:08:54 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
May 17 00:08:54 DietPi systemd[1]: Stopped Lighttpd Daemon.
May 17 00:08:54 DietPi systemd[1]: Starting Lighttpd Daemon...
May 17 00:09:05 DietPi lighttpd[2412]: 2019-05-17 00:08:54: (configfile.c.1321) opening configfile  /etc/lighttpd/conf-enabled/50-phpmyadmin.conf failed: No suc
h file or directory
May 17 00:09:05 DietPi lighttpd[2412]: 2019-05-17 00:08:54: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 47 pos: 15 parser failed somehow near
here: (EOL)
May 17 00:09:05 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
May 17 00:09:05 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 17 00:09:05 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 17 00:09:05 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
May 17 00:09:05 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
May 17 00:09:05 DietPi systemd[1]: Stopped Lighttpd Daemon.
May 17 00:09:05 DietPi systemd[1]: Starting Lighttpd Daemon...
May 17 00:09:14 DietPi lighttpd[2672]: 2019-05-17 00:09:05: (configfile.c.1321) opening configfile  /etc/lighttpd/conf-enabled/50-phpmyadmin.conf failed: No suc
h file or directory
May 17 00:09:14 DietPi lighttpd[2672]: 2019-05-17 00:09:05: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 47 pos: 15 parser failed somehow near
here: (EOL)
May 17 00:09:14 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
May 17 00:09:14 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 17 00:09:14 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 17 00:09:14 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
May 17 00:09:14 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
May 17 00:09:14 DietPi systemd[1]: Stopped Lighttpd Daemon.
May 17 00:09:14 DietPi systemd[1]: Starting Lighttpd Daemon...
May 17 00:09:22 DietPi lighttpd[2922]: 2019-05-17 00:09:15: (configfile.c.1321) opening configfile  /etc/lighttpd/conf-enabled/50-phpmyadmin.conf failed: No suc
h file or directory
[... 118 lines cut ...]
May 17 00:10:30 DietPi systemd[1]: Starting Lighttpd Daemon...
[... cut ...]
May 17 00:10:48 DietPi lighttpd[4791]: 2019-05-17 00:10:39: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 47 pos: 15 parser failed somehow near
here: (EOL)
May 17 00:10:48 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
May 17 00:10:48 DietPi systemd[1]: Failed to start Lighttpd Daemon.
May 17 00:10:48 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
May 17 00:10:48 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
```
```
root@DietPi:/# cat /var/log/lighttpd/error.log
root@DietPi:/#
```

MichaIng, 2019-05-16:
@maartenlangeveld 
Stop all services: `dietpi-services stop`
Did you already apply the phpMyAdmin fix above? https://github.com/MichaIng/DietPi/issues/2808#issuecomment-493220292

Possibly there were still the two php-fpm versions running concurrently. So the above, which includes purging php7.0 packages, perhaps solves the hanging.

Then since you just switched back to default Lighttpd Buster config, phpMyAdmin most likely wants to reapply its configuration changes. You should be able to do that via:
`dpkg-reconfigure phpmyadmin`
- It should prompt a question which webserver you want to apply phpMyAdmin configs to. Select Lighttpd and continue.

maartenlangeveld, 2019-05-16:
@MichaIng 

I did apply the phpMyAdmin fixes. Had also purged the php7.0 packages.

Now services stopped, stil very high CPU >95%, it seems /lib/systemd/systemd-journald consumes lot of cpu power.

maartenlangeveld, 2019-05-17:
@MichaIng 

Hi Micha, have installed everything again

- v6.24 update removes symlink to '50-phpmyadmin.conf' (conf-available directory) so symlink in conf-enabled is pointing to non-existing symlink in conf-available;
- therefore lighttpd initially would not start despite having modified lighttpd.conf beforehand;
- /etc/apt/preferences.d/dietpi-php gets overwritten during v6.24 installation, so I had to re-apply 'phpMyAdmin' patched /etc/apt/preferences.d/dietpi-php after v6.24 installation

but more high CPU (95-100%) load remains with same error for dietpi-wifi-monitor written:

```
root@DietPi:~# journalctl -xe
-- Logs begin at Fri 2019-05-17 02:17:24 CEST, end at Fri 2019-05-17 02:49:56 CEST. --
May 17 02:48:30 DietPi dietpi-wifi-monitor.sh[728]: Try 'sleep --help' for more information.
May 17 02:48:30 DietPi dietpi-wifi-monitor.sh[728]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 17 02:48:30 DietPi dietpi-wifi-monitor.sh[728]: Try 'sleep --help' for more information.
May 17 02:48:30 DietPi dietpi-wifi-monitor.sh[728]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 17 02:48:30 DietPi dietpi-wifi-monitor.sh[728]: Try 'sleep --help' for more information.
May 17 02:48:31 DietPi dietpi-wifi-monitor.sh[728]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 17 02:48:31 DietPi dietpi-wifi-monitor.sh[728]: Try 'sleep --help' for more information.
May 17 02:48:31 DietPi dietpi-wifi-monitor.sh[728]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 17 02:48:31 DietPi dietpi-wifi-monitor.sh[728]: Try 'sleep --help' for more information.
May 17 02:48:31 DietPi dietpi-wifi-monitor.sh[728]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 17 02:48:31 DietPi dietpi-wifi-monitor.sh[728]: Try 'sleep --help' for more information.
May 17 02:48:31 DietPi dietpi-wifi-monitor.sh[728]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 17 02:48:31 DietPi dietpi-wifi-monitor.sh[728]: Try 'sleep --help' for more information.
May 17 02:48:31 DietPi dietpi-wifi-monitor.sh[728]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 17 02:48:31 DietPi dietpi-wifi-monitor.sh[728]: Try 'sleep --help' for more information.
May 17 02:48:31 DietPi dietpi-wifi-monitor.sh[728]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 17 02:48:31 DietPi dietpi-wifi-monitor.sh[728]: Try 'sleep --help' for more information.
May 17 02:48:32 DietPi dietpi-wifi-monitor.sh[728]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 17 02:48:32 DietPi dietpi-wifi-monitor.sh[728]: Try 'sleep --help' for more information.
May 17 02:48:32 DietPi dietpi-wifi-monitor.sh[728]: sleep: invalid time interval ‘{TICKRATE:-10}’
```
This is a real issue. Will go back to v6.22 again again again... I'm sorry since I know how committed you are. But with 100% CPU I have no other option.

MichaIng, 2019-05-17:
@maartenlangeveld 
Ah there is some typo:
```
sed -i 's/TICKRATE=/TICKRATE=\$/' /var/lib/dietpi/services/dietpi-wifi-monitor.sh
systemctl restart dietpi-wifi-monitor
```

maartenlangeveld, 2019-05-17:
@MichaIng
Thanks. Micha!

MichaIng, 2019-05-17:
@maartenlangeveld 
Great to hear. Many thanks for your patience with the many issues that PHP7.3 from Buster came with. Your reports allowed to quickly analyse, test and fix in the right direction.

I solved btw the priorities a bid different to assure that only the absolutely required packages are pulled from Buster: https://github.com/MichaIng/DietPi/pull/2809

This allows install e.g. build-essentials from Stretch, since some core libraries were not pulled from Buster. But in your case (where the mentioned libraries were already upgraded), this requires downgrading of some of them. When required, I will assist in how to do that.

maartenlangeveld, 2019-05-17:
**@MichaIng**

You are more than welcome. I have to thank you and Daniel for putting so much effort so that I enjoy my SBCs. I do not need build essentials, I installed it by accident, so no assistance relating to thus issue is needed.

PeterLacknase, 2019-05-18:
> This will change the webroot to /var/www, @PeterLacknase you should do this as well since this is how DietPi installs expect it.

The right path came with the updated conf file (on my machines, that is). 

Thanks for all Your patience!

MichaIng, 2019-05-18:
Okay, v6.24.1 just released with Lighttpd and further PHP on ARMv6 fixes.

closed 2019-05-18 by MichaIng