MichaIng/DietPi #2810: /bin/bash: dietpi-update: command not found

Sopor, 2019-05-16:
### Creating a bug report/issue

#### Required Information
- G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH=master
G_GITOWNER=MichaIng

- Distro version | 9.9
- Kernel version | Linux dietpi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
- SBC device | RPi 3 Model B+ (armv7l)
- Power supply used | 5V 2.4A
- SDcard used | None
- USB | Kingston A400 SSD with external power

#### Steps to reproduce

I noticed a message today about updating DietPi `dietpi-update   : Run now to update DietPi from v6.22.3 to v6.24.0.` so i ran first `dietpi-update` but then i got a message that i should use `sudo -s dietpi-update` so i did that too.

#### Expected behaviour

I assume it should start to update DietPi?

#### Actual behaviour

Instead i got this message
```
xxxxx@dietpi:~ $ dietpi-update
[FAILED] DietPi-Update | Root privileges required. Please run the command with "G_SUDO", or, "sudo -s".
xxxxx@dietpi:~ $ sudo -s dietpi-update
[sudo] password for xxxxx:
/bin/bash: dietpi-update: command not found
xxxxx@dietpi:~ $
```

UNiXMIT, 2019-05-17:
Try running the script with the full path:

`sudo /DietPi/dietpi/dietpi-update`

Sopor, 2019-05-17:
If full path is needed the help text should includes that information instead of only showing this text:
```
[FAILED] DietPi-Update | Root privileges required. Please run the command with "G_SUDO", or, "sudo -s".
```

UNiXMIT, 2019-05-17:
Did it help though? It's not usually needed.
I've just tried it and yes sudo -s dietpi-update returns that error. Try it without the -s and it should work fine.

Sopor, 2019-05-17:
But the instructions say `sudo -s` so i did use what it tells me to do. The `-s` (runs a shell with root privileges) is added for some reasons or else it could be left out.

MichaIng, 2019-05-17:
@Sopor @UNiXMIT 
Jep the "-s" is not correct here. The info was changed with v6.23.

Please simply do: `sudo dietpi-update`

UNiXMIT, 2019-05-17:
Ah yes it has changed in 6.23. Great.

closed 2019-05-17 by MichaIng