MichaIng/DietPi #2811: Installing "16 Build-Essentials: common packages for compile" fails

maartenlangeveld, 2019-05-17:
#### Details:
- Date           | Fri 17 May 16:50:00 CEST 2019
- Bug report     | N/A
- DietPi version | v6.24.0 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi Zero W (armv6l) (index=1)
- Kernel version | #1200 Tue Feb 12 20:11:02 GMT 2019
- Distro         | stretch (index=4)
- Command        | G_AGI build-essential make autoconf automake
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2019-05-17:
@maartenlangeveld 
Could you please paste the full output of: `apt install build-essential make autoconf automake`

maartenlangeveld, 2019-05-17:
@MichaIng 

Will do that later tonight or tomorrow. For me this issue has no priority at all since I do not need the build-essentials at all, I just selected it by accident but since it failed I have reported it.

Thanks,

Maarten

MichaIng, 2019-05-18:
@maartenlangeveld 
Okay great, then I mark this issue as closed for now. As said feel free to reopen if you ever require build-essential. Most likely other users run into this as well, so most likely we will have a step-by-step solution ready until then.

closed 2019-05-18 by MichaIng

Velociraptor85, 2024-08-18:
@MichaIng hey, I was wondering if there was some change as apparently according to https://github.com/MichaIng/DietPi/wiki/DietPi-Software-list 16 is now a different package ?

Joulinar, 2024-08-18:
We removed the dedicated software option a while ago. If needed you simply can install Debian software package https://packages.debian.org/bookworm/build-essential