MichaIng/DietPi #2813: RaspAP web interface to control wifi

symbios24, 2019-05-18:
### Creating a software request

#### Formal software information
- Software title | RaspAp
- Short description | A simple, responsive web interface to control wifi
- GitHub URL (if available) | https://github.com/billz/raspap-webgui
- (Official) install documentation | https://github.com/billz/raspap-webgui#quick-installer

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

Quick installer
Install RaspAP from your RaspberryPi's shell prompt:

wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap
The installer will complete the steps in the manual installation (below) for you.

After the reboot at the end of the installation the wireless network will be configured as an access point as follows:

IP address: 10.3.141.1
Username: admin
Password: secret
DHCP range: 10.3.141.50 to 10.3.141.255
SSID: raspi-webgui
Password: ChangeMe
Note: As the name suggests, the Quick Installer is a great way to quickly setup a new AP. However, it does not automagically detect the unique configuration of your RPi. Best results are obtained by connecting an RPi to ethernet (eth0) or as a WiFi client, also known as managed mode, with wlan0

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
 yes
- ...

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2019-05-18:
@symbios24 
Many thanks for your request.

Hmm, I bet this one is too intrusive, breaking/colliding with dietpi-config and dietpi-software in many ways. E.g.: https://github.com/billz/raspap-webgui/blob/master/installers/raspbian.sh#L12
Install Lighttpd hardcoded instead of leaving webserver setup optional for end user. Installs hostapd from APT where we in some cases require a custom build for RTL8818* chips to run. Most likely it also overrides Lighttpd config and turns the device into a single-purpose one, not what we can support.

But I will have a closer look when I can.

cocoflan, 2019-05-21:
> ### Creating a software request
> #### Formal software information
>
>     * Software title | RaspAp
>
>     * Short description | A simple, responsive web interface to control wifi
>
>     * GitHub URL (if [...]

ikkerus, 2023-03-03:
RaspAP has been updated many times now and it uses this script (from install.raspap.com):
I beleive this can be adopted to suit the needs of DietPi and the dependencies to e.g. lighthttpd.

`#!/bin/bash
#
# RaspAP Quick Installer
# Author: @billz <billzimmerman@gmail.com>
# Author URI: https://github.com/billz/
# License: GNU General Public License v3.0
# License URI: https://github.com/raspap/raspap-webgui/blob/master/LICENSE
#
# Usage: raspbian.sh options
#
# Installs an instance of RaspAP.
#
# OPTIONS:
# -y, --yes, --assume-yes           Assume "yes" as answer to all prompts and run non-interactively
# -c, --cert, --certficate          Installs mkcert and generates an SSL certificate for lighttpd
# -o, --openvpn <flag>              Used with -y, --yes, sets OpenVPN install option (0=no install)
# -a, --adblock <flag>              Used with -y, --yes, sets Adblock install option (0=no install)
# -r, --repo, --repository <name>   Overrides the default GitHub repo (RaspAP/raspap-webgui)
# -b, --branch <name>               Overrides the default git branch (master)
# -t, --token <accesstoken>         Specify a GitHub token to access a private repository
# -u, --upgrade                     Upgrades an existing installation to the latest release version
# -i, --insiders                    Installs from the Insiders Edition (RaspAP/raspap-insiders)
# -v, --version                     Outputs release info and exits
# -n, --uninstall		    Loads and executes the uninstaller
# -h, --help                        Outputs usage notes and exits
#
# Depending on options passed to the installer, ONE of the following
# additional shell scripts will be downloaded and sourced:
#
# https://raw.githubusercontent.com/raspap/raspap-webgui/master/installers/common.sh
# - or -
# https://raw.githubusercontent.com/raspap/raspap-webgui/master/installers/mkcert.sh
# - or -
# https://raw.githubusercontent.com/raspap/raspap-webgui/master/installers/uninstall.sh
#
# You are not obligated to bundle the LICENSE file with your RaspAP projects as long
# as you leave these references intact in the header comments of your source files.

set -eo pipefail

function _main() {
    # set defaults
    repo="RaspAP/raspap-webgui" # override with -r, --repo option
    repo_common="$repo"
    _parse_params "$@"
    _setup_colors
    _log_output
    _load_installer
}

function _parse_params() {
    # default option values
    assume_yes=0
    upgrade=0
    ovpn_option=1
    adblock_option=1
    insiders=0
    acctoken=""

    while :; do
        case "${1-}" in
            -y|--yes|--assume-yes)
            assume_yes=1
            apt_option="-y"
            ;;
            -o|--openvpn)
            ovpn_option="$2"
[... 183 lines cut ...]
    header=()
    if [[ ! -z "$acctoken" ]]; then
        header=(--header "Authorization: token $acctoken")
    fi

    if [ "${install_cert:-}" = 1 ]; then
        source="mkcert"
        wget "${header[@]}" -q ${UPDATE_URL}installers/${source}.sh -O /tmp/raspap_${source}.sh
        source /tmp/raspap_${source}.sh && rm -f /tmp/raspap_${source}.sh
        _install_certificate || _install_status 1 "Unable to install certificate"
    elif [ "${uninstall}" = 1 ]; then
        source="uninstall"
        wget "${header[@]}" -q ${UPDATE_URL}installers/${source}.sh -O /tmp/raspap_${source}.sh
        source /tmp/raspap_${source}.sh && rm -f /tmp/raspap_${source}.sh
        _remove_raspap || _install_status 1 "Unable to uninstall RaspAP"
    else
        source="common"
        wget "${header[@]}" -q ${UPDATE_URL}installers/${source}.sh -O /tmp/raspap_${source}.sh
        source /tmp/raspap_${source}.sh && rm -f /tmp/raspap_${source}.sh
        _install_raspap || _install_status 1 "Unable to install RaspAP"
    fi
}

_main "$@"
`

symbios24, 2023-03-03:
Hello

I think i tried before Raspap it was not working correctly for me,maybe i missed something

In case you put it in dietpi,please don't remove or merge it with the traditional wifi hotspot software,put it as a separate software.

Thank you

billz, 2023-03-03:
@ikkerus the install loader (raspbian.sh) calls one of several shell scripts depending on options passed to it.

RaspAP's [common install functions](https://github.com/RaspAP/raspap-webgui/blob/7cf64f789343c4f4e04360c0492bb28f744969a6/installers/common.sh) could be forked, if necessary, and we could look into adding support for an external installer.

DN2048, 2023-09-12:
I had some free time to spare so I installed a fresh copy of DietPi on a Raspberry PI 4B and after it was finished I installed RaspAP. The RaspAP worked perfectly fine to my surprise.
I then installed Pi Hole and everything broke (apparently). I didn't had much time left to investigate what went wrong or if I had to do something before installing Pi Hole or if I should have installed Pi Hole first then RaspAP but what I can conclude is that having RaspAP working with DietPi doesn't seem that hard (no problems whatsoever on a fresh install).

DietPi v8.21
RaspAP v2.9.4

MichaIng, 2023-09-12:
If you enable the DHCP server of Pi-hole, it breaks executions of any other DHCP server, including the one from RaspAP. Some extra `dmsmasq` config would be required to bind Pi-hole related configs to the Ethernet-side interface, so that the second DHCP server from RaspAP can run on the WiFi interface. Or best of course is to use one and the same DHCP server for both interfaces.

We discussed it here: #6485
It is about the DietPi WiFi HotSpot, but the same underlying issue generally applies for any combination of Pi-hole/AGH and any AP implementation, when both run a DHCP server.