MichaIng/DietPi #2814: DietPi-LetsEncrypt Issue v6.24

Borotes, 2019-05-18:
Hey guys,

I Just reinstalled my nextcloud after the problems I had in the past, to get back to stable.

I would like to run certbot but am getting this issue:

DietPi-LetsEncrypt
─────────────────────────────────────────────────────
 Mode: Running CertBot

[  OK  ] DietPi-LetsEncrypt | Lighttpd webserver detected
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator webroot, Installer None
Obtaining a new certificate
Performing the following challenges:
http-01 challenge for cloud.****.de
Using the webroot path /var/www for all unmatched domains.
Waiting for verification...
Cleaning up challenges
Failed authorization procedure. cloud.****.de (http-01): urn:ietf:params:acme:error:unauthorized :: The client lacks sufficient authorization :: Invalid response from http://cloud.****.de/.well-known/acme-challenge/aQgWxlZPtvLgB_nt4KHaeaneCiHRZR2cjtWbijNs4I0 [84.159.19.29]: "<?xml version=\"1.0\" encoding=\"iso-8859-1\"?>\n<!DOCTYPE html PUBLIC \"-//W3C//DTD XHTML 1.0 Transitional//EN\"\n         \"http://www."

IMPORTANT NOTES:
 - The following errors were reported by the server:

   Domain: cloud.****.de
   Type:   unauthorized
   Detail: Invalid response from
   http://cloud.****.de/.well-known/acme-challenge/aQgWxlZPtvLgB_nt4KHaeaneCiHRZR2cjtWbijNs4I0
   [84.159.19.29]: "<?xml version=\"1.0\"
   encoding=\"iso-8859-1\"?>\n<!DOCTYPE html PUBLIC \"-//W3C//DTD
   XHTML 1.0 Transitional//EN\"\n         \"http://www."

   To fix these errors, please make sure that your domain name was
   entered correctly and the DNS A/AAAA record(s) for that domain
   contain(s) the right IP address.
[FAILED] CertBot failed with error code (1), please check it's terminal output. Aborting...
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | ncc maintenance:mode --on
[  OK  ] DietPi-Services | restart : fail2ban
[  OK  ] DietPi-Services | restart : redis-server
[ ..   ] DietPi-Services | mariadb

I already checked my router and DynDNS but still can access it via http...

Do you have any idea, what could be wrong?

Thanks in advance! 😊

MichaIng, 2019-05-18:
@Borotes 
Many thanks for your report.

Could you verify the Lighttpd server is up correctly:
```
journalctl -u lighttpd
cat /var/log/lighttpd/error.log # Or similar
```

Borotes, 2019-05-18:
Hi @MichaIng 😊

Thanks for your quick reply!

I am running the latest dietpi version on a raspberry pi 3b+. I installed Nextcloud, Fail2Ban, Certbot and 
Midnight Commander.

The lighttpd log is empty 😕

Here you have the LetsEncrypt log:
{
  "keyChange": "https://acme-v02.api.letsencrypt.org/acme/key-change",
  "meta": {
    "caaIdentities": [
      "letsencrypt.org"
    ],
    "termsOfService": "https://letsencrypt.org/documents/LE-SA-v1.2-November-15-2017.pdf",
    "website": "https://letsencrypt.org"
  },
  "newAccount": "https://acme-v02.api.letsencrypt.org/acme/new-acct",
  "newNonce": "https://acme-v02.api.letsencrypt.org/acme/new-nonce",
  "newOrder": "https://acme-v02.api.letsencrypt.org/acme/new-order",
  "revokeCert": "https://acme-v02.api.letsencrypt.org/acme/revoke-cert",
  "ts_aoXyh9zU": "https://community.letsencrypt.org/t/adding-random-entries-to-the-directory/33417"
}
2019-05-18 22:18:26,436:INFO:certbot.main:Obtaining a new certificate
2019-05-18 22:18:36,398:DEBUG:certbot.crypto_util:Generating key (4096 bits): /etc/letsencrypt/keys/0005_key-certbot.pem
2019-05-18 22:18:36,579:DEBUG:certbot.crypto_util:Creating CSR: /etc/letsencrypt/csr/0005_csr-certbot.pem
2019-05-18 22:18:36,584:DEBUG:acme.client:Requesting fresh nonce
2019-05-18 22:18:36,584:DEBUG:acme.client:Sending HEAD request to https://acme-v02.api.letsencrypt.org/acme/new-nonce.
2019-05-18 22:18:36,792:DEBUG:requests.packages.urllib3.connectionpool:https://acme-v02.api.letsencrypt.org:443 "HEAD /acme/new-nonce HTTP/1.1" 200 0
2019-05-18 22:18:36,795:DEBUG:acme.client:Received response:
HTTP 200
Server: nginx
Link: <https://acme-v02.api.letsencrypt.org/directory>;rel="index"
Replay-Nonce: BU-zk0Q9OiQtQjv3zAJgD8-IsJfdLyuwA983dNZiU84
X-Frame-Options: DENY
Strict-Transport-Security: max-age=604800
2019-05-18 22:18:25,921:DEBUG:certbot.main:certbot version: 0.28.0
2019-05-18 22:18:25,927:DEBUG:certbot.main:Arguments: ['--webroot', '-w', '/var/www', '--agree-tos', '--no-eff-email', '--rsa-key-size', '4096', '-m', '****@outlook.com', '-d', 'cloud.****.de']
2019-05-18 22:18:25,931:DEBUG:certbot.main:Discovered plugins: PluginsRegistry(PluginEntryPoint#manual,PluginEntryPoint#null,PluginEntryPoint#standalone,PluginEntryPoint#webroot)
2019-05-18 22:18:25,981:DEBUG:certbot.log:Root logging level set at 20
2019-05-18 22:18:25,986:INFO:certbot.log:Saving debug log to /var/log/letsencrypt/letsencrypt.log
2019-05-18 22:18:25,989:DEBUG:certbot.plugins.selection:Requested authenticator webroot and installer None
2019-05-18 22:18:25,991:DEBUG:certbot.plugins.selection:Single candidate plugin: * webroot
Description: Place files in webroot directory
Interfaces: IAuthenticator, IPlugin
Entry point: webroot = certbot.plugins.webroot:Authenticator
Initialized: <certbot.plugins.webroot.Authenticator object at 0x769a5e50>
Prep: True
2019-05-18 22:18:25,997:DEBUG:certbot.plugins.selection:Selected authenticator <certbot.plugins.webroot.Authenticator object at 0x769a5e50> and installer None
2019-05-18 22:18:25,997:INFO:certbot.plugins.selection:Plugins selected: Authenticator webroot, Installer None
2019-05-18 22:18:26,039:DEBUG:certbot.main:Picked account: <Account(RegistrationResource(uri='https://acme-v02.api.letsencrypt.org/acme/acct/57337791', terms_of_service=None, body=Registration(terms_of_service_agreed=None, contact=(), o$
2019-05-18 22:18:26,047:DEBUG:acme.client:Sending GET request to https://acme-v02.api.letsencrypt.org/directory.
2019-05-18 22:18:26,070:DEBUG:requests.packages.urllib3.connectionpool:Starting new HTTPS connection (1): acme-v02.api.letsencrypt.org
2019-05-18 22:18:26,431:DEBUG:requests.packages.urllib3.connectionpool:https://acme-v02.api.letsencrypt.org:443 "GET /directory HTTP/1.1" 200 658
2019-05-18 22:18:26,434:DEBUG:acme.client:Received response:
HTTP 200
Server: nginx
Content-Type: application/json
Content-Length: 658
X-Frame-Options: DENY
Strict-Transport-Security: max-age=604800
Expires: Sat, 18 May 2019 20:18:26 GMT
Cache-Control: max-age=0, no-cache, no-store
Pragma: no-cache
Date: Sat, 18 May 2019 20:18:26 GMT
Connection: keep-alive

MichaIng, 2019-05-18:
@Borotes 
> [ OK ] DietPi-LetsEncrypt | Lighttpd webserver detected

and

> Server: nginx

Hmm which webserver do you have? Or are Nginx and Lighttpd both installed somehow?

Borotes, 2019-05-18:
@MichaIng I just checked DietPi-Software and only Lighttpd is installed.

https://i.imgur.com/mKiCINo.png
https://i.imgur.com/BlKfTED.png

MichaIng, 2019-05-18:
@Borotes 
`dpkg -l | grep nginx` also does not show any Nginx package installed?

Ahh, sorry this must be the webserver of Let's Encrypt, not your local one of course, my fail 😜.

If you run `certbot renew` it shows the same error as `dietpi-letsencrypt`?

Borotes, 2019-05-18:
@MichaIng 
root@DietPi:~# dpkg -l | grep nginx
root@DietPi:~# certbot renew
Saving debug log to /var/log/letsencrypt/letsencrypt.log

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -

No renewals were attempted.
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
root@DietPi:~#

MichaIng, 2019-05-18:
@Borotes 
Ah okay so indeed to certs exist yet.

You can access you webserver (Nextcloud) via browser from outside your local network successfully, right?
Also assure permissions: `chown -R www-data:www-data /var/www`

Could you try the naked certbot command. You need to enter info interactively then: `certbot --webroot -w /var/www`

Borotes, 2019-05-19:
@MichaIng Yes, I can access Nextcloud from outside of my local network, that means my port forwarding and dyndns are working.

root@DietPi:~# chown -R www-data:www-data /var/www
root@DietPi:~# certbot --webroot -w /var/www
Saving debug log to /var/log/letsencrypt/letsencrypt.log
With the webroot plugin, you probably want to use the "certonly" command, eg:

    certbot certonly --webroot

(Alternatively, add a --installer flag. See https://eff.org/letsencrypt-plugins
 and "--help plugins" for more information.)

Tried again with the command from the cosole:

root@DietPi:~# certbot certonly --webroot -w /var/www
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator webroot, Installer None
Please enter in your domain name(s) (comma and/or space separated)  (Enter 'c'
to cancel): cloud.****.de
Obtaining a new certificate
Performing the following challenges:
http-01 challenge for cloud.****.de
Using the webroot path /var/www for all unmatched domains.
Waiting for verification...
Cleaning up challenges
Failed authorization procedure. cloud.****de (http-01): urn:ietf:params:acme:error:unauthorized :: The client lacks sufficient authorization :: Invalid response from http://cloud.****.de/.well-known/acme-challenge/2MN_l8lBQ2Rd-qNcsDvdLfs14uMVdyD12zgHtbCpVok [84.159.19.29]: "<?xml version=\"1.0\" encoding=\"iso-8859-1\"?>\n<!DOCTYPE html PUBLIC \"-//W3C//DTD XHTML 1.0 Transitional//EN\"\n         \"http://www."

IMPORTANT NOTES:
 - The following errors were reported by the server:

   Domain: cloud.****.de
   Type:   unauthorized
   Detail: Invalid response from
   http://cloud.****.de/.well-known/acme-challenge/2MN_l8lBQ2Rd-qNcsDvdLfs14uMVdyD12zgHtbCpVok
   [84.159.19.29]: "<?xml version=\"1.0\"
   encoding=\"iso-8859-1\"?>\n<!DOCTYPE html PUBLIC \"-//W3C//DTD
   XHTML 1.0 Transitional//EN\"\n         \"http://www."

   To fix these errors, please make sure that your domain name was
   entered correctly and the DNS A/AAAA record(s) for that domain
   contain(s) the right IP address.

The only config I edited myself is the ligttpd config. I changed....

‘server.document-root = "/var/www/html"' to "/var/www/nextcloud"

... that I don't have to type https://cloud.domain.de/nextcloud

https://i.imgur.com/Se9axSV.png

Maybe an typing error?

MichaIng, 2019-05-19:
@Borotes 
> ‘server.document-root = "/var/www/html"' to "/var/www/nextcloud"

Ahhh of course then `-w /var/www` cannot work. Run:
```
certbot certonly --webroot -w /var/www/nextcloud
dietpi-letsencrypt
```
- In `dietpi-letsencrypt` enter the same domain as you did before in the plain certbot command (+ additional infos). It will then just attempt a renewal and apply the webserver settings afterwards., which should work out of the box then.

Note that changing the webroot will cause issues with other software installs as well. DietPi simply expect it to be `/var/www` and in some cases also reverts it to be there. But most probably you want that if you install a second website anyway.

We have an open request about to add support for reaching single web applications without path in the URL, but it is a larger task to integrate: https://github.com/MichaIng/DietPi/issues/1608

Borotes, 2019-05-19:
@MichaIng 

It worked like a charme each time till I reinstalled yesterday. I wrote a guide, to remember what I did and this time it didn't work. Don't know what changed...

> certbot certonly --webroot -w /var/www/nextcloud

With that command it worked, but with `dietpi-letsencrypt` the issue remains 🤔

MichaIng, 2019-05-19:
@Borotes 
Does dietpi-letsencrypt still try to receive a new certificate? Actually when it finds existing ones, it should only call `certbot renew`, telling you that no renewal is required. Are you sure you used the exact same domain on both?

Borotes, 2019-05-19:
@MichaIng Yes, sorry... I tried it with `dietpi-letsencrypt` first, checked if everything was typed correctly and then ran certbot manually.

MichaIng, 2019-05-19:
@Borotes 
> I tried it with dietpi-letsencrypt first

And you tried with dietpi-letsencrypt afterwards again, resulting in the same error as above? https://github.com/MichaIng/DietPi/issues/2814#issue-445700175

Borotes, 2019-05-19:
@MichaIng When I did it manually I got my certificate. Afterwards it dietpi-letsencrypt does what it is meat to do, telling me I don't have to renew it.

MichaIng, 2019-05-19:
@Borotes 
Okay, perfect. So HTTPS works now as desired, doesn't it?

Borotes, 2019-05-19:
@MichaIng Yes, thanks for your help!

MichaIng, 2019-05-19:
@Borotes 
You're welcome. I mark this issue as closed then. Feel free to reopen if required.

closed 2019-05-19 by MichaIng