MichaIng/DietPi #2815: Software Request: Firefox Browser

symbios24, 2019-05-18:
#### Formal software information
- Software title | Firefox Browser
- Short description | <-Web Browser->

#### Are there similar/alternative software titles available with DietPi-Software?

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

MichaIng, 2019-05-18:
@symbios24 
Many thanks for your request.

Actually we ship Firefox as default web browser with all desktop installs. Do you want it available as standalone install? But since a desktop is required anyway for this, I don't see a reason currently?

symbios24, 2019-05-23:
yes i want it as standalone install ,i have tested it and runs standalone with the same method as chrome browser

MichaIng, 2021-05-18:
Done for next release, including a preference menu to be respected when a desktop is installed: #4314

closed 2021-05-18 by MichaIng