MichaIng/DietPi #2816: v6.24 new lighhtpd version config file breaks 'Now Playing' page O!MPD when playlist is empty

maartenlangeveld, 2019-05-18:
@MichaIng 

Hi Micha,

With the new lighttpd version in v6.24, there is also a new lighhtpd.conf config file shipped.
This breaks one specific webpage ('Now Playing') under a specific condition (empty playlist): "400 Bad Request" when clicking ('Now Playing') oon O!MPD startpage.

After som trial and error I managed to discover which element in the new config file was the cause. Just to let you know:

New version lighttpd added following lines to lighttpd.conf. 
When I disable "url-ctrls-reject" the "Now Playing" page in O!MPD page loads again when playlist is empty. If enabled (as default with new lighttpd versions), I get ""400 Bad Request" error message.

# strict parsing and normalization of URL for consistency and security
# https://redmine.lighttpd.net/projects/lighttpd/wiki/Server_http-parseoptsDetails
# (might need to explicitly set "url-path-2f-decode" = "disable"
#  if a specific application is encoding URLs inside url-path)
server.http-parseopts = (
  "header-strict"           => "enable",# default
  "host-strict"             => "enable",# default
  "host-normalize"          => "enable",# default
  "url-normalize-unreserved"=> "enable",# recommended highly
  "url-normalize-required"  => "enable",# recommended
 #"url-ctrls-reject"        => "enable",# recommended -->Disabled. breaks O!MPD 'Now Playing' when playlist is emtpy.
  "url-path-2f-decode"      => "enable",# recommended highly (unless breaks app)
 #"url-path-2f-reject"      => "enable",
  "url-path-dotseg-remove"  => "enable",# recommended highly (unless breaks app)
 #"url-path-dotseg-reject"  => "enable",
 #"url-query-20-plus"       => "enable",# consistency in query string
)

MichaIng, 2019-05-18:
@maartenlangeveld 
Many thanks for reporting and providing the solution.

Will have to check what this setting actually does. Also we currently ship an outdated version of O!MPD due to database sync errors with the (that time) current one. But will retest this and perhaps the new version works with new Lighttpd defaults as well.
The same might be true for myMPD.

maartenlangeveld, 2019-05-18:
**Hi @MichaIng,**

Have just checked latest O!MPD versions (git clone from O!MPD github repo). Works fine at first sight. No databases errors. Plays my catolog and streams (radio) as well. At first sight seems okay with me.

MichaIng, 2019-05-18:
@maartenlangeveld 
Thanks for testing. Have to find the existing issues about that. Fourdee faced the issues that time, at best he can verify it works now to have the device/setup tested that failed.

I will also test on VM and RPi now.

maartenlangeveld, 2019-05-20:
@MichaIng @Fourdee 

Meanwhile has been solved by O!MPD @ArturSierzant in GitHub master.

Maarten

MichaIng, 2019-05-24:
Jep fixed in latest master, although we generally should go with latest release. Here is how we'll do it: https://github.com/ArturSierzant/OMPD/issues/86#issuecomment-495430679

MichaIng, 2019-06-03:
PR up: https://github.com/MichaIng/DietPi/pull/2884

closed 2019-07-03 by MichaIng