MichaIng/DietPi #2817: convert dietpi-update to use apt-fast

mzramna, 2019-05-18:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- apt is very slow when you update a system

#### Describe the solution you'd like:

- so it could use apt-fast to speed up this,apt-fast is an apt but with the modification to download using aria2c,and this could download multiple dependencies at same time,and is faster than the official apt-get,apt or aptitude

MichaIng, 2019-05-18:
@mzramna 
Many thanks for your request.

We did have this in the past and re-validated some time ago, and results was that `apt-fast` is not faster, but slower + includes the overhead of Aria2 etc...: https://github.com/MichaIng/DietPi/issues/1987#issuecomment-410004392

Could you show some (or your own) test results that prove it to be significantly faster?

mzramna, 2019-05-24:
well,i've tried this into mine pcs,not the pi,it just have any speedup when you do an system upgrade,or install any aplication with multiple dependencies/multiple aplications at same time.
the apt-fast made diference JUST when you parallelize the download,like download 10 or 20 files download at same time,lass then this....it is realy useless

mzramna, 2019-05-24:
i could do this tests,but...i really dont know how to record this time diference...could you show me how to do it?

MichaIng, 2019-05-24:
@mzramna 
E.g.:
```
time apt-get -y install lxde
time apt-fast -y install lxde
```
- `lxde` just because it ships a lot of dependencies, thus a lot of concurrent downloads possible
- Of course you need to purge all related packages + cache afterwards: `G_AGP lxde; G_AGA; apt clean`
- I did the tests on a VM, so I was able to simply revert to a snapshot before doing the install so both runs happen on the exact same system state. `dietpi-backup` would be possible as well. This is just nothing that should be done on a SBC with SDcard due to high amount of SD writes.

Alternatively you choose a large amount of smaller packages to test with, e.g.:
net-tools, p7zip-full, wakeonlan, etherwake, iotop, iptraf, ... or such. Just do the G_AGP ..., G_AGA, apt clean steps to revert those installs + dependencies and clean the cache.
____________
As said, apt-fast sounds reasonable at first, but the effort to contact the different mirrors find the packages on each, load Aria2 to call the final APT command, once for each package, is so large that for any benefit it requires:
- a fast download capability, so that downloading a single package does only fill your bandwidth by a small share and
- slow APT mirrors, so that those can only fill your download bandwidth by a small share.

Otherwise the server side preparations and overhead is simply too high. See my test results, even with 12 concurrent downloads it was overall slower than pure apt-get. And I have 100 Mbit download bandwidth.

MichaIng, 2019-06-25:
I mark this as closed. I opened an issue on apt-fast GitHub to ask for benchmarks (and get notifications in case of reply). Did some more testing there and could still very hardly find any benefit. Basically it depends on the chosen APT mirror. I found some that are not able to use my full bandwidth, where apt-fast then was slightly faster sometimes. But im most cases it was slower, sometimes dramatically, especially after choosing the best local APT mirror for apt-get.

Also apt-fast. to have any benefit, requires some custom configuration to find multiple fast mirrors. There is some tool that should test then and list the 10 fastest, but it does not show correct result, but slower mirrors in my case. Not sure if it's ping vs bandwidth, however all this effort and overhead still makes me believe that it is not worth it.

But leaned something about Aria2:
- It is able to also download single files from multiple mirrors, downloading and merging chunks.
- So the benefit (if found) is independent from the amount of packages being downloaded.

closed 2019-06-25 by MichaIng