MichaIng/DietPi #2823: Radar eats all my RAM

esturniolo, 2019-05-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | `6.24`
- Distro version | 9.9
- Kernel version | Linux EasSRV 4.9.0-6-amd64 #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux
- SBC device | Native PC (x86_64)
- Power supply used | Original Laptop supply.
- SDcard used | N/A

#### Additional Information (if applicable)
- Software title | RADARR
- Was the software title installed freshly or updated/migrated? Freshly
- Can this issue be replicated on a fresh installation of DietPi? Yes, I guess.

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce
Just install Radarr

#### Expected behaviour
Consume reasonable amount of RAM

#### Actual behaviour
Eat more RAM than Chrome (and that's is much to say)

#### Extra details
Here may be a solution -> https://github.com/Radarr/Radarr/issues/1580#issuecomment-487583782
The problem is the "mono" version.
May be if you use another version (downgrade) the issue will be solved.

Please note that I'm using a x64 DietPi version with 2Gb RAM and isn't enough.
[image]

esturniolo, 2019-05-18:
Sorry, I mean CPU, not RAM.
It eats all my CPU.
And the temp it's a problem too...

[image]

MichaIng, 2019-05-18:
@esturniolo 
Many thanks for your report. Will try to replicate. I just tested it a week ago or so without any issues.

But note that Radarr takes a little while to load/compile which indeed consumes much CPU. But should not be more then half a minute at most.

esturniolo, 2019-05-18:
> @esturniolo
> Many thanks for your report. Will try to replicate. I just tested it a week ago or so without any issues.
> 
> But note that Radarr takes a little while to load/compile which indeed consumes much CPU. But should not be more then half a minute at most.

Yes, but this was installed and running properly.
It just finished to download a 7Gb file and keep the same consumption.
In the link that I added, you can see that this issue came from long time. And still remains.

Please tell me if can I help you with something.

MichaIng, 2019-05-19:
@esturniolo 
Okay just tested on x86 VM and works fine here:
```
root@VM-Stretch:~# mono -V
Mono JIT compiler version 5.20.1.19 (tarball Thu Apr 11 09:07:22 UTC 2019)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
        TLS:           __thread
        SIGSEGV:       altstack
        Notifications: epoll
        Architecture:  amd64
        Disabled:      none
        Misc:          softdebug
        Interpreter:   yes
        LLVM:          yes(600)
        Suspend:       hybrid
        GC:            sgen (concurrent by default)
```
- CPU usage at 0% constantly.

Does your Radarr log (from web UI or `cat /var/log/radarr/radarr.txt`) show anything suspicious?

esturniolo, 2019-05-19:
Not really...
`19-5-18 22:20:46.2|Info|PreDBService|Starting PreDB Sync
19-5-18 22:20:46.3|Info|NetImportSearchService|No lists are enabled for auto-import.
19-5-18 22:22:46.8|Info|RssSyncService|Starting RSS Sync
19-5-18 22:22:50.5|Info|DownloadDecisionMaker|Processing 100 releases
19-5-18 22:23:18.1|Info|RssSyncService|RSS Sync Completed. Reports found: 100, Reports grabbed: 0`

I changed link the DB to RAM and works. Even the temp started to down. I'm now in 53 Celsius.

When you mention

> settings and download status changes are lost

do you mean that all the things that I made only will sync to disk with a regular shutdown?
Because we have some energy troubles here (Argentina) and and the power is cut off from time to time.

MichaIng, 2019-05-19:
@esturniolo 
> do you mean that all the things that I made only will sync to disk with a regular shutdown?

Jep exactly. Not the downloads themselves are lost, but Radarr will not know any more about new tasks you added, about which downloads already have finished and all that. Not sure how well it can simply re-scan the download dir to update info 🤔.

To have the database regularly synced to disk you can use the tasks implementation of Radarr. Read the info at the bottom of this post: https://dietpi.com/phpbb/viewtopic.php?f=8&t=5828

If it's still to risky for due, when your power supply is not reliable enough:
`/DietPi/dietpi/misc/dietpi-arr_to_RAM disable` will restore stuff back to disk and disable the related service.
______________
Interesting actually that it helped so well. Was the CPU usage going down as well?
Perhaps also the Radarr service restart fixed some hanging task or such, not sure.

esturniolo, 2019-05-20:
Hi. I'm here again.
Thanks for your help. As I said before you can't trust in the power here and I had a power outage yesterday...
I reversed the dietpi-arr_to_RAM to disable and automagically now I've 0 CPU usage.

[image]

[image]

So, we know that the problem here is mono, but for some reason that I can't sometimes works like a charm (like now) and others don't (like Saturday).
I don't know what will happen when I start to use radarr frequently. But for now it works.

If you need it I can provide you some logs or whatever you need.

MichaIng, 2019-05-20:
@esturniolo 
Okay great it works now.

Jackett ships a standalone version now that does not require mono anymore. So I checked it for Radarr: https://github.com/Radarr/Radarr/wiki/Installation#debian--ubuntu
- First I thought, well no mono package installed (like for Fedora) but having a closer look: [libmono-cil-dev](https://packages.debian.org/stretch/libmono-cil-dev) and also the systemd service runs Radarr with mono...

Gabba35, 2019-05-22:
Having the same problem with my C2 but with Jackett and Sonarr rather than Radarr. High CPU usage and Sonarr crashes.

(For some reason my bug reporting doesn't work either and hasn't for the last few installs.)

esturniolo, 2019-05-22:
I've a very strange behaviour.
I recovered the previuos backup **before** install Radarr and NZBGet in the same x86 machine.
Then I install again the same packages with `dietpi-software`. So I made the same steps that I did in the first installation.
Again the CPU at 70%/80% all the time.
I rebooted the machine (like I did in the same installation too) and magically the CPU started to works at 2%/3%.

I searched for a movie in Raddar. I'd peaks of 50% CPU or more, but totally normal. They are just peaks. 
Started to download with NZBGet.
Today I wake up and check the CPU and...

[image]

Reboot again and...

[image]

It has no sense.

MichaIng, 2019-05-22:
@esturniolo 
Indeed... I will again install it on test machine and let it run until going to sleep. Lets see if it also starts to eat CPU after a while.

Just a small hint:
- You have a 54M swap file. I know this is default since DietPi by default fills memory up to 2G with the swap file. But actually that does not make much sense.
- Either it should have a significant size, some hundred MiB at least, or it should be disabled.
- Please try to disable it. It was full even with more than enough available RAM, possibly mono uses it in a strange way: `dietpi-drive_manager` > /dev/sda1 > Swap file > enter `0` and confirm

MichaIng, 2019-06-08:
@esturniolo 
Does the issue still persists and did you try to disable the swap file?

esturniolo, 2019-06-10:
Hi @MichaIng 
Sorry for my delay in the response.

Finally I chaged the swap to 1024, but only to have some more performance. finally I uninstalled Radarr and I started to use CouchPotato.

MichaIng, 2019-06-11:
@esturniolo 
Okay, whatever was the reason for Radarr causing this high CPU usage. I mark the issue as closed then. Feel free to reopen if required.

closed 2019-06-11 by MichaIng