MichaIng/DietPi #2824: dietpi-update from v6.22.3 to v6.24.0 error

Korpip, 2019-05-18:
--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Sun 19 May 00:43:53 BST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1219 SMP Tue May 14 21:20:58 BST 2019
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. dietpi-update
2. Apply

#### Expected behaviour:

- The power to take over the world

#### Actual behaviour:

- Not updating

#### Extra details:

- I am currently on DietPi version | v6.22.3

#### Additional logs:
```
Log file contents:
Setting up openssh-server (1:7.4p1-10+deb9u6) ...
Job for ssh.service failed because the control process exited with error code.
See "systemctl status ssh.service" and "journalctl -xe" for details.
invoke-rc.d: initscript ssh, action "start" failed.
● ssh.service - OpenBSD Secure Shell server
   Loaded: loaded (/lib/systemd/system/ssh.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2019-05-19 00:43:40 BST; 26ms ago
  Process: 13552 ExecStart=/usr/sbin/sshd -D $SSHD_OPTS (code=exited, status=255)
  Process: 13549 ExecStartPre=/usr/sbin/sshd -t (code=exited, status=0/SUCCESS)
 Main PID: 13552 (code=exited, status=255)

May 19 00:43:40 DietPi systemd[1]: Starting OpenBSD Secure Shell server...
May 19 00:43:40 DietPi systemd[1]: ssh.service: Main process exited, code=e…/n/a
May 19 00:43:40 DietPi systemd[1]: Failed to start OpenBSD Secure Shell server.
May 19 00:43:40 DietPi systemd[1]: ssh.service: Unit entered failed state.
May 19 00:43:40 DietPi systemd[1]: ssh.service: Failed with result 'exit-code'.
Hint: Some lines were ellipsized, use -l to show in full.
dpkg: error processing package openssh-server (--configure):
 subprocess installed post-installation script returned error exit status 1
dpkg: dependency problems prevent configuration of ssh:
 ssh depends on openssh-server (>= 1:7.4p1-10+deb9u6); however:
  Package openssh-server is not configured yet.

dpkg: error processing package ssh (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 openssh-server
 ssh
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

--------------------------------------------------------------------

[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

MichaIng, 2019-05-19:
@Korpip 
Many thanks for your report.

What does the following show:
```
sshd
```

It starts in foreground, in case press ctrl+c to exit.

Korpip, 2019-05-19:
```
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

root@DietPi:~# sshd
sshd re-exec requires execution with an absolute path
root@DietPi:~#   
```

MichaIng, 2019-05-19:
@Korpip 
Intersting: https://unix.stackexchange.com/questions/109380/why-does-sshd-requires-an-absolute-path

So try:
```
/usr/sbin/sshd
```

Korpip, 2019-05-19:
```
root@DietPi:~# /usr/sbin/sshd
root@DietPi:~# sshd
sshd re-exec requires execution with an absolute path
```

```
root@DietPi:/usr/sbin# ls
addgnupghome        delgroup           groupmod              mklost+found         remove-shell            update-locale
addgroup            deluser            grpck                 newusers             rfkill                  update-mime
add-shell           dpkg-preconfigure  grpconv               nologin              rmt                     update-passwd
adduser             dpkg-reconfigure   grpunconv             pam-auth-update      rmt-tar                 update-rc.d
applygnupgdefaults  dropbear           iconvconfig           pam_getenv           rtcwake                 update-usbids
arpd                e2freefrag         invoke-rc.d           pam_timestamp_check  service                 useradd
chgpasswd           e4crypt            ldattach              phpdismod            setvesablank            userdel
chpasswd            e4defrag           lighttpd              phpenmod             sshd                    usermod
chroot              fdformat           lighttpd-angel        php-fpm7.0           start_pms               validlocale
cifs.idmap          filefrag           lighttpd-disable-mod  phpquery             tarcat                  vcstime
cifs.upcall         genl               lighttpd-enable-mod   pwck                 tunelp                  vigr
cpgr                groupadd           lighty-disable-mod    pwconv               tzconfig                vipw
cppw                groupdel           lighty-enable-mod     pwunconv             update-ca-certificates  visudo
cron                groupmems          locale-gen            readprofile          update-info-dir         zic
root@DietPi:/usr/sbin# sshd
sshd re-exec requires execution with an absolute path
root@DietPi:/usr/sbin#
```

I have Dropbear setup in dietpi-software if that information helps.  I do not think I ever went tried OpenSSH.

MichaIng, 2019-05-19:
@Korpip 
> I have Dropbear setup in dietpi-software if that information helps. I do not think I ever went tried OpenSSH.

Ah okay, then an error is expected 😄. Please purge OpenSSH then: `apt purge ssh openssh-server openssh-client`
- Before confirming, check that there is no other package purged besides this three. But I am not aware that there exist anything that explicitly depends on OpenSSH.

Korpip, 2019-05-19:
That did the trick and I was able to complete the update.  I don't remember doing anything with OpenSSH but I am new at this and could have without knowing.

Thank you for your time and effort.

MichaIng, 2019-05-19:
@Korpip 
Great, and you are welcome.

Yeah sometimes we follow some setup guide or something and forget about it later. Or some installer does unexpected things, who knows 😉.

I mark this as closed. Feel free to reopen if required.

closed 2019-05-19 by MichaIng