MichaIng/DietPi #2825: DietPi-Software | WireGuard not working after update to 6.24.1

Joulinar, 2019-05-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.24.1
- Distro version | stretch / 9.9
- Kernel version | Linux DietPi 4.19.42-v7+
- SBC device | RPi 3 Model B+ (armv7l)
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | WireGuard
- WireGuard was already installed before updating from 6.24.0 -> 6.24.1

#### Steps to reproduce
- Just performed `dietpi-update` to update from 6.24.0 -> 6.24.1

#### Expected behaviour
- I would expect WireGuard to be running after update and reboot

#### Actual behaviour
- Not able to start WireGuard

#### Extra details
- Looks that something was removed during `G_AGUP`. See attachement

[wg.txt](https://github.com/MichaIng/DietPi/files/3194622/wg.txt)

I tried to manual restart using  `systemctl restart wg-quick@wg0`

```
root@DietPi:/etc/wireguard# systemctl status wg-quick@wg0.service
● wg-quick@wg0.service - WireGuard via wg-quick(8) for wg0
   Loaded: loaded (/lib/systemd/system/wg-quick@.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2019-05-19 01:46:28 CEST; 18s ago
     Docs: man:wg-quick(8)
           man:wg(8)
           https://www.wireguard.com/
           https://www.wireguard.com/quickstart/
           https://git.zx2c4.com/WireGuard/about/src/tools/man/wg-quick.8
           https://git.zx2c4.com/WireGuard/about/src/tools/man/wg.8
  Process: 1325 ExecStart=/usr/bin/wg-quick up wg0 (code=exited, status=1/FAILURE)
 Main PID: 1325 (code=exited, status=1/FAILURE)

Mai 19 01:46:27 DietPi systemd[1]: Starting WireGuard via wg-quick(8) for wg0...
Mai 19 01:46:28 DietPi wg-quick[1325]: [#] ip link add wg0 type wireguard
Mai 19 01:46:28 DietPi wg-quick[1325]: RTNETLINK answers: Operation not supported
Mai 19 01:46:28 DietPi wg-quick[1325]: Unable to access interface: Protocol not supported
Mai 19 01:46:28 DietPi wg-quick[1325]: [#] ip link delete dev wg0
Mai 19 01:46:28 DietPi wg-quick[1325]: Cannot find device "wg0"
Mai 19 01:46:28 DietPi systemd[1]: wg-quick@wg0.service: Main process exited, code=exited, status=1/FAILURE

Mai 19 01:46:28 DietPi systemd[1]: Failed to start WireGuard via wg-quick(8) for wg0.
Mai 19 01:46:28 DietPi systemd[1]: wg-quick@wg0.service: Unit entered failed state.
Mai 19 01:46:28 DietPi systemd[1]: wg-quick@wg0.service: Failed with result 'exit-code'.
root@DietPi:/etc/wireguard#
```

checked as well `journalctl`
```
Mai 19 01:34:31 DietPi wg-quick[571]: [#] ip link add wg0 type wireguard
Mai 19 01:34:31 DietPi wg-quick[571]: RTNETLINK answers: Operation not supported
Mai 19 01:34:31 DietPi wg-quick[571]: Unable to access interface: Protocol not supported
Mai 19 01:34:31 DietPi wg-quick[571]: [#] ip link delete dev wg0
Mai 19 01:34:31 DietPi wg-quick[571]: Cannot find device "wg0"
Mai 19 01:34:31 DietPi systemd[1]: wg-quick@wg0.service: Main process exited, code=exited, status=1/FAILURE
m
Mai 19 01:34:31 DietPi systemd[1]: Failed to start WireGuard via wg-quick(8) for wg0.
-- Subject: Unit wg-quick@wg0.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit wg-quick@wg0.service has failed.
--
-- The result is failed.
Mai 19 01:34:31 DietPi systemd[1]: wg-quick@wg0.service: Unit entered failed state.
Mai 19 01:34:31 DietPi systemd[1]: wg-quick@wg0.service: Failed with result 'exit-code'.
```

MichaIng, 2019-05-19:
@Joulinar 
Since the kernel was upgraded (update just yesterday) you need to rebuild the WireGuard kernel module. 

Do:
```
dpkg-reconfigure wireguard-dkms
systemctl restart wg-quick@wg0
```

Sadly this is obviously not done automatically on RPi (on x86 it is). Something we could ask either the WireGuard or Raspberry Pi devs about, to create a APT hook on one side or use the same that x86 kernel packages use, if possible.

Joulinar, 2019-05-19:
cool, that was quick fix. Thanks so much.
Now working again. 

```
root@DietPi:~# dpkg-reconfigure wireguard-dkms
Loading new wireguard-0.0.20190406 DKMS files...
It is likely that 4.19.42-v7+ belongs to a chroot's host
Building for 4.19.42+ and 4.19.42-v7+
Building initial module for 4.19.42+
Done.

wireguard:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.42+/kernel/net/

depmod....

DKMS: install completed.
Building initial module for 4.19.42-v7+
Done.

wireguard:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.42-v7+/kernel/net/

depmod....

DKMS: install completed.
```

So I guess someting I would need to do every time kernel is updated. correct?

MichaIng, 2019-05-19:
@Joulinar 
Great!

> So I guess someting I would need to do every time kernel is updated. correct?

Exactly. Luckily a kernel upgrade is nothing you could ever miss due to it's verbose output 😄. And the last one was a large step, v4.14 => v4.19, which includes maaany enhancement in the background, performance, power and security-wise 😃.

closed 2019-05-19 by MichaIng