MichaIng/DietPi #2826: dietpi-config - show what options i have to choose from

Sopor, 2019-05-19:
I was running the `dietpi-config` and i browsed around to see what i can do in the settings and most of them doesn't do anything but it seems that when i press the `RTC mode` it will not bring up any options. Instead it starts to change Emulated to Hardware without even asking me.

The `RTC mode` should have its own menu so i know what to choose from instead of changing it when i press on it.

[image]

MichaIng, 2019-05-19:
@Sopor 
Many thanks for your request.

Yeah most options are implemented as simple toggle switches, so when you select them, no additional menu will prompt, but the state simply changes. But since in case of RTC mode at least runs an APT install/purge is done (takes a little while), a separate menu makes sense. Also we can add some info about the RTC, what it's for and why you need the `fake-hwclock` emulation for most SBCs.

However purging/installing the package back and forth does not hurt.

> most of them doesn't do anything

What you mean by this? All of them should toggle some state or open a sub menu.

MichaIng, 2019-06-14:
Done: https://github.com/MichaIng/DietPi/commit/392683ccb9b0da3fe80b344f12a586207e53f1a7
Changelog: https://github.com/MichaIng/DietPi/commit/5e3902ebbcfcc18921fd3ac4304fe630bcfc2ea4

closed 2019-07-03 by MichaIng