MichaIng/DietPi #2827: 6.24.1.: Lighttpd still not starting

zerbes, 2019-05-19:
- DietPi version: 6.24.1
- SBC device: RPI1

- Software title : lighttpd
- Was the software title installed freshly or updated/migrated: updated from 6.19.
- Can this issue be replicated on a fresh installation of DietPi?: not tried yet

#### Steps to reproduce
Tried to upgrade from 6.19 (no later backup unfortunately after bricked my RPI as result of upgdate from 6.23 this week).

#### Actual behaviour
Update runs smooth to 6.24.1 but lighttpd is still not starting. Reboot did not help

#### Extra details
`[ INFO ] DietPi-Update | Patching 6.23.1 to 6.24
[  OK  ] DietPi-Update | Patching to v6.24.1 completed

[  OK  ] DietPi-Survey | Connection test: ssh.dietpi.com
[ SUB2 ] DietPi-Update > Completed
[ INFO ] DietPi-Update | Current version : v6.24.1
[ INFO ] DietPi-Update | Latest version  : v6.24.1
[  OK  ] DietPi-Update | Syncing new DietPi scripts to disk
[ SUB3 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : php7.3-fpm
[FAILED] DietPi-Services | restart : lighttpd
[  OK  ] DietPi-Services | restart : cron
cron: can't lock /var/run/crond.pid, otherpid may be 32726: Resource temporarily unavailable

dnsmasq: junk found in command line
pihole-FTL: invalid option -- '0'
Try 'pihole-FTL --help' for more information
2019-05-19 10:57:29: (server.c.1073) No configuration available. Try using -f option.
/DietPi/dietpi/.dietpi-process_tool: line 6: php-fpm: command not found
[ SUB4 ] DietPi-Process_tool > Apply
[FAILED] DietPi-Process_tool | Webserver Lighttpd (344) : Nice      0
[FAILED] DietPi-Process_tool | Webserver Lighttpd (344) : Scheduler SCHED_OTHER 0
[FAILED] DietPi-Process_tool | An issue has occurred
Beendet`

Unfortunately i have to restore my backup based on 6.19 to have my pi-hole up and running, so so I am afraid I won't be able to assist any further on solving this. With some spare time i can try to update to 6.24.1 from current 6.22-ISO to see if it is only an issue coming from updating a very old version (6.19).

Regards

zerbes

MichaIng, 2019-05-19:
@zerbes 
There is junk in your process tool config, responsible for:
```
cron: can't lock /var/run/crond.pid, otherpid may be 32726: Resource temporarily unavailable

dnsmasq: junk found in command line
pihole-FTL: invalid option -- '0'
Try 'pihole-FTL --help' for more information
2019-05-19 10:57:29: (server.c.1073) No configuration available. Try using -f option.
/DietPi/dietpi/.dietpi-process_tool: line 6: php-fpm: command not found
```

We did some change but AFAIK the update patch should remove this accordingly. However when you see such again, do `rm /{DietPi,boot}/dietpi/.dietpi-process_tool`. And start `dietpi-process_tool` to re-apply custom nice/scheduler/priority if required.
_____________
There are two things to verify about Lighttpd:
- It got upgraded to the Buster version, which should have been part of the regular APT upgrade at the beginning of the dietpi-update run.
- It's config file got adjusted, which should be part of the incremental patch: `sed -i 's/create-mime\.assign\.pl/create-mime.conf.pl/g' /etc/lighttpd/lighttpd.conf`

MichaIng, 2019-06-08:
I mark this issue as closed now, feel free to reopen if required.

closed 2019-06-08 by MichaIng

ConsultorML, 2020-10-12:
I had a similar problem with Lighttpd, crashing to boot and not working. I solved it with this code. 
`chown -R www-data:www-data /var/log/lighttpd`

Maybe it can help. Greetings

Source: https://askubuntu.com/questions/1072840/failed-to-start-lighttpd-daemon-on-boot