MichaIng/DietPi #2828: Dietpi inaccessible and dietpi commands no longer working after failed update

1985kasper, 2019-05-19:
### Creating a bug report/issue

#### Required Information
- DietPi version | /DietPi is empty
- Distro version | 9.9
- Kernel version | 4.14.98
- SBC device | RPi2
- Power supply used | unknown generic
- SDcard used | scandisk

#### Additional Information (if applicable)
- Software title | (Dietpi
- Was the software title installed freshly or updated/migrated? Updated
- Can this issue be replicated on a fresh installation of DietPi? not sure

#### Steps to reproduce

1. Run the dietpi update command
2. The update fails

#### Expected behaviour

- The update is successfull and machine stays accessible

#### Actual behaviour

- Update fails and machine is no longer accessible through SSH. Password is prompting but no longer able to login. The Dietpi directories are empty and the MOTD about Dietpi commands no longer visible when logging in non-remote

#### Extra details

- Wireguard installed

MichaIng, 2019-05-19:
@1985kasper 
Many thanks for your report.

Please paste:
```
systemctl status dietpi-ramdisk
cat /var/tmp/dietpi/logs/dietpi-ramdisk.log
```

1985kasper, 2019-05-19:
Had to nuke the SD card and start over. It runs pi-hole so little inconvenient to have no DNS in the house. 
So not able to do more testing/troubleshooting

MichaIng, 2019-05-19:
@1985kasper 
Okay, just a side note:
- When facing the kernel upgrade to v4.19 (via apt upgrade), remember to re-compile the WireGuard module against it, otherwise it will fail: `dpkg-reconfigure wireguard-dkms`

closed 2019-05-19 by MichaIng