MichaIng/DietPi #2829: FQDN/hostname and NIS domainnam : mismatch between DietPi-banner output and hostname command

maartenlangeveld, 2019-05-19:
DietPi-Banner states
(WAN IP deliberately changed: not my real IP)
 ```
─────────────────────────────────────────────────────
 DietPi v6.24.1 : 14:45 - Sun 19/05/19
 ─────────────────────────────────────────────────────
 - Device model : RPi Zero W (armv6l)
 - Uptime : up 1 day, 11 hours, 56 minutes
 - CPU temp : 34'C : 93'F (Cool runnings)
 - FQDN/hostname : DietPi
 - NIS domainname : hostname: Local domain name not set
 - LAN IP : 192.168.168.10 (wlan0)
 - WAN IP : 145.133.001.002
 - Freespace (RootFS) : 7.5G
 - Freespace (music folder) : 236G
 ─────────────────────────────────────────────────────
```
while hostename gives:

```
root@DietPi:~# hostname --short
DietPi
root@DietPi:~# hostname --domain
local
root@DietPi:~# hostname --domain
local
root@DietPi:~# hostname --fqdn
DietPi.local
```
So FQDN/hostname does not match (domain is lacking in DietPi-banner output) as is NIS domainname.

Curious why there's a mismatch.

MichaIng, 2019-05-19:
@maartenlangeveld 
Thanks for your report.

Why I wonder is why `hostname --fqdn` shows `DietPi.local` while the banner shows `DietPi` only. It uses `hostname -f` which is only an alias for `hostname --fqdn`, show they should show both the long version.

For "NIS domainname" it uses `hostname -y` (== `--yp`/`--nis`), which is not the same as `hostname --domain/-d`. Read about it here: https://en.wikipedia.org/wiki/Network_Information_Service

The reason is that FQDN already contains the full name, including the domain. It would have been unnecessary overhead to separate those info into two fields. You always require the full FQDN anyway to access.

NIS/YP has not directly something to do with it, but needs separate implementation, which is why it says in your case: "Local domain name not set"
Actually NIS/YP is an outdated and insecure nameserver + user authentication solution. Nowadays the purpose is mostly solved differently, e.g. via LDAP.

maartenlangeveld, 2019-05-19:
@MichaIng 
Thanks, not an important issue though. I now understand there are different domain names (DNS vs NIS). 
Will check another time why there's a difference between the outputs.

maartenlangeveld, 2019-05-19:
@MichaIng 

Cannot reproduce. Now both outputs are consistent (no domain name).
It surprises me since I have set domain :  local in DCHP server, so I would expect my RPI gets 'local' as domain name assigned. Apparently not.

MichaIng, 2019-05-19:
@maartenlangeveld 
Not sure if the `hostname` respects any externally assigned info at all. Usually `/etc/hostname` should simply contain the full FQDN. The hostname command handles this correctly, producing long/short/domain outputs.

maartenlangeveld, 2019-05-19:
@MichaIng 
Ok thanks, learning a lot. I shall close since not an issue at all. Thanks for your patience.

closed 2019-05-19 by maartenlangeveld