MichaIng/DietPi #2830: Update from 6.22 to 6.24 crash

Gill-Bates, 2019-05-19:
Hi there,

during the first attempt to upgrade Dietpi to 6.2.4 the Update procedure crashed:

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6

[  OK  ] DietPi-Update | Update available
[ INFO ] DietPi-Update | Current version : v6.22.3
[ INFO ] DietPi-Update | Latest version  : v6.24.1
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=108905 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.0-fpm
[  OK  ] DietPi-Update | wget https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/pre-patch_file -O pre-patch_file

 DietPi-Pre-patch
─────────────────────────────────────────────────────
 Mode: Applying pre-patches

[  OK  ] Successfully applied pre-patches

[ INFO ] DietPi-Update | APT update, please wait...
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:3 https://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Get:4 https://archive.raspberrypi.org/debian stretch/main armhf Packages [221 kB]
Get:5 https://archive.raspberrypi.org/debian stretch/ui armhf Packages [45.0 kB]
Get:6 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [95.5 kB]
Fetched 12.1 MB in 11s (1,083 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | G_AGUP
[ INFO ] DietPi-Update | APT upgrade, please wait...
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20190517-1_armhf.deb  404  Not Found
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-bin_1.20190517-1_armhf.deb  404  Not Found
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi0_1.20190517-1_armhf.deb  404  Not Found
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20190517-1_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
[FAILED] DietPi-Update | G_AGUG
[ INFO ] DietPi-Update | APT upgrade, please wait...

After the second try, the upgrade process works fine. Any idea what went wrong?

MichaIng, 2019-05-19:
@Gill-Bates 
Yeah I faced the same yesterday. Somehow the accessibility of the new new kernel package is unstable.

Also when checking the packages file yesterday, sometimes the new kernel was listed, sometimes the old. Seems to be either a cache issue or, if in the back archive.raspberrypi.org points to different servers, a sync issue.

But this should resolve itself, actually my guess was that today it should work reliable (due to usual APT repo sync and cache times and such) 🤔.

closed 2019-05-19 by MichaIng

officiallor, 2019-08-29:
I'm facing the same problem on my 6.24.1 when I try to update to 6.25.3.

MichaIng, 2019-08-29:
@officiallor 
Please try:
```
apt clean
apt update
apt upgrade
```
- The last command includes what `dietpi-update` fails on, so it this succeeds this time, `dietpi-update` will succeed as well.
- This contains a kernel+bootloader+firmware update, so it might take a little while on RPi Zero 😉.

benbrookshire, 2019-11-03:
Same issue for me. A retry worked. 

```
root@DietPi:~# dietpi-update
[  OK  ] DietPi-Update | Root access verified.
[  OK  ] DietPi-Update | RootFS R/W access verified.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v6.25.3
[ INFO ] DietPi-Update | Latest version  : v6.26.3
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=3973 MiB | required=100 MiB

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Applying DietPi update

[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : smbd
[  OK  ] DietPi-Services | stop : nmbd
[  OK  ] DietPi-Services | stop : avahi-daemon
[ .... ] DietPi-Update | wget https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/pre-patch_file -[  OK  ] DietPi-Update | wget https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/pre-patch_file -O pre-patch_file
[ INFO ] Pre-patch 13 | Disabling dpkg status file translations by default, which have hardly any use, but raise APT update times + disk I/O 
[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://packages.sury.org/php stretch InRelease
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...
[  OK  ] DietPi-Update | G_AGUP
[ INFO ] DietPi-Update | APT upgrade, please wait...
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20190819~stretch-1_armhf.deb  GnuTLS recv error (-110): The TLS connection was non-properly terminated.
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
[FAILED] DietPi-Update | G_AGUG
[ INFO ] DietPi-Update | APT upgrade, please wait...
Extracting templates from packages: 100%

```