MichaIng/DietPi #2834: Dietpi-update error

gael1980, 2019-05-20:
Bonjour,
Impossible pour moi de réaliser la mise à jour de la version 6.24.0 vers la version 6.24.1

J'obtiens toujours ce message d'erreur : [FAILED] DietPi-Update | Download failed, unable to run update. Please try running dietpi-update again.

root@DietPi:~# dietpi-update
[  OK  ] DietPi-Update | Root access verified.
[  OK  ] DietPi-Update | RootFS R/W access verified.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Mich                                                                                        aIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/                                                                                        MichaIng/DietPi/master/dietpi/server_version-6

[  OK  ] DietPi-Update | Update available
[ INFO ] DietPi-Update | Current version : v6.24.0
[ INFO ] DietPi-Update | Latest version  : v6.24.1
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/DD/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=27052 MiB | requir                                                                                        ed=100 MiB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : webmin
[  OK  ] DietPi-Services | stop : grafana-server
[  OK  ] DietPi-Services | stop : influxdb
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : redis-server
[  OK  ] DietPi-Services | stop : nfs-kernel-server
[  OK  ] DietPi-Services | stop : smbd
[  OK  ] DietPi-Services | stop : nmbd
[  OK  ] DietPi-Update | wget https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/pre-patch_file -O pre-patch_file

 DietPi-Pre-patch
─────────────────────────────────────────────────────
 Mode: Applying pre-patches

[  OK  ] Successfully applied pre-patches

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Hit:3 https://repos.influxdata.com/debian stretch InRelease
Hit:4 https://packages.sury.org/php stretch InRelease
Hit:5 https://packages.grafana.com/oss/deb stable InRelease
Reading package lists...
[  OK  ] DietPi-Update | G_AGUP
[ INFO ] DietPi-Update | APT upgrade, please wait...
[  OK  ] DietPi-Update | G_AGUG
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   120    0   120    0     0    283      0 --:--:-- --:--:-- --:--:--   283
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (56) Received HTTP code 503 from proxy after CONNECT
[FAILED] DietPi-Update | Download failed, unable to run update. Please try running dietpi-update again.
root@DietPi:~#

MichaIng, 2019-05-20:
@gael1980 
Many thanks for your report.

Do you use a proxy server?

Please try to run the download manually:
```
curl -L https://github.com/MichaIng/DietPi/archive/master.zip > /tmp/update.zip
```

Perhaps it was a temporary issue. If the above succeed, then the update should as well pass this time, so rerun it:
```
dietpi-update 1
```

gael1980, 2019-05-21:
Thanks for your help.
the problem did not come from the proxy
I solved the problem by adding "codeload.github.com" in the whitelist of pihole

MichaIng, 2019-05-21:
@gael1980 
Okay something we need to forward to Pi-hole. Must be one of their ad lists, so redirected again to the ad lists maintainers. Strange to block github...

MichaIng, 2019-06-08:
Possibly related: https://github.com/MichaIng/DietPi/issues/2772

MichaIng, 2019-06-08:
Hmm strange, on a fresh Pi-hole install non of the GitHub URLs is blocked, nor codeload.github.com explicitly. So this must be due some custom blocklist or entry. I mark the issue as closed now.

closed 2019-06-08 by MichaIng