MichaIng/DietPi #2835: MPD update error

echable9999, 2019-05-20:
Hi
Just bought my first RPi yesterday, bought 2nd one today .... DietPi is fantastic! Thank you for all the great work!

I love the DOS-like startup menu with optimised package installation - for the first time ever I have a fully working Subsonic installation - on RPi #1. 

On RPi #2, however, I am trying to install MPD using the DietPi menu. It gives an error during install that says "exit code 8 [..] 523 Origin Unreachable Remote File Does Not Exist - Broken Link!"

It gives "retry" or DietPi-config options. I tried looking at DietPi-config, anyone with any tips how I reconfigure the source where it looks for and can't find MPD ? Or another solution to this problem ? Thank you very much.

MichaIng, 2019-05-20:
@echable9999 
Many thanks for your report.

I just checked our download links and they work fine here: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L3566-L3604

Could you paste some more details, at best re-run the installation, when you run into the error, hit exit and paste the blue lines that are printed on terminal. Easiest btw. to do this via SSH/[PuTTY](https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html) if this was not the case, so you can copy&paste directly from your desktop.

The info I need is the RPi model (since you just bought them, I guess RPi3?) and the exact download command that failed. If it was indeed the MPD binary download, then the URL should have been: https://dietpi.com/downloads/binaries/all/mpd_0.20.23-1_armv7l.deb which downloads fine here. You can test manually on the RPi:
wget https://dietpi.com/downloads/binaries/all/mpd_0.20.23-1_armv7l.deb -O /tmp/mpd.deb

echable9999, 2019-05-20:
Hi
Turns out it was apparently just temporary downtime on your site/that site
:)

Same thing happened now when I did an initial install of DietPi again, this
time "retry" worked again straight away, unlike earlier today.

Still stuck trying to get MPD to work. Need it to get data from a NAS, but
I already did a NFS mount for one of the Pis for Subsonic and that worked
with no problems. But, with MPD, I just get the "already bound/already in
use" error regarding localhost/port number. Is the DietPi MPD install
supposed to be fully functional now, or are there still known bugs ?  (I
saw some mentioned on Github but it's too technical for me).

I've forwarded the MPD port 6600 to the IP of the RPi, as I assume I
should, and in the mpd.conf file I've tried variations of default "any",
"127.0.0.1", and the local IP of the RPi with MPD on it. No luck so far.

DietPi is fantastic, however, if I'd known I would have got some Pis years
ago :)

Regards,
Eric

MichaIng, 2019-05-20:
@echable9999 
> Turns out it was apparently just temporary downtime on your site/that site

I just found the server being restarted, based on the uptime fitting very well with your post. Looks like it was a crash but not sure for the reason. I ran some `inotifywait` over night to have some image download numbers. Not sure if related, but actually it is the standard Linux kernel internal file access monitor available, with 0 CPU and marginal RAM usage 🤔.

However glad it works now, so I mark this issue as closed.

MichaIng, 2019-05-26:
@echable9999 
Are you sure you need to open your MPD daemon to the web? For `bind_to_address` then your external IP or domain should work.

If you get a bind issue, please check: `ls -l /proc/$(fuser 6600/tcp | mawk '{print $1}')/exe`
Perhaps there is another program using this port is MPD is already running? Perhaps a remaining instance that needs to be killed first.

MichaIng, 2019-06-25:
I mark this as closed. Feel free to reopen if issue persists.

closed 2019-06-25 by MichaIng