MichaIng/DietPi #2836: Update failure for 6.24.1

raysqa, 2019-05-20:
#### Details:
- Date           | Mon May 20 06:48:43 MDT 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi Zero W (armv6l) (index=1)
- Kernel version | #1200 Tue Feb 12 20:11:02 GMT 2019
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 127
- Software title |

#### Steps to reproduce:

1. enter command DietPi-Update
2. ...

#### Expected behaviour:

- Distro should update from v.6.22.3 to 6.24.1

#### Actual behaviour:

- Update failure - -get: error while loading shared libraries: libstdk++.so.6: cannot open shared
ect file: No such file or directory

Retry does not work

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
apt-get: error while loading shared libraries: libstdk++.so.6: cannot open shared object file: No such file or directory
apt-get: error while loading shared libraries: libstdk++.so.6: cannot open shared object file: No such file or directory
```

MichaIng, 2019-05-20:
@raysqa 
First I thought it is about ARMv6 + PHP, but actually `libstdk++.so.6` does not exist and should not it's `libstdc++.so.6` that is present on systems. Strange where this typo in APT came from. Also web search does not result in any single match (and to be failsafe, jep Raspbian repo does not have it: http://raspbian.raspberrypi.org/raspbian/dists/stretch/main/binary-armhf/Packages), so it's definitely a typo in your APT binary. Not sure if this comes from e.g. SDcard corruption.

Just to verify, run:
```
apt-get update
```

If that fails with the same error, let's see if the `apt` binary is affected as well:
```
apt update
```

To check for I/O errors:
```
dmesg
```
- Paste any red lines that you see in the output.

If those appear, and actually to be failsafe, so a fsck on reboot:
```
> /forcefsck
reboot
```
_____________
Since APT itself is affected it is a bid tricky to solve, when you sorted out or fixed any file system corruption.

We can link the existing correct library to where APT looks for it:
```
ln -s /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 /usr/lib/arm-linux-gnueabihf/libstdk++.so.6
```

Hopefully then `apt update` works through.

Then we need to reinstall the APT package:
```
apt install --reinstall apt
```

Hopefully that solves the typo, so we can remove the symlink again:
```
rm /usr/lib/arm-linux-gnueabihf/libstdk++.so.6
apt-get update
```

If this finally worked, then re-run the update:
```
dietpi-update 1
```

raysqa, 2019-05-21:
Thanks for getting back to me.  It turns out that my installation is pretty much corrupted.  I ended up reinstalling from a backup image and upgrading from that.  Having a corrupted installation is a pretty regular occurrence – some point after just running pihole, the image corrupts.  But at this point I have everything back up.  Thanks again.

MichaIng, 2019-05-21:
@raysqa 
Yeah, I also regularly face this on my Pi/SDcard.... However fsck fixes it still, currently... Earlier or later I need a new SDcard. Not sure how old/how much in use yours is, but you should consider this as well. The SDcard might contain bad blocks, so parts where when written to, will always result in corrupted data.

And as a general advice:
- Try to avoid writing to the SDcard as much as possible. Move swap file and dietpi userdata to external drives, use DietPi-RAMlog and such.
- We are also thinking about further methods to reduce SDcard writes, e.g. allowing to move all the APT cache onto external drive or RAM. The whole cache is 120M on my system, which is regularly re-created.

closed 2019-05-21 by MichaIng