MichaIng/DietPi #2837: 'dietpi-cleaner 1' in an endless loop (DietPi v6.24.1)

maartenlangeveld, 2019-05-20:
'dietpicleaner 1' command seems to cause an endless loop...

```
[root@DietPi:~# dietpi-cleaner 1
[  OK  ] DietPi-Cleaner | Root access verified.
[  OK  ] DietPi-Cleaner | RootFS R/W access verified.
[ SUB1 ] DietPi-Services > stop
[ INFO ] DietPi-Services | Including custom service: upmpdcli
[ INFO ] DietPi-Services | Including custom service: minimserver
[  OK  ] DietPi-Services | stop : minimserver
[  OK  ] DietPi-Services | stop : upmpdcli
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : mpd
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : nfs-kernel-server
[  OK  ] DietPi-Services | stop : smbd
[  OK  ] DietPi-Services | stop : nmbd
[  OK  ] DietPi-Services | stop : avahi-daemon

───────────────────────────────────────
DietPi-Cleaner
 - Man pages and docs
───────────────────────────────────────
E: No packages found
[ INFO ] DietPi-Cleaner | APT autoremove + purge, please wait...
[  OK  ] DietPi-Cleaner | G_AGA

───────────────────────────────────────
DietPi-Cleaner
 - Files
───────────────────────────────────────

Searching for filenames matching:
- *.tmp

Please wait...

No matching filenames were found.

───────────────────────────────────────
DietPi-Cleaner
 - Man pages and docs
───────────────────────────────────────
E: No packages found
[ INFO ] DietPi-Cleaner | APT autoremove + purge, please wait...
[  OK  ] DietPi-Cleaner | G_AGA

───────────────────────────────────────
DietPi-Cleaner
 - Files
───────────────────────────────────────

Searching for filenames matching:
- *.tmp

Please wait...

No matching filenames were found.

───────────────────────────────────────
DietPi-Cleaner
 - Man pages and docs
───────────────────────────────────────
E: No packages found
[ INFO ] DietPi-Cleaner | APT autoremove + purge, please wait...
[  OK  ] DietPi-Cleaner | G_AGA

───────────────────────────────────────
DietPi-Cleaner
 - Files
───────────────────────────────────────

Searching for filenames matching:
- *.tmp

Please wait...

No matching filenames were found.

───────────────────────────────────────
DietPi-Cleaner
 - Man pages and docs
───────────────────────────────────────
E: No packages found
[ INFO ] DietPi-Cleaner | APT autoremove + purge, please wait...
[  OK  ] DietPi-Cleaner | G_AGA

───────────────────────────────────────
DietPi-Cleaner
 - Files
───────────────────────────────────────

Searching for filenames matching:
- *.tmp

Please wait...

No matching filenames were found.

───────────────────────────────────────
DietPi-Cleaner
 - Man pages and docs
───────────────────────────────────────
E: No packages found
[ INFO ] DietPi-Cleaner | APT autoremove + purge, please wait...
[  OK  ] DietPi-Cleaner | G_AGA

> **and so on....**

```

Fourdee, 2019-05-20:
@maartenlangeveld 

Many thanks for the report, please can you paste the output of:
```
cat /DietPi/dietpi/.dietpi-cleaner_custom_files
```

Fourdee, 2019-05-20:
@maartenlangeveld 

Many thanks for the report :+1: 

Confirmed bug, looking into it now.

MichaIng, 2019-05-20:
Jep, `i` in the Run_Cleaner_$i function must be local. We generally need to take care to localize every i before loops in functions. I didn't notice since I had no files added, so I was not overridden.

MichaIng, 2019-05-20:
@maartenlangeveld @Fourdee 
That should be it: https://github.com/MichaIng/DietPi/commit/75c5daaec6ad497c10295b2b2b544428d9e97297
- Line 460 is the one required here.
- But to be failsafe in general, we should localise every loop index inside a function once. When some script gets reordered, this proactively prevents similar issues, which not necessarily show up on every test, depending on used options/variables and such. In this case indeed files array need to be filled (files added for removal) to run into the loop.

maartenlangeveld, 2019-05-20:
@MichaIng / @Fourdee,

Thanks. 

> sed -i "/for i in \${!aCustomFiles\[@\]}/i\\\t\\t\\tlocal i=''" /DietPi/dietpi/dietpi-cleaner
Solved the problem :-)

MichaIng, 2019-05-20:
@maartenlangeveld 
Great, thanks for testing 👍.

diveyez, 2019-05-21:
<https://github.com/MichaIng/DietPi/issues/2806> as seen in my images there, system was attempting to clean php and stuck in infinite loop as well. Dirty Dirty Packages. They should never have removed the crypto in php.

MichaIng, 2019-05-21:
@diveyez 
I can't see the images you mention, just the two about startup (with "A start job ..." hang and network interface error)?

> system was attempting to clean php

You mean the systemd service that regularly cleans PHP sessions? That is not related to this issue then (`dietpi-cleaner`).

> Dirty Dirty Packages

Yeah sounds like file corruption then, which might be also the reason for all the other issues you face/mention in the other topics.
`dmesg` can be used to check undervoltage, disk I/O errors and such, basically check for any red lines.
`> forcefsck` can be used to force a file system check on next reboot.

> They should never have removed the crypto in php.

What you mean? I don't get the relation and PHP of course is still cryptography capable via OpenSSL module. They removed the obsolete outdated `mcrypt` module which is a security vulnerability only when being used, if that is what you mean.

Fourdee, 2019-05-21:
Confirmed resolved with ```dev```, nice one @MichaIng :+1:

closed 2019-06-13 by MichaIng