MichaIng/DietPi #2838: Retrieving data from a broken Nextcloud install

minnux, 2019-05-20:
I apologize in advance as this is not strictly dietpi related, but when updating to the newest dietpi version on a rockpro64 there was either a power surge or some kind of corruption with my EMMC card and the install was halted or damaged. My data is on an external drive that I can easily plug into my linux laptop and I can see the dietpi userdata folder and within the nextcloud folder which appears to be the right size but I don't have the permission to access it now. All I want to do is retrieve the nextcloud data from the drive onto my ubuntu laptop and start from scratch. I still have my username and password for the dietpi and nextcloud installs but am just wondering what commands I would issue to copy over the data and be able to access it.

MichaIng, 2019-05-20:
@minnux 
So the RockPro64 does boot anymore at all? In most cases it should be possible to fix corruptions after a crash. Luckily Nextcloud data and database cannot be affected since services are stopped during updates.

Note that the `nextcloud_data` dir only does not contain any meta-data like comments, tags, previews or Nextcloud settings, but just the raw files.

However if you want to start from scratch, `sudo` should be your friend. I guess it's an Ubuntu desktop where you log in as custom user account. After mounting the drive, you should be able to open a terminal emulator and run: `sudo chown -R username:username /mnt/sd...`
- Replace "username" with your login account and "/mnt/sd..." should be the mount point of the external drive. Not sure what's default there, perhaps it's also in `/media/...`?
- Then your user has read+write permissions to access the data.

minnux, 2019-05-20:
Yes that's the info I was looking for thank you. Comments, tags and such are not essential to me but the data definitely is. Will username be the old Nextcloud user name, the dietpi user name, or my ubuntu laptop's user name? I didn't know you could take ownership of a directory that was already owned by www-data just by using chown.

MichaIng, 2019-05-21:
@minnux 
Note that all Nextcloud user + authentication things are totally independent from the UNIX user permissions (that you change with chown) of the underlying system.

When you setup a new Nextcloud instance you can again freely chose user names etc there. The only thing that is required for the data directory is `www-data` permissions, which is the webserver UNIX user. Nextcloud itself handles access permissions on top of that, as a second Nextcloud-only internal authentication layer.

Jep `chown` and `chmod` are the tools to change UNIX user permissions. Mostly it requires root permissions (`sudo`), if the related files/dirs are not already yours.
__________
Little note for setting up the new instance, if you do it on DietPi:
- Save both, the data directory (`/mnt/dietpi_userdata/nextcloud_data`) and the Nextcloud install dir (`/var/www/nextcloud`) and place them exactly to the same location on the new server.
- Then do:
```
rm /var/www/nextcloud/occ # Required to skip some maintenance mode handling
sed -i "s/'installed' => true,/'installed' => false,/" /var/www/nextcloud/config/config.php # Required to force a database rebuild
```
- Then install Nextcloud via `dietpi-software`. It will migrate the old instance and re-create the database.

closed 2019-07-12 by MichaIng