MichaIng/DietPi #2839: How to start OMPD

echable9999, 2019-05-21:
### Creating a bug report/issue

#### Required Information
- DietPi version 6/24/1/master
- Distro version 9.9
- SBC device RPi3
- Power supply used Original
- SDcard used SanDisk ultra uhs1

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
OMPD
- Was the software title installed freshly or updated/migrated?
Fresh install of OS and software
- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Steps to reproduce

I have installed OMPD through the DietPi "software optimised for your system" menu.

According to OMPD docs, OMPD should then be accessible at http://(ip address of DietPi device)/OMPD.

I have tried this, I have also tried http://(ip address of DietPi device):port/OMPD and http://(ip address of DietPi device):port.

I get "refused to connect" on everything except what the instructions say, http://(ip address of DietPi device)/OMPD, where the error is instead 404 Not Found. 

I have exactly the same problem with DietPi's Ampache install too.

Can anyone help with this please ? Thank you.

MichaIng, 2019-05-21:
@echable9999 
Many thanks for your report.

Please paste:
```
journalctl -u lighttpd
cat /var/log/lighttpd/error.log
journalctl -u php7.3-fpm
cat /var/log/php7.3-fpm.log
ls -l /var/www
```

And of course do you see the Lighttpd welcome page when only accessing your raw IP via browser without any port or path?

Fourdee, 2019-05-22:
@MichaIng 

Unable to replicate in testing.

@echable9999 

Also, please ensure to change to your IP of system. This is displayed on the system after you login.

eg, my IP is 192.168.0.100:
http://192.168.0.100/ompd

echable9999, 2019-05-23:
Tried the path in non-capital letters, still didn't work. 

Yes, I see the webserver "it works" page if I go directly to the base ip without port or path. 

Used the internal IP. 

Reinstalled on Odroid XU4, it works on that straight away, didn't work on Raspberry Pi 3B+. Good software.

closed 2019-05-23 by echable9999

MichaIng, 2019-05-23:
@echable9999 
So it works now on RPi3 as well? If webserver is up, then as above having a look at PHP status for debugging.

echable9999, 2019-05-23:
Haven't tried so many but OMPD is by far the best MPD client/manager I've tried (wish it allowed edition the notorious mpd config options through the interface maybe..) But it's running on a xu4 now. Haven't tried OMPD again on RPi3. Got one of those set aside now to act exclusively as a Subsonic server - it works smoother than I have ever seen it anywhere - just DietPi, mounted NAS drive, and Subsonic. I've fought with that software since I was 15 and whether on Windows, on NAS, wherever I've installed it there's been slow response times, always one or more of some kind of Java / ffmpeg / transcoding / permissions / database corruption issue. But I've loved it when it very, very occasionally has worked for ten minutes, no more no less. This install is now running solid and with zero errors or problems ever since I installed it, I will never touch that RPi3 again :) Don't know if DietPi or Subsonic or whoever has made this happen, maybe I've just been unlucky in the past with my hw/sw combinations. One strange thing I noticed, however: when I first installed DietPi Subsonic, DietPi-process_tool had Subsonic on the list of programs that you could tweak nice values for. Now, on the clean exclusive Subsonic install, Subsonic is not there in the process tool. But off-topic I guess.

MichaIng, 2019-05-24:
@echable9999 
Okay great.

Strange about process tool. I checked down to v6.13 and even then Subsonic was never inside the process tool list. Looks like we simply forget to add it.

This will be solved with v6.25 since we merged process tool and dietpi-services into one script. And dietpi-services always handled Subsonic as well. However if you want to add it:
```
sed -i '\|Squeezelite|a\aNAME\[$index\]='Subsonic';aPROCESS_NAME\[$index\]='/var/subsonic';((index++))' /DietPi/dietpi/dietpi-process_tool
```