MichaIng/DietPi #2840: APT | apt clean and disk I/O

MichaIng, 2019-05-21:
- Aside from allowing to move the whole APT cache onto external drive or into RAM, we should re-evaluate the need to run apt clean on regular basis in `G_AGUP`.
- It means that 60 - 100M are removed and rewritten every time:

The question is if, at least without pdiffs used (Raspbian), those `pkgcache.bin` and `srcpkgcache.bin` are re-written completely regardless.
_______
```
2019-05-21 16:32:39 root@micha:/mnt/sda/apt# iostat -m -p sda && apt-get clean && apt-get update && iostat -m -p sda
Linux 4.19.42-v7+ (micha.gnoedi.org)    21/05/19        _armv7l_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           0.60    0.07    0.30    0.16    0.00   98.88

Device             tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               1.93         0.01         0.02       1712       3185

Hit:1 https://archive.raspberrypi.org/debian stretch InRelease
Hit:2 https://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian buster InRelease
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Linux 4.19.42-v7+ (micha.gnoedi.org)    21/05/19        _armv7l_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           0.60    0.07    0.30    0.16    0.00   98.87

Device             tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               1.93         0.01         0.02       1712       3201
```
- 16M written, seems that async prevents most of the disk writes 👍.
```
2019-05-21 16:33:40 root@micha:/mnt/sda/apt# iostat -m -p sda && apt-get clean && sync && apt-get update && sync && iostat -m -p sda
Linux 4.19.42-v7+ (micha.gnoedi.org)    21/05/19        _armv7l_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           0.60    0.07    0.30    0.16    0.00   98.87

Device             tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               1.93         0.01         0.02       1712       3243

Hit:1 https://archive.raspberrypi.org/debian stretch InRelease
Hit:2 https://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian buster InRelease
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Linux 4.19.42-v7+ (micha.gnoedi.org)    21/05/19        _armv7l_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           0.60    0.07    0.30    0.16    0.00   98.87

Device             tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               1.93         0.01         0.02       1712       3302
```
- Forcing sync to disk => 59M as expected due to ~30M pkg+srcpkg cache files.
```
2019-05-21 16:34:04 root@micha:/mnt/sda/apt# iostat -m -p sda && apt-get update && sync && iostat -m -p sda
Linux 4.19.42-v7+ (micha.gnoedi.org)    21/05/19        _armv7l_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           0.60    0.07    0.30    0.16    0.00   98.87

Device             tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               1.93         0.01         0.02       1712       3302

Hit:1 https://archive.raspberrypi.org/debian stretch InRelease
Hit:2 https://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian buster InRelease
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Linux 4.19.42-v7+ (micha.gnoedi.org)    21/05/19        _armv7l_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           0.60    0.07    0.30    0.16    0.00   98.87

Device             tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               1.94         0.01         0.02       1712       3360
```
- Okay without `apt-get clean` the result is exactly the same. Seems every APT update rewrites the while cache files on Raspbian at least.
```
2019-05-21 16:35:17 root@micha:/mnt/sda/apt# iostat -m -p sda && apt-get update && iostat -m -p sda
Linux 4.19.42-v7+ (micha.gnoedi.org)    21/05/19        _armv7l_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           0.60    0.07    0.30    0.16    0.00   98.87

Device             tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               1.93         0.01         0.02       1712       3362

Hit:1 https://archive.raspberrypi.org/debian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Hit:3 https://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian buster InRelease
Reading package lists... Done
Linux 4.19.42-v7+ (micha.gnoedi.org)    21/05/19        _armv7l_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           0.61    0.07    0.30    0.16    0.00   98.87

Device             tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               1.94         0.01         0.02       1712       3376
```
- For completeness: Without forcing sync, apt-get clean as well has no effect on actual writes, or lets say nearly no.

MichaIng, 2019-05-21:
Okay needs testing on non-RPi (Debian repo with pdiff support) systems.

Fourdee, 2019-05-22:
@MichaIng 

It was mostly in there to prevent any APT-cache corruption (4.5v< PSU users :wink: ) which can prevent successful ```apt-get update```, always ensuring a fresh cache is generated each update.

But yep, can be removed I believe. 

Although ,maybe we should clear the cache only during removal of ```/etc/apt/sources.list.d/program``` in ```dietpi-software uninstall``` to ensure its completed removed from cache?

MichaIng, 2019-05-22:
@Fourdee 
I anyway thought it would be a larger issue, but as you can see:
- In both cases, on RPi, the whole file is (re)written once => same I/O.
- Disk async prevents most MB written.

But on non-Raspbian systems the repositories allow pdiffs, small chunks of newly added/changes packages to be downloaded and added to the list. Note sure currently how these are handled file wise on the disk, but possibly it is not required there to rewrite the whole ~60M cache file.

MichaIng, 2019-05-23:
### Test on VM with pdiff support
```
root@VM-Stretch:/tmp# iostat -m -p sda && apt-get clean && apt-get update && iostat -m -p sda
Linux 4.9.0-8-amd64 (VM-Stretch)        23/05/19        _x86_64_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           1.54    0.00    0.20    0.04    0.00   98.23

Device:            tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               4.78         0.03         0.13        318       1414
sda1              3.86         0.03         0.13        315       1414

Hit:1 https://download.mono-project.com/repo/debian stretch InRelease
Ign:2 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian stretch Release
Reading package lists... Done
Linux 4.9.0-8-amd64 (VM-Stretch)        23/05/19        _x86_64_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           1.55    0.00    0.20    0.04    0.00   98.22

Device:            tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               4.78         0.03         0.13        318       1414
sda1              3.86         0.03         0.13        315       1414
```
- Ah btw. when using `iostat -m -p sda` a second time, the ~50M are written, too. The above is just called too quickly after APT.
```
root@VM-Stretch:/tmp# iostat -m -p sda && apt-get clean && sync && apt-get update && sync && iostat -m -p sda
Linux 4.9.0-8-amd64 (VM-Stretch)        23/05/19        _x86_64_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           1.54    0.00    0.20    0.04    0.00   98.22

Device:            tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               4.77         0.03         0.13        318       1462
sda1              3.85         0.03         0.13        315       1462

Hit:1 https://download.mono-project.com/repo/debian stretch InRelease
Ign:2 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian stretch Release
Reading package lists... Done
Linux 4.9.0-8-amd64 (VM-Stretch)        23/05/19        _x86_64_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           1.55    0.00    0.20    0.04    0.00   98.21

Device:            tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               4.77         0.03         0.14        318       1510
sda1              3.85         0.03         0.14        315       1510
```
```
root@VM-Stretch:/tmp# iostat -m -p sda && apt-get update && sync && iostat -m -p sda
Linux 4.9.0-8-amd64 (VM-Stretch)        23/05/19        _x86_64_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           1.54    0.00    0.20    0.04    0.00   98.23

Device:            tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               4.71         0.03         0.14        318       1558
sda1              3.80         0.03         0.14        315       1558

Hit:1 https://download.mono-project.com/repo/debian stretch InRelease
Ign:2 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian stretch Release
Reading package lists... Done
Linux 4.9.0-8-amd64 (VM-Stretch)        23/05/19        _x86_64_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           1.54    0.00    0.20    0.04    0.00   98.22

Device:            tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               4.71         0.03         0.15        318       1606
sda1              3.81         0.03         0.15        315       1606
```
- Okay. again APT clean does not make ANY difference on disk writes. The two large cache files are simple recreated every time. pdiffs seem to only impact the download size, but not the cache files:
```
root@VM-Stretch:/tmp# l /var/cache/apt/
total 48932
drwxr-xr-x 3 root root    28672 May 23 03:06 archives
-rw-r--r-- 1 root root 25044480 May 23 03:14 pkgcache.bin
-rw-r--r-- 1 root root 25023911 May 23 03:14 srcpkgcache.bin
```

MichaIng, 2019-05-23:
Different approach:
```
root@VM-Stretch:/var/cache/apt# cat /etc/apt/apt.conf.d/99-dietpi-nocache
Dir::Cache::pkgcache "";
Dir::Cache::srcpkgcache "";
```
```
root@VM-Stretch:/var/cache/apt# iostat -m -p sda && apt-get update && sync && iostat -m -p sda
Linux 4.9.0-8-amd64 (VM-Stretch)        23/05/19        _x86_64_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           1.51    0.00    0.21    0.04    0.00   98.24

Device:            tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               5.13         0.03         0.16        347       1860
sda1              3.99         0.03         0.16        341       1860

Hit:1 https://download.mono-project.com/repo/debian stretch InRelease
Ign:2 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian stretch Release
Reading package lists... Done
Linux 4.9.0-8-amd64 (VM-Stretch)        23/05/19        _x86_64_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           1.52    0.00    0.21    0.04    0.00   98.23

Device:            tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               5.13         0.03         0.16        348       1860
sda1              3.99         0.03         0.16        341       1860

root@VM-Stretch:/var/cache/apt# iostat -m -p sda && apt-get clean && sync && apt-get update && sync && iostat -m -p sda
Linux 4.9.0-8-amd64 (VM-Stretch)        23/05/19        _x86_64_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           1.52    0.00    0.21    0.04    0.00   98.23

Device:            tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               5.13         0.03         0.16        348       1860
sda1              3.99         0.03         0.16        341       1860

Hit:1 https://download.mono-project.com/repo/debian stretch InRelease
Ign:2 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian stretch Release
Reading package lists... Done
Linux 4.9.0-8-amd64 (VM-Stretch)        23/05/19        _x86_64_        (4 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
           1.53    0.00    0.21    0.04    0.00   98.22

Device:            tps    MB_read/s    MB_wrtn/s    MB_read    MB_wrtn
sda               5.13         0.03         0.16        348       1860
sda1              3.99         0.03         0.16        341       1860
```
- 0M written!

Interesting, with this, even after running APT clean, package search and install works without any APT update required! It looks like APT clean really only cleans the cache. If it is active, then APT upgrade/install expects the cache files and otherwise fails, asking to run APT update to recreate it. But actually the really relevant files stay in place, untouched by APT clean:
```
root@VM-Stretch:/var/lib/apt/lists# l
total 42048
-rw-r--r-- 1 root root   231764 Apr 27 11:04 deb.debian.org_debian_dists_stretch_contrib_binary-amd64_Packages
-rw-r--r-- 1 root root 38896249 Apr 27 11:04 deb.debian.org_debian_dists_stretch_main_binary-amd64_Packages
-rw-r--r-- 1 root root   381796 Apr 27 11:04 deb.debian.org_debian_dists_stretch_non-free_binary-amd64_Packages
-rw-r--r-- 1 root root   117943 Apr 27 11:30 deb.debian.org_debian_dists_stretch_Release
-rw-r--r-- 1 root root     2434 Apr 27 11:35 deb.debian.org_debian_dists_stretch_Release.gpg
-rw-r--r-- 1 root root    91028 May 22 22:30 deb.debian.org_debian_dists_stretch-updates_InRelease
-rw-r--r-- 1 root root   168555 May 18 16:09 deb.debian.org_debian_dists_stretch-updates_main_binary-amd64_Packages
-rw-r--r-- 1 root root    10084 May 18 16:09 deb.debian.org_debian_dists_stretch-updates_main_binary-amd64_Packages.diff_Index
-rw-r--r-- 1 root root     4840 Mar 31 21:35 deb.debian.org_debian-security_dists_stretch_updates_contrib_binary-amd64_Packages
-rw-r--r-- 1 root root    94345 May 22 23:37 deb.debian.org_debian-security_dists_stretch_updates_InRelease
-rw-r--r-- 1 root root  2671692 May 22 23:37 deb.debian.org_debian-security_dists_stretch_updates_main_binary-amd64_Packages
-rw-r--r-- 1 root root     3423 May 15 11:22 deb.debian.org_debian-security_dists_stretch_updates_non-free_binary-amd64_Packages
-rw-r--r-- 1 root root     5878 Apr 11 18:14 download.mono-project.com_repo_debian_dists_stretch_InRelease
-rw-r--r-- 1 root root   329118 Apr 11 18:14 download.mono-project.com_repo_debian_dists_stretch_main_binary-amd64_Packages
-rw-r----- 1 root root        0 Mar 24 20:12 lock
drwx------ 2 _apt root    12288 May 23 03:29 partial
```
- These are only rewritten when there are actually changes and wich cache disabled these are sufficient for all APT tasks I could find and without any negative performance impact I could recognise.

Now I am wondering what these pkgcache.bin and srcpkgcache.bin files are actually for O_O... They are rewritten on every APT update but actually the package lists are already present at another directory and sufficient. With cache active, on an actually updated list on the server, the disk I/O is even doubled, once the files in /var/lib/apt and then once again in /var/cache/apt...

I will now disable the APT cache on all my devices and see if I face any difference.

MichaIng, 2019-05-23:
To completely disable the ATP cache:
```
cat << _EOF_ > /etc/apt/apt.conf.d/99-dietpi-no_cache
Dir::Cache "";
Dir::Cache::archives "";
Dir::Cache::pkgcache "";
Dir::Cache::srcpkgcache "";
_EOF_
```
- If we anyway do apt-get clean, the slightly performance impact for the `Reading package lists...` tasks is exactly the same, as exactly these cached files are removed.
- And for my impression it should be absolute standard to disable this cache on every SDcard system, if not on every system. >50M disk writes on every APT update is simply too much for such a tiny performance improvements, and the cache even causes issues by times, which is the reason why we run apt-get clean...
- So solve all of this at once at disable the cache. Then apt-get clean has no use anymore and can be left out.
- Even `apt-cache showpkg ...` works flawlessly on any installed and non-installed package.

MichaIng, 2019-05-23:
### Performance tests:
#### No cache
```
root@VM-Stretch:/var/lib/apt/lists# time for i in {1..10}; do apt-get update &> /dev/null; done

real    0m42.188s
user    0m41.656s
sys     0m2.896s
```
#### Enabled cache
```
root@VM-Stretch:/var/lib/apt/lists# time for i in {1..10}; do apt-get update &> /dev/null; done

real    0m43.962s
user    0m42.036s
sys     0m4.220s
```
- 🤣 What the hell is the use of this.... All package list information with everything relevant is stored in /var/lib/apt, all APT commands successfully get their information from there, for what this need to be again wrapped up and saved to anther file...
________
Just to be sure....
#### Cache still enabled
```
root@VM-Stretch:/var/lib/apt/lists# time for i in {1..10}; do apt-get update &> /dev/null; done

real    0m44.786s
user    0m42.308s
sys     0m4.432s
```
#### Cache disabled
```
root@VM-Stretch:/var/lib/apt/lists# time for i in {1..10}; do apt-get update &> /dev/null; done

real    0m42.149s
user    0m41.428s
sys     0m2.864s
```

Okay let's get rid of this!

Fourdee, 2019-05-23:
Notes:
```
cat << _EOF_ > /etc/apt/apt.conf.d/99-dietpi-ramcache
Dir::Cache::pkgcache "/tmp/apt_cache_pkg/";
Dir::Cache::srcpkgcache "/tmp/apt_cache_src/";
_EOF_
mkdir -p /tmp/apt_cache_pkg /tmp/apt_cache_src

apt-get clean; sync; sleep 1; time apt-get update
```

...
```
root@DietPi:~# apt-get clean; sync; sleep 1; time apt-get update
Hit:1 http://fuzon.co.uk/meveric all InRelease
Hit:2 http://fuzon.co.uk/meveric stretch InRelease 
Ign:3 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:6 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease
Hit:7 https://cdn-aws.deb.debian.org/debian stretch Release
Reading package lists... Error!
E: Problem renaming the file /tmp/apt_cache_src/.MDyrGZ to /tmp/apt_cache_src/ - rename (20: Not a directory)
E: Problem renaming the file /tmp/apt_cache_pkg/.49rdpb to /tmp/apt_cache_pkg/ - rename (20: Not a directory)
W: You may want to run apt-get update to correct these problems
E: The package cache file is corrupted

real    0m4.406s
user    0m3.100s
sys     0m0.852s
```

MichaIng, 2019-05-23:
@Fourdee 
Those two values need to be files, use instead:
```
Dir::Cache "/tmp/apt_cache_pkg"
Dir::Cache::pkgcache "pkgcache.bin";
Dir::Cache::srcpkgcache "srcpkgcache.bin";
```

PeterLacknase, 2019-05-24:
From [https://wiki.ubuntu.com/ReducingDiskFootprint](https://wiki.ubuntu.com/ReducingDiskFootprint):

> ### Disable apt caches
> Apt stores two caches in `/var/cache/apt/`: `srcpkgcache.bin` is rather useless these days, and `pkgcache.bin` is only needed for faster lookups with `apt-cache` (software-center has its own cache). Removing those two buys 26 MB, for the price of apt-cache taking an extra two seconds for each lookup.
> 
> An image build needs to `rm /var/cache/apt/*.bin`. However, the next `apt-get update` would bring it back, thus the caches need to be disabled in the configuration. Create a file `/etc/apt/apt.conf.d/02nocache` with the following contents:
> 
>  `Dir::Cache {`
>    `srcpkgcache "";`
>    `pkgcache "";`
>  `}`
> However, please note that if you need to use synaptic, you need to keep pkgcache; see bug [596898](https://bugs.launchpad.net/bugs/596898).

MichaIng, 2019-05-24:
@PeterLacknase 
Thanks for pasting. Okay so it is indeed mostly useless. How much users call `apt-cache` at all. Regardless will do some speed tests. Actually the info should be included in the APT list files, so offline info should still be possible.

MichaIng, 2019-06-10:
### Speed tests
#### No cache
```
root@Marlin-Laptop:/tmp# cat /etc/apt/apt.conf.d/99-dietpi-no_cache
Dir::Cache "/tmp/apt";
Dir::Cache::archives "/tmp/apt";
Dir::Cache::pkgcache "";
Dir::Cache::srcpkgcache "";
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache show coturn &> /dev/null; done

real    0m6.294s
user    0m5.960s
sys     0m0.456s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache show coturn &> /dev/null; done

real    0m6.279s
user    0m5.952s
sys     0m0.448s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache search coturn &> /dev/null; done

real    0m7.804s
user    0m7.412s
sys     0m0.516s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache search coturn &> /dev/null; done

real    0m7.788s
user    0m7.400s
sys     0m0.504s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache depends coturn &> /dev/null; done

real    0m7.541s
user    0m7.232s
sys     0m0.432s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache depends coturn &> /dev/null; done

real    0m7.550s
user    0m7.212s
sys     0m0.464s
```
#### Cache in RAM
```
root@Marlin-Laptop:/tmp# cat /etc/apt/apt.conf.d/99-dietpi-no_cache
Dir::Cache "/tmp/apt";
Dir::Cache::archives "/tmp/apt";
Dir::Cache::pkgcache "pkg.bin";
Dir::Cache::srcpkgcache "src.bin";
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache show coturn &> /dev/null; done

real    0m0.457s
user    0m0.384s
sys     0m0.040s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache show coturn &> /dev/null; done

real    0m0.455s
user    0m0.372s
sys     0m0.048s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache search coturn &> /dev/null; done

real    0m1.969s
user    0m1.776s
sys     0m0.160s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache search coturn &> /dev/null; done

real    0m1.967s
user    0m1.756s
sys     0m0.180s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache depends coturn &> /dev/null; done

real    0m1.731s
user    0m1.620s
sys     0m0.080s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache depends coturn &> /dev/null; done

real    0m1.728s
user    0m1.612s
sys     0m0.084s
```
#### Cache on disk (default)
```
root@Marlin-Laptop:/tmp# cat /etc/apt/apt.conf.d/99-dietpi-no_cache
#Dir::Cache "/tmp/apt";
#Dir::Cache::archives "/tmp/apt";
#Dir::Cache::pkgcache "pkg.bin";
#Dir::Cache::srcpkgcache "src.bin";
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache show coturn &> /dev/null; done

real    0m0.455s
user    0m0.360s
sys     0m0.060s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache show coturn &> /dev/null; done

real    0m0.440s
user    0m0.352s
sys     0m0.052s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache search coturn &> /dev/null; done

real    0m1.966s
user    0m1.760s
sys     0m0.168s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache search coturn &> /dev/null; done

real    0m1.963s
user    0m1.812s
sys     0m0.120s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache depends coturn &> /dev/null; done

real    0m1.728s
user    0m1.616s
sys     0m0.080s
root@Marlin-Laptop:/tmp# time for i in {1..5}; do apt-cache depends coturn &> /dev/null; done

real    0m1.725s
user    0m1.604s
sys     0m0.084s
```
#### Result
- Significantly slower package info queries without cache.
- Interestingly RAM vs disk makes no difference. Seems that reading the file is just a tiny part while gathering and preparing the information makes up the time, if the info is already available locally.
- I am just wondering why this cache is even required, since the package lists themselves, which are as well stored on disk, already contain all of this info:
```
root@Marlin-Laptop:/tmp# head -20 /var/lib/apt/lists/deb.debian.org_debian_dists_stretch_contrib_binary-amd64_Packages
Package: alien-arena
Version: 7.66+dfsg-3
Installed-Size: 2067
Maintainer: Debian Games Team <pkg-games-devel@lists.alioth.debian.org>
Architecture: amd64
Depends: libc6 (>= 2.17), libcurl3-gnutls (>= 7.16.2), libfreetype6 (>= 2.3.5), libgcc1 (>= 1:3.0), libjpeg62-turbo (>= 1.3.1), libogg0 (>= 1.0rc3), libstdc++6 (>= 4.1.1), libvorbis0a (>= 1.1.2), libvorbisfile3 (>= 1.1.2), libx11-6, libxxf86vm1, zlib1g (>= 1:1.1.4), libopenal1, alien-arena-data
Description: Standalone 3D first person online deathmatch shooter
Homepage: http://red.planetarena.org
Description-md5: de2b3d0db5845c79b22ffc0c38842f1b
Tag: game::fps, hardware::input:keyboard, hardware::input:mouse,
 hardware::opengl, implemented-in::c, interface::3d,
 interface::graphical, interface::x11, network::client, role::program,
 uitoolkit::sdl, use::gameplaying, x11::application
Section: contrib/games
Priority: extra
Filename: pool/contrib/a/alien-arena/alien-arena_7.66+dfsg-3_amd64.deb
Size: 774388
MD5sum: d54108f51edd8ad9d147da8261e0473f
SHA256: 4c58933e396ba5977dade1318a3f0545a5a98283b87a8bbe5eecd2a5ccc5ec8a
```
- Also the size is quite comparable.
- But `apt-cache` commands with disabled cache clearly download this info, as can be seen by monitoring e.g. via `iptraf-ng`, instead of scanning the list files on disk...

MichaIng, 2019-06-10:
PR up: https://github.com/MichaIng/DietPi/pull/2904

It is important to allow handling cache, lists and download (archive) dirs separately since this vastly influences RAM usage:
- Cache `(src)pkgcache.bin`: 50-100M
- Lists `/var/lib/apt/lists`: 50-100M
- Download archive `/var/cache/apt/archive`: All `.deb` files of a single APT install/upgrade call are downloaded there at once, so this can grow very large when doing large APT installs. With disabled cache how these archives are removed automatically after install has finished.

closed 2019-07-03 by MichaIng