MichaIng/DietPi #2841: DietPi-PREP: Failed to get canonical path of `/dev/simfs'.

LexiconCode, 2019-05-21:
#### Details:
- Date           | Tue 21 May 16:02:04 BST 2019
- Bug report     | N/A
- DietPi version | v6.24.1 (MichaIng/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | NULL (index=20)
- Kernel version | #1 SMP Wed Feb 27 09:04:24 MSK 2019
- Distro         | stretch (index=4)
- Command        | update-grub
- Exit code      | 1
- Software title | DietPi-PREP

#### Steps to reproduce:

1. Follow the instructions on [PREP_SYSTEM_FOR_DIETPI.sh | Automate](https://github.com/MichaIng/DietPi/issues/1285#issue-280771944)

#### Expected behaviour:

- The script should convert to a diet by system.

#### Actual behaviour:

- failed to get canonical path of `/dev/simfs'

#### Extra details:

- This is being installed on a commercial VPS (OpenVZ) 
- The original system was Debian 8 64 Bit which I updated in place using to Debian 9 64 Bit using this [Guide](https://phoenixnap.com/kb/how-to-upgrade-debian-8-jessie-to-debian-9-stretch)

#### Additional logs:
```
Log file contents:
/usr/sbin/grub-probe: error: failed to get canonical path of `/dev/simfs'.
```

MichaIng, 2019-05-21:
@LexiconCode 
Many thanks for your report.

Yeah an initial Jessie system with some custom VPS setup always includes unexpected results. `/dev/simfs` might be something like the virtual file system of the VPS boot partition (with bootloader on it => grub), although strange then that it does not exist. Possibly some package is required or something has gone with the Stretch upgrade.

Btw I recommend you use DietPi-PREP to upgrade Jessie => Stretch. It supports this! It detects Jessie and allows you to select Stretch and even Buster as target repo.

The benefit is that since DietPi-PREP vastly reduces the package amount, the distro upgrade is much faster and less risky. However if the manual upgrade went fine, the result will be the same.

Can you paste:
```
lsblk
df
```

LexiconCode, 2019-05-21:
Further testing yields of the system is unbootable and I can't access via SSH to assess. I will try the `use DietPi-PREP to upgrade Jessie => Stretch.` with a quick reinstall.

MichaIng, 2019-05-21:
@LexiconCode 
Since the error appeared during `update-grub` command which is the bootloader update, it is very expected that a failure there results in an unbootable system. Please inspect the drive and partition setup first!

LexiconCode, 2019-05-21:
Filesystem     1K-blocks   Used Available Use% Mounted on
/dev/simfs      62914560 594996  62319564   1% /
devtmpfs         3145728      0   3145728   0% /dev
tmpfs            3145728      0   3145728   0% /dev/shm
tmpfs            3145728   9328   3136400   1% /run
tmpfs               5120      0      5120   0% /run/lock
tmpfs            3145728      0   3145728   0% /sys/fs/cgroup
tmpfs            3145728      0   3145728   0% /tmp
none             3145728      0   3145728   0% /run/shm

LexiconCode, 2019-05-21:
cat /etc/fstab
```
proc  /proc       proc    defaults    0    0
none  /dev/pts    devpts  rw,gid=5,mode=620    0    0
none  /run/shm    tmpfs   defaults    0    0
```

MichaIng, 2019-05-21:
@LexiconCode 
Strange, no /dev/simfs mount entry in fstab 🤔. There must be some other mechanism mounting it.

Which base image did you install on the VPS? Are there read-to-run Debian Jessie images available?

It needs to be checked which mechanism mounts the rootfs (/dev/simfs). Is grub even the bootloader?
`dpkg -l | grep grub`

LexiconCode, 2019-05-21:
Base Image: Debian 8 64 Bit
Provider: https://www.x4servers.com/

`dpkg -l | grep grub` returns nothing
df -T
```

Filesystem     Type     1K-blocks   Used Available Use% Mounted on
/dev/simfs     simfs     62914560 595080  62319480   1% /
devtmpfs       devtmpfs   3145728      0   3145728   0% /dev
tmpfs          tmpfs      3145728      0   3145728   0% /dev/shm
tmpfs          tmpfs      3145728   9328   3136400   1% /run
tmpfs          tmpfs         5120      0      5120   0% /run/lock
tmpfs          tmpfs      3145728      0   3145728   0% /sys/fs/cgroup
tmpfs          tmpfs      3145728      0   3145728   0% /tmp
none           tmpfs      3145728      0   3145728   0% /run/shm

```

If this is of help I would be willing to give you SSH access.

MichaIng, 2019-05-21:
@LexiconCode 
I think there is not very much we can do. This VPS seems to use a custom bootloader and custom virtual drive. We could hack this to working perhaps, but you will run into issues again, e.g. running DietPi drive manager and such.

However to play around:
- `ls -al /boot` # To perhaps get a hint about the boot loader
- In DietPi-PREP script you need to remove the `grub-pc`/`grub-efi` ATP install lines.
- Check `dpkg-query -s initramfs-tools`. If it is not installed, then remove the line containing `initramfs-tools` from DietPi-PREP script as well.
- After DietPi-PREP has finished, you need to remove the `/dev/simfs` line from `/etc/fstab`, since this does not exist and the rootfs is mounted differently.

LexiconCode, 2019-05-21:
That's a shame but understandable. I'd rather find another VPS that works.

ls -al /boot
```
total 8
drwxr-xr-x  2 root root 4096 Nov 26  2016 .
drwxr-xr-x 23 root root 4096 May 21 11:24 ..
```
dpkg-query -s initramfs-tools.
`dpkg-query: package 'initramfs-tools.' is not installed and no information is available`

LexiconCode, 2019-05-21:
I wonder if it's worthwhile Creating PREP_SYSTEM_FOR_DIETPI.sh for VPS? The DietPi-Drive_Manager isn't necessary for a VPS. Backups are handled by the VPS and they typically only have one virtual drive.

MichaIng, 2019-05-21:
@LexiconCode
Okay all bootloader stuff seems to run completely outside the virtual drive.

You can give it a try with skipping grub and initramfs install in DietPi-PREP script

LexiconCode, 2019-05-21:
Sure I'll give that a try. Verify I should be using the following
`wget https://raw.githubusercontent.com/MichaIng/DietPi/jessie-support/PREP_SYSTEM_FOR_DIETPI.sh -O PREP_SYSTEM_FOR_DIETPI.sh
`

LexiconCode, 2019-05-21:
This may be because I didn't make the edits correctly. I commented out lines only. However when a script completes.
```
nano /etc/fstab
-bash: /usr/bin/nano: No such file or directory
```
https://gist.github.com/LexiconCode/bd28190e351fae3b8fc34a82dc06cb6b

LexiconCode, 2019-05-21:
After further investigation I come across a strange scenario as fstab seems to exist(I made a backup as well) however when I try to edit it it doesn't exist.

ls /etc/fstab
/etc/fstab

 nano /etc/fstab
-bash: /usr/bin/nano: No such file or directory

MichaIng, 2019-05-21:
@LexiconCode 
`nano` is what does not exist, although strange because if it was not installed, the error message should look different. Commenting lines in an array might create issues not sure.

But to verify: `sudo apt install nano`

LexiconCode, 2019-05-21:
I uncommented those two lines
https://gist.github.com/LexiconCode/bd28190e351fae3b8fc34a82dc06cb6b#file-prep_system_for_dietpi-sh-L797-L799

but receive
./PREP_SYSTEM_FOR_DIETPI.sh: line 1765: syntax error near unexpected token `}'
./PREP_SYSTEM_FOR_DIETPI.sh: line 1765: `       }'

MichaIng, 2019-05-21:
@LexiconCode 
This line as well needs to be uncommented: https://gist.github.com/LexiconCode/bd28190e351fae3b8fc34a82dc06cb6b#file-prep_system_for_dietpi-sh-L789

LexiconCode, 2019-05-21:
I just want to say thanks for your help I really appreciate it! 
The script runs as expected with the changes that you recommended.

> After further investigation I come across a strange scenario as fstab seems to exist(I made a backup as well) however when I try to edit it it doesn't exist.

I'm now experiencing the issue as described in the above quote. I have not rebooted yet because I cannot check the following.

>     * After DietPi-PREP has finished, you need to remove the `/dev/simfs` line from `/etc/fstab`, since this does not exist and the rootfs is mounted differently.

LexiconCode, 2019-05-21:
> 
> 
> @MichaIng
> I just want to say thanks for your help I really appreciate it!
> The script parses as expected with the changes that you recommended.
> 
> > After further investigation I come across a strange scenario as fstab seems to exist(I made a backup as well) however when I try to edit it it doesn't exist.
> 
> I'm now experiencing the issue as described in the above quote. I have not rebooted yet because I cannot check the following.
> 
> > ```
> > * After DietPi-PREP has finished, you need to remove the `/dev/simfs` line from `/etc/fstab`, since this does not exist and the rootfs is mounted differently.
> > ```
reference
https://unix.stackexchange.com/questions/441493/bash-usr-bin-nano-no-such-file-or-directory

MichaIng, 2019-05-21:
@LexiconCode 
> hash -r fixes the issue with nano.

Good find. So the $PATH location content seems to be cached and `hash -r` = `forget all remembered locations`.

Okay the fstab looks good. Was not 100% sure if/how the `/dev/simfs` was added. Could you paste:
`dpkg -l`
So just to see tat really all bootloader, initramfs etc packages that affect boot steps before kernel do not exist.

LexiconCode, 2019-05-21:
dpkg -l
```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                                            Version                              Architecture                         Description
+++-===============================================================-====================================-====================================-===================================================================================================================================
ii  adduser                                                         3.115                                all                                  add and remove users and groups
ii  apt                                                             1.4.9                                amd64                                commandline package manager
ii  apt-transport-https                                             1.4.9                                amd64                                https download transport for APT
ii  apt-utils                                                       1.4.9                                amd64                                package management related utility programs
ii  base-files                                                      9.9+deb9u9                           amd64                                Debian base system miscellaneous files
ii  base-passwd                                                     3.5.43                               amd64                                Debian base system master password and group files
ii  bash                                                            4.4-5                                amd64                                GNU Bourne Again SHell
ii  bash-completion                                                 1:2.1-4.3                            all                                  programmable completion for the bash shell
ii  bc                                                              1.06.95-9+b3                         amd64                                GNU bc arbitrary precision calculator language
ii  bsdutils                                                        1:2.29.2-1+deb9u1                    amd64                                basic utilities from 4.4BSD-Lite
ii  bzip2                                                           1.0.6-8.1                            amd64                                high-quality block-sorting file compressor - utilities
ii  ca-certificates                                                 20161130+nmu1+deb9u1                 all                                  Common CA certificates
ii  console-setup                                                   1.164                                all                                  console font and keymap setup program
ii  console-setup-linux                                             1.164                                all                                  Linux specific part of console-setup
ii  coreutils                                                       8.26-3                               amd64                                GNU core utilities
ii  cpio                                                            2.11+dfsg-6                          amd64                                GNU cpio -- a program to manage archives of files
ii  cron                                                            3.0pl1-128+deb9u1                    amd64                                process scheduling daemon
ii  curl                                                            7.52.1-5+deb9u9                      amd64                                command line tool for transferring data with URL syntax
ii  dash                                                            0.5.8-2.4                            amd64                                POSIX-compliant shell
ii  debconf                                                         1.5.61                               all                                  Debian configuration management system
ii  debian-archive-keyring                                          2017.5                               all                                  GnuPG archive keys of the Debian archive
ii  debianutils                                                     4.8.1.1                              amd64                                Miscellaneous utilities specific to Debian
ii  diffutils                                                       1:3.5-3                              amd64                                File comparison utilities
ii  dirmngr                                                         2.1.18-8~deb9u4                      amd64                                GNU privacy guard - network certificate management service
ii  dmsetup                                                         2:1.02.137-2                         amd64                                Linux Kernel Device Mapper userspace library
ii  dpkg                                                            1.18.25                              amd64                                Debian package management system
ii  dropbear-bin                                                    2016.74-5+deb9u1                     amd64                                lightweight SSH2 server and client - command line tools
[... cut ...]
ii  whiptail                                                        0.52.19-1+b1                         amd64                                Displays user-friendly dialog boxes from shell scripts
ii  xkb-data                                                        2.19-1+deb9u1                        all                                  X Keyboard Extension (XKB) configuration data
ii  zlib1g:amd64                                                    1:1.2.8.dfsg-5                       amd64                                compression library - runtime

```

MichaIng, 2019-05-21:
> ii  initramfs-tools
> ii  initramfs-tools-core 

Not sure why but they're still there. Need to be purged: `apt purge initramfs-tools initramfs-tools-core`

And two other leftovers: `apt purge module-init-tools insserv`

> Interesting. if I don't reboot and logout of the SSH session then log back in the passwords accepted but then SSH session immediately ends without an error message.

Jep possibly the login script fails due to missing RAMdisk... Hmm, actually only the login script should exit, not the whole session. However a reboot is required anyway.

LexiconCode, 2019-05-21:
No boot, but a valiant effort!

Well it was a good try and I'm glad we discovered a bug. I guess I'll try to pursue getting my money back.

MichaIng, 2019-05-21:
@LexiconCode 
> No boot, but a valiant effort!
Well it was a good try and I'm glad we discovered a bug.

Jep worth it, also good to learn a bid about different VPS instances.

> I guess I'll try to pursue getting my money back.

You could also ask them to help with/provide info info about special needs to boot their systems. Possibly some special required packages or files that need to stay or any that definitely need to be absent, like bootloader and most likely initramfs in this case.

MichaIng, 2019-05-21:
Is it this one? https://openvz.org/
Although you link: https://www.x4servers.com/vps-hosting/
And they state `Platform Independent`. The empty `/boot` directory was on a fresh Jessie image they provide, right? And also `ls -l /` does not list `initrd.img` and/or `vmlinuz`?

LexiconCode, 2019-05-22:
`boot` is empty.

`ls -l /` which does not list initrd.img and/or vmlinuz.
```
drwxr-xr-x  2 root root 4096 Nov 26  2016 bin
drwxr-xr-x  2 root root 4096 Nov 26  2016 boot
drwxr-xr-x  6 root  500  660 May 21 18:56 dev
drwxr-xr-x 85 root root 4096 May 21 18:56 etc
drwxr-xr-x  2 root root 4096 Nov 26  2016 home
drwxr-xr-x 13 root root 4096 May 21 18:56 lib
drwxr-xr-x  2 root root 4096 Nov 26  2016 lib64
drwx------  2 root root 4096 Nov 26  2016 lost+found
drwxr-xr-x  2 root root 4096 Nov 26  2016 media
drwxr-xr-x  2 root root 4096 Nov 26  2016 mnt
drwxr-xr-x  2 root root 4096 Nov 26  2016 opt
dr-xr-xr-x 58 root root    0 May 21 18:56 proc
drwx------  2 root root 4096 Nov 26  2016 root
drwxr-xr-x 17 root root  480 May 21 18:56 run
drwxr-xr-x  2 root root 4096 Nov 26  2016 sbin
drwxr-xr-x  2 root root 4096 Nov 26  2016 srv
drwxr-xr-x  7 root root    0 May 21 18:56 sys
drwxrwxrwt  7 root root  140 May 21 19:50 tmp
drwxr-xr-x 11 root root 4096 Nov 26  2016 usr
drwxr-xr-x 12 root root 4096 Nov 26  2016 var
```
Their control panel shows `OpenVZ` however contact with support indicates that they use solsuvm's official template to deploy their servers.

MichaIng, 2019-05-22:
@LexiconCode 
Hmm SolusVM offers VMs on different basis, but they do not seem to support Debian: https://documentation.solusvm.com/display/DOCS/SolusVM+Installer++-+Version+4

Yeah however, as said the support might help when you ask them how to install/setup a custom Debian-based image.

MichaIng, 2019-11-08:
I mark this issue as closed. Generic VPS support is for my knowledge a difficult thing, since every VPS invokes the image in a different way:
- Some may emulte that hardware only (like VirtualBox/VMware and such) and run fine with every arch-fitting Linux distro.
- Others emulate the bootloader too and invoke an external boot partition or initramfs for this, so any OS-internal kernel/firmware update can break things, if this does not fit to external bootloader, and especially any image-internal boot partition and related fstab entry might conflict.
- Again others might emulte the kernel (load an external one), like `qemu` does for ARM devices. Then the possible incompatibilities raise when the image comes with another kernel and software find and/or loads those modules/headers/... that do not fit the emulated kernel.

In theory we could add some env vars/menu that allows to skip initramfs, bootloader and/or kernel package installs, but if this then requires some other packages, flag files or special configs is again unknown, so currently out of scope IMO.

closed 2019-11-08 by MichaIng