MichaIng/DietPi #2842: Missing passwords - qBittorrent, NZBGet

echable9999, 2019-05-21:
Tried all of DietPi's install instructions suggested passwords, my own passwords that I set during DietPi OS install, tried the default suggested first-start passwords for NzbGet etc.., looked in some plaintext .conf files and tried what it said there. Still could not get in to either of these two apps. Most recent OS update flashed yesterday, apps installed today.

MichaIng, 2019-05-21:
@echable9999 
Many thanks for your report.

Did you use any special character? Note that it is the first password that you were asked to change during first boot setup, the global software password.

Damn that GitHub does not have an at least optional PM function. Do you have an account on dietpi.com/phpbb forum? There is a way to recover the password but I don't want to spread this so open here 😉.

echable9999, 2019-05-22:
No special characters. PMed you on forum. Thanks.

closed 2019-05-22 by echable9999