MichaIng/DietPi #2843: Zerotier

cocoflan, 2019-05-21:
### Creating a software request

#### Formal software information
- Software title | Zerotier
- Short description | People use ZeroTier to seamlessly connect laptops, desktops, phones, embedded devices, cloud resources, and apps any way they want anywhere they go. Our software transforms the entire world into a single data center.
- Official URL (if available) | https://zerotier.com/
- GitHub URL (if available) | https://github.com/zerotier/ZeroTierOne
- (Official) install documentation | https://zerotier.com/download.shtml

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

curl -s 'https://raw.githubusercontent.com/zerotier/download.zerotier.com/master/htdocs/contact%40zerotier.com.gpg' | gpg --import && \
if z=$(curl -s 'https://install.zerotier.com/' | gpg); then echo "$z" | sudo bash; fi

ome users may need to replace gpg with gpg2 in the above commands.

This first makes sure you have the contact@zerotier.com GPG key. If this succeeds it then fetches our install script and pipes it through GPG. This verifies the script against its embedded signature, and if it is valid only the verified portion is forwarded through to "sudo bash." Otherwise the script is not executed and you will see a GPG error about signature check failure.

The install.zerotier.com script first characterizes your Linux distribution and adds the ZeroTier official package repository if we have one available for you. It then installs the zerotier-one package, enables and starts the service, and prints your new 10-digit ZeroTier address.

For those who are a bit more careless lazy brave, the script also works without GPG:

curl -s https://install.zerotier.com/ | sudo bash

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- yes

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/+70

MichaIng, 2019-05-22:
@cocoflan 
Many thanks for your request. Definitely an interesting project if you want to connect two peers without having everything relayed over a VPN.

cocoflan, 2019-05-22:
I am testing it on my Raspberry pi 3 and Nanopi M4 for weeks now, all works perfect!

Greetings, hope it will be in dietpi soon.

JimmyRecard, 2019-07-07:
I would love this as well!

cocoflan, 2020-03-25:
Any news if Zerotier could be added to the  packages ?

MichaIng, 2020-03-26:
@cocoflan 
Probably with DietPi 6.30. However if someone is able to do a start, this could be done generally faster.