MichaIng/DietPi #2844: The Stress Terminal UI: s-tui

cocoflan, 2019-05-22:
### Creating a software request

#### Formal software information
- Software title | S-tui
- Short description | Stress-Terminal UI, s-tui, monitors CPU temperature, frequency, power and utilization in a graphical way from the terminal.
- GitHub URL (if available) | https://github.com/amanusk/s-tui
- (Official) install documentation | https://github.com/amanusk/s-tui/blob/master/README.md

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
Install with:

pip install s-tui --user

(This usuall creates an executable in ~/.local/bin/ dir. Make sure it is in your PATH)

To install as root

sudo pip install s-tui

If you are installing s-tui on a Raspberry-Pi you might need to install python-dev first

Installation in virtualenv with pipsi:

pipsi install s-tui

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
Yes, I could help out with this, e.g. if I experience issues. / Sorry, I am totally unexperienced in bash coding. -->
- yes

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2019-05-22:
@cocoflan 
Many thanks for your request. We'll have a look when we can.

cocoflan, 2019-05-23:
Standard tool i use on all my SBC's, love it!

Greetings

MichaIng, 2026-10-27:
Available via APT:
```sh
apt install s-tui
```
I see no point to implement this into `dietpi-software`. Just install the package and run `s-tui`, nothing we could reasonable to to make this any easier.

closed 2026-10-27 by MichaIng