MichaIng/DietPi #2846: NZBHydra2

echable9999, 2019-05-22:
### Creating a software request

#### Formal software information
- Software title: nzbhydra2
- Short description: metasearch for NZBs + more
- Official URL (if available): https://github.com/theotherp/nzbhydra2   (NOT nzbhydra.com=scam site)
- GitHub URL (if available): https://github.com/theotherp/nzbhydra2
- (Official) install documentation: https://github.com/theotherp/nzbhydra2/releases

#### Are there similar/alternative software titles available with DietPi-Software?

Not as far as I know. There used to be available for installation on my Synology one other NZB metasearch app that did something similar, but it was not updated for years and is now not available there anymore at least and I can't remember the name.  

#### What makes your requested software better than the above solutions, if available?

Fantastic functionality, ties all the NZB-threads together and makes it so much much more efficient and effective. Responsive developer that updates every few days. Very complex feature set. Completely free.

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

There is an install file release called linux.zip available on the github, I don't think it would be difficult to translate this to the DietPi "optimised software" install system. Probably installing the linux.zip myself is easy too, but just not gonna go there with command-line installs on my DietPi's if I can avoid it, noob here. Most important reason it should be in DietPi is that it is extremely useful and there is no other software like it.  

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

https://github.com/theotherp/nzbhydra2/releases/download/v2.6.10/nzbhydra2-2.6.10-linux.zip

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

I'm good at learning (book learning) and would help in any way I could so I could probably learn stuff, handle communications maybe, but I don't know programming. 

- ...

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

https://feathub.com/MichaIng/DietPi/+82

Thanks!

MichaIng, 2019-05-22:
@echable9999 
Many thanks for your request. We'll have a look when we can.

Camry2731, 2021-12-06:
To add onto this, another potential option could be Prowlerr, as its made by the same team that makes Radarr, Lidarr, and Readarr. Its like a combination of Jackett and NZBHydra.

Link: https://github.com/Prowlarr/Prowlarr

Joulinar, 2021-12-06:
https://github.com/MichaIng/DietPi/issues/4473

Already requested

Camry2731, 2021-12-06:
> https://github.com/MichaIng/DietPi/issues/4473
> 
> Already requested

Ah, welp, I appreciate you linking that. I apparently need to get better at the GitHub search function, as I searched earlier but couldn't find that, lol.

MichaIng, 2021-12-06:
I'll have a look into Prowlarr for DietPi v8.0. Just by chance I recognised that the HTTPS certificate for feathub.com has been renewed, so it can be accessed via browser again without warnings. But GitHub authentication does not yet work here 🤔. However, no a reason to again rely on it, but we migrate for votings to GitHub discussions.