MichaIng/DietPi #2848: Portainer

cocoflan, 2019-05-22:
### Creating a software request

#### Formal software information
- Software title | Portainer
- Short description | Portainer is a lightweight management UI which allows you to easily manage your different Docker environments (Docker hosts or Swarm clusters). Portainer is meant to be as simple to deploy as it is to use. It consists of a single container that can run on any Docker engine (can be deployed as Linux container or a Windows native container, supports other platforms too). Portainer allows you to manage your all your Docker resources (containers, images, volumes, networks and more) ! It is compatible with the standalone Docker engine and with Docker Swarm mode.
- Official URL (if available) | https://www.portainer.io/
- GitHub URL (if available) | https://github.com/portainer/portainer
- (Official) install documentation | https://portainer.readthedocs.io/en/latest/deployment.html

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
Install with:

$ docker volume create portainer_data
$ docker run -d -p 9000:9000 --name portainer --restart always -v /var/run/docker.sock:/var/run/docker.sock -v portainer_data:/data portainer/portainer

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- yes

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/+83

vmavromatis, 2020-11-15:
+1 for Portainer arguably the simplest and best docker GUI out there. It's always the first app I install on my dietpi in order to run my homeserver which comprises of 100% docker containers (nextcloud, pihole, ghost and many more).

Joulinar, 2020-11-15:
well it's 2 commands to be executed. Running it via `dietpi-software` would take much longer. Not sure if there would be any benefit as there is nothing to automate, except to have 2 commands executed behind each other 😃

vmavromatis, 2020-11-15:
It's true but then again most software installs are now 2 or 3 or 4 commands :D That said, it would be really cool to have it install both docker & docker-compose before as well as to consider to possibly package it with nginx-proxy-manager which is an excellent reverse proxy with a UI. Then the package becomes plug-n-play to start spinning up containers via portainer and expose them to the web in an easy fashion!

MichaIng, 2020-11-17:
Even if the install is very simple, then still having it in `dietpi-software` is kinda promotion, make users aware of solutions that exist and would have been overseen otherwise. As we have Docker in the list, a GUI for it is a reasonable addition IMO. Is always nice to have a set of software titles with companions/extensions as well. E.g. you use the "Search" option in `dietpi-software` and type "docker" and have Portainer GUI listed there as well, which makes you think "Oh nice, a GUI, never through about it, let's give it a try!" 😄.

closed 2020-12-20 by MichaIng