MichaIng/DietPi #2849: 64 bit image for NanoPi Fire3

meetyg, 2019-05-23:
#### Formal device information
- Device name | NanoPi Fire3
- Official product URL | http://wiki.friendlyarm.com/wiki/index.php/NanoPi_Fire3
- GitHub resource URL (if available) | 
- Image download URL | http://download.friendlyarm.com/NanoPiFire3

#### Is the SBC officially supported by the Debian installer?
Not that I could find...

#### If not, is a reliable 3rd party Debian image available for this SBC?
- Armbian: https://dl.armbian.com/nanopifire3/Debian_stretch_next.7z
I think that this is 64 bit (couldn't get it running properly on my NanoPi Fire3)

#### If not, are there install instructions for Debian available?

- ...

#### Vote for this iamge on FeatHub: https://feathub.com/MichaIng/DietPi/

You already support the NanoPi Fire3 with a 32bit (armhf) image.
Currently thats the image I'm running on my Fire3 (Thanks!)
Would it be possible to supply a 64 bit image in addition / instead of the 32bit image currently supplied?

The NanoPi feature an Octa-core 64 bit SOC, so it would be nice to take advantage of the 64 bit.
I think Armbian's Debian image is 64 bit, but for some odd reason, it doesn't work well on my Fire3 (hadn't had the time to contact Armbian devs about it yet...)

MichaIng, 2019-05-23:
@meetyg 
Many thanks for your request. Jep that actually makes sense. Also the more current kernel version would be great.

FriendlyARM generally seems to not ship Debian images (any more). So probably we need to switch to ARMbian as basis.

Since I have no M3/Fire3 here, I need to ask @Fourdee if he finds time to refresh this image. Since the currently offered one is from 2018, an update would be anyway good.

Or we find some else who is willing to create this image, using: https://github.com/MichaIng/DietPi/issues/1285#issue-280771944
If you can manage to get the ARMbian image running, I would be happy to guide you through the process.

meetyg, 2019-05-23:
I contacted Armbian devs about my problem. They suggested I try an older version to see if it works. It may also be a problem with specific memory chip version.
Anyways, I would be happy to help. I'll try to get Armbian working first.
BTW, Armbian also support Docker on the NanoPi Fire3, so this will also solve issue #2160 I suppose, since FriendlyArm's kernel doesnt have Docker support.

MichaIng, 2019-05-23:
@meetyg 
Note that ARMbian and DietPi mostly do different things:
- **ARMbian** does kernel development (e.g. producing 4.14 kernel for M3/Fire3 where FriendlyARM just ships 4.4, providing a Ordroid C1 mainline kernel 5.0 image where Hardkernel stucks at 3.10 etc.) and builds their images from scratch. This is a very difficult and heavy work and requires a larger team in most cases. And many issues can occur that need to be handles/taken care of when updating the kernel, this is why there are moreless regularly issues with ARMbian images, sometimes audio not working, sometimes something else, we found many reports. However as this is so difficult, most SBC manufacturers simply produce one single kernel version and never update it, at least not the base version, just internal additions/fixes in case outside the kernel version string. Only RPi Foundation has the SBC selling counts and man power to regularly upgrade their kernel for all RPi versions. No wonder if you see our install stats which should not be far away from the overall marked share: https://dietpi.com/survey/#device
- **DietPi** on the other hand does no kernel development and we use pre-created images instead, e.g. ARMbians ones or the ones shipped by manufacturers or third party contributors (in case of Odroids). So we can fully concentrate on the software layer part, the GUI and software installs, while avoiding the most time consuming kernel/firmware work. But we have to live with the limitations and in case errors that the shipped kernel/firmware includes, which is the downside. Best we can do is forwarding bugs and feature requests.

So as you can see, this is not comparable. Actually the combination is best:
- Having the one guys with related knowledge and experience working on the kernel/firmware side, while skipping the OS/software part, which then leads to bare Debian images.
- And having the other guys concentrating on the software layer part to make this bare Debian system user friendly (in all aspects, config, software installs, setup), as we are aiming to.
- Only important thing is to share credits fair on both sides, and as of effort + man power actually the larger credit needs to go to the kernel/firmware devs 😉.

meetyg, 2019-05-23:
Of course. I just meant to say that I like DietPi's "flavour" and tools.

MichaIng, 2019-11-07:
New images are available for testing, including Fire3 ARMv8 Debian Buster Linux 4.14 😃: #3221
I mark this issue as closed. Testing and feedback should go to the other thread.

closed 2019-11-07 by MichaIng