MichaIng/DietPi #2851: Wifi Hotspot crashes

Soltcius, 2019-05-23:
#### Details:
- Date           | Thu 23 May 12:25:16 BST 2019
- Bug report     | N/A
- DietPi version | v6.24.1 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1219 SMP Tue May 14 21:20:58 BST 2019
- Distro         | stretch (index=4)
- Command        | systemctl start hostapd
- Exit code      | 1
- Software title | DietPi-Config

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
```

MichaIng, 2019-05-23:
@Soltcius 
Many thanks for your report.

Please paste the output of:
```
journalctl -u hostapd
hostapd /etc/hostapd/hostapd.conf
```

Soltcius, 2019-05-25:
Thank you very much for yor help

May 25 05:05:14 DietPi systemd[1]: hostapd.service: Control process exited, code=exited status=1
May 25 05:05:14 DietPi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 25 05:05:14 DietPi systemd[1]: hostapd.service: Unit entered failed state.
May 25 05:05:14 DietPi systemd[1]: hostapd.service: Failed with result 'exit-code'.
May 25 05:05:16 DietPi systemd[1]: hostapd.service: Service hold-off time over, scheduling restart.
May 25 05:05:16 DietPi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 25 05:05:16 DietPi systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...
May 25 05:05:16 DietPi hostapd[13483]: Configuration file: /etc/hostapd/hostapd.conf
May 25 05:05:16 DietPi hostapd[13483]: Line 2: unknown configuration item '//driver'
May 25 05:05:16 DietPi hostapd[13483]: 1 errors found in configuration file '/etc/hostapd/hostapd.conf'
May 25 05:05:16 DietPi systemd[1]: hostapd.service: Control process exited, code=exited status=1
May 25 05:05:16 DietPi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 25 05:05:16 DietPi systemd[1]: hostapd.service: Unit entered failed state.
May 25 05:05:16 DietPi systemd[1]: hostapd.service: Failed with result 'exit-code'.
May 25 05:05:18 DietPi systemd[1]: hostapd.service: Service hold-off time over, scheduling restart.
May 25 05:05:18 DietPi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 25 05:05:18 DietPi systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...
May 25 05:05:18 DietPi hostapd[13492]: Configuration file: /etc/hostapd/hostapd.conf
May 25 05:05:18 DietPi hostapd[13492]: Line 2: unknown configuration item '//driver'
May 25 05:05:18 DietPi hostapd[13492]: 1 errors found in configuration file '/etc/hostapd/hostapd.conf'
May 25 05:05:18 DietPi systemd[1]: hostapd.service: Control process exited, code=exited status=1
May 25 05:05:18 DietPi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 25 05:05:18 DietPi systemd[1]: hostapd.service: Unit entered failed state.
May 25 05:05:18 DietPi systemd[1]: hostapd.service: Failed with result 'exit-code'.
May 25 05:05:20 DietPi systemd[1]: hostapd.service: Service hold-off time over, scheduling restart.
May 25 05:05:20 DietPi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 25 05:05:20 DietPi systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...
May 25 05:05:20 DietPi hostapd[13494]: Configuration file: /etc/hostapd/hostapd.conf
May 25 05:05:20 DietPi hostapd[13494]: Line 2: unknown configuration item '//driver'
May 25 05:05:20 DietPi hostapd[13494]: 1 errors found in configuration file '/etc/hostapd/hostapd.conf'
May 25 05:05:20 DietPi systemd[1]: hostapd.service: Control process exited, code=exited status=1
May 25 05:05:20 DietPi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 25 05:05:20 DietPi systemd[1]: hostapd.service: Unit entered failed state.
May 25 05:05:20 DietPi systemd[1]: hostapd.service: Failed with result 'exit-code'.
May 25 05:05:23 DietPi systemd[1]: hostapd.service: Service hold-off time over, scheduling restart.
May 25 05:05:23 DietPi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 25 05:05:23 DietPi systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...
May 25 05:05:23 DietPi hostapd[13496]: Configuration file: /etc/hostapd/hostapd.conf
May 25 05:05:23 DietPi hostapd[13496]: Line 2: unknown configuration item '//driver'
May 25 05:05:23 DietPi hostapd[13496]: 1 errors found in configuration file '/etc/hostapd/hostapd.conf'
May 25 05:05:23 DietPi systemd[1]: hostapd.service: Control process exited, code=exited status=1
May 25 05:05:23 DietPi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 25 05:05:23 DietPi systemd[1]: hostapd.service: Unit entered failed state.
May 25 05:05:23 DietPi systemd[1]: hostapd.service: Failed with result 'exit-code'.
May 25 05:05:25 DietPi systemd[1]: hostapd.service: Service hold-off time over, scheduling restart.
May 25 05:05:25 DietPi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 25 05:05:25 DietPi systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...
May 25 05:05:25 DietPi hostapd[13498]: Configuration file: /etc/hostapd/hostapd.conf
May 25 05:05:25 DietPi hostapd[13498]: Line 2: unknown configuration item '//driver'
May 25 05:05:25 DietPi hostapd[13498]: 1 errors found in configuration file '/etc/hostapd/hostapd.conf'
May 25 05:05:25 DietPi systemd[1]: hostapd.service: Control process exited, code=exited status=1
May 25 05:05:25 DietPi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 25 05:05:25 DietPi systemd[1]: hostapd.service: Unit entered failed state.
May 25 05:05:25 DietPi systemd[1]: hostapd.service: Failed with result 'exit-code'.
May 25 05:05:27 DietPi systemd[1]: hostapd.service: Service hold-off time over, scheduling restart.
[... cut ...]
Configuration file: /etc/hostapd/hostapd.conf
Line 2: unknown configuration item '//driver'root@DietPi:~# cat /etc/hostapd/hostapd.conf 
interface=wlan0
//driver=nl80211
ssid=ORANGE
hw_mode=g
channel=3
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=Orange1234
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
ieee80211n=1

1 errors found in configuration file '/etc/hostapd/hostapd.conf'
-----------------------------------------------------------------------------------------------------

PeterLacknase, 2019-05-25:
Just a Tyrann: delete the `driver` line and restart without any `driver` option in `hostapd.conf`.

MichaIng, 2019-05-25:
@PeterLacknase @Soltcius 
Better solution is to correct the driver line as it should be:
- Remove the `//`, which is wrong there: `sed -i 's|//driver|driver|' /etc/hostapd/hostapd.conf`

PeterLacknase, 2019-05-25:
Shouldn‘t be necessary, since it defaults to `hostap` if nothing else is stated:

[https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf](https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf)

If this works, one could add the driver line. Just to sort things out.

MichaIng, 2019-05-25:
@PeterLacknase 
That's true, however I am not sure how well the hostap driver works. There is some reason that all over the web you find guides mostly stating nl80211. But actually good to know. In case not a syntax is the issue, it is worth to try out hostap driver.

MichaIng, 2019-06-08:
I mark this issue as closed now, syntax error is quite clear. Feel free to reopen if required.

closed 2019-06-08 by MichaIng