MichaIng/DietPi #2852: Bug & Improvement: Idle Spindown vs. RootFS on external disk

Gill-Bates, 2019-05-23:
Hi there,

If I move the RootFS to an external HDD using the Drive Manager, it would be helpful if the value of the idle spindown is automatically set from default 30 minutes (241) to 0 (Disabled). Otherwise the Raspberry Pi is dead after 30 minutes. :-)

Bug: 
After I have set the IdleSpin Down to 0, the selection menu jumps back to 30 minutes as soon as I open the Drive Manager again.

[image]

`/dev/sda:
SG_IO: bad/missing sense data, sb[]:  70 00 01 00 00 00 00 0a 00 00 00 00 00 1d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 drive state is:  unknown

/dev/sda1:
SG_IO: bad/missing sense data, sb[]:  70 00 01 00 00 00 00 0a 00 00 00 00 00 1d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 drive state is:  unknown`

Fourdee, 2019-05-23:
@Gill-Bates 

> SG_IO: bad/missing sense data, sb[]: 

Would appear you drive is incompatible with ```hdparm``` which we use to apply spindown options.

Try:
```
hdparm -C /dev/sd*
```
```
root@Fuzon:~# hdparm -C /dev/sd*

/dev/sda:
 drive state is:  active/idle

/dev/sda1:
 drive state is:  active/idle

```

Fourdee, 2019-05-23:
Confirmed bug, always sets 30min, will check.

Fourdee, 2019-05-23:
Visual bug only, setting does get applied, but menu always defaults to 30, will fix.

Gill-Bates, 2019-05-23:
I still recommended to set the idle Spindown to Zero, if an user move the RootFS to an external disk.

Fourdee, 2019-05-23:
@Gill-Bates 

> I still recommended to set the idle Spindown to Zero, if an user move the RootFS to an external disk.

I disagree. Since the dawn of power saving options, default spin downs in all OS are set, usually 10/20 minutes for all drives.

Also, we can help save the planet by default, reducing everyones energy consumption :smiley: 

Option to change spindown is available, per user preference.

---

Original issue resolved in v6.25.

closed 2019-05-23 by Fourdee

MichaIng, 2019-05-23:
@Gill-Bates @Fourdee 
> Otherwise the Raspberry Pi is dead after 30 minutes. :-)

Did you verify this? Actually I don't believe that RPi will die. E.g. most HDD docking stations have idle times for spin down/off (e.g. on mine I cannot even control this, 10 mins fix) and AFAIK also this is default for all spinning drives on Debian, although not 100% sure. When the system is active, it regularly writes tiny bits to the drive, so that RootFS should never spin down anyway and if so, the system/kernel should handle this gracefully.
E.g. MariaDB never has any issues, even that the database/dietpi_userdata is moved to external drive which is most of the times in idle. Small writes are cached due to async, every now an then (~every two hours or such, when also cron jobs lead to higher overall writes) the drive spins up for 10 minutes, sync is done, then spins down.

However the above are just assumptions so far. Would be good if we can verify the apparent behaviour on this.