MichaIng/DietPi #2855: I-Sabre  K2M not recognized after upgrade to v6.24.1

willefg, 2019-05-25:
### Creating a bug report/issue

#### Required Information
- DietPi version v6.24.1
- Distro version | Stretch
- Kernel version |Linux DietPi 4.19.42-v7+  
- SBC device |RPi3 
- Power supply used |iFi iPower
- SDcard used | Kingston class10 8GB

#### Steps to reproduce

1. Upgrade  to v6.24.1
2. Reselected driver  i-Sabre K2M for my soundcard i-Sabre 9038Q2M in DietPi-Config after upgrade
3.Squeezelite not recognized in  LMS due to no soundcard available in DietPi/Squeezelite, Squeezelite is recognized when selecting onboard 3.5mm analog out as audio output.
4.

#### Expected behaviour

- I should have the i-Sabre as working audio device

#### Actual behaviour

- i-Sabre isn’t working

#### Extra details

- ...

MichaIng, 2019-05-25:
@willefg 
Many thanks for your report.

Actually with the new kernel 4.19 the driver is built-in: https://github.com/MichaIng/DietPi/issues/2690
Not sure if the source build somehow conflicts with this.

Ah actually those are two different overlays. Please try the following:
```
sed -i 's/i-sabre-k2m/i-sabre-q2m/g' /DietPi/config.txt
dtoverlay i-sabre-q2m
```
- Do not select it in `dietpi-config` anymore until we correctly implemented the new kernel driver with v6.25.

Check if now the sound card is detected.

willefg, 2019-05-25:
That did the trick ! Thanks !
I’ll leave it alone until the next upgrade.

closed 2019-05-25 by willefg

MichaIng, 2019-05-25:
@willefg 
Perfect, thanks for feedback. So then we know now that the `i-sabre-q2m` overlay supports `i-sabre-k2m` as well. That is very helpful.