MichaIng/DietPi #2858: Dietpi Upgrade v6.24.1 Issue

Kaddoxx, 2019-05-26:
#### Details:
- Date           | Sun 26 May 13:56:18 BST 2019
- Bug report     | N/A
- DietPi version | v6.23.3 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```
apt --fix-broken doesn't work
Message : 

dpkg: error processing archive /var/cache/apt/archives/libraspberrypi-bin_1.20190517-1_armhf.deb (--unpack):
 unable to make backup link of './opt/vc/bin/containers_dump_pktfile' before installing new version: Operation not permitted
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libraspberrypi-bin_1.20190517-1_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

MichaIng, 2019-05-26:
@Kaddoxx 
Many thanks for your report.
```
dpkg: error processing archive /var/cache/apt/archives/libraspberrypi-bin_1.20190517-1_armhf.deb (--unpack):
unable to make backup link of './opt/vc/bin/containers_dump_pktfile' before installing new version: Operation not permitted
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
```
- So there seems to be a deeper issue.

Could you try:
```
G_AGUP
G_AGA
apt-get upgrade
```
And paste the full output of the last command.

Check `dmesg` for any red lines, e.g. I/O errors.

Kaddoxx, 2019-05-26:
root@DietPi:~# apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libraspberrypi-bin : Depends: libraspberrypi0 (= 1.20190401-1) but 1.20190517-1 is installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

Kaddoxx, 2019-05-26:
root@DietPi:~# G_AGUP
[  OK  ] Root access verified.
[ INFO ] APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Hit:3 https://packages.sury.org/php stretch InRelease
Reading package lists...
[  OK  ] G_AGUP
root@DietPi:~# G_AGA
[ INFO ] APT autoremove + purge, please wait...
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
[FAILED] G_AGA
[ INFO ] APT autoremove + purge, please wait...
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
[FAILED] G_AGA

MichaIng, 2019-05-26:
> libraspberrypi-bin : Depends: libraspberrypi0 (= 1.20190401-1) but 1.20190517-1 is installed

What the hack, why `libraspberrypi-bin` is no upgraded as well?

What does `apt-get install libraspberrypi-bin` show?

If the above fails, please paste:
`apt show -a libraspberrypi-bin`

Kaddoxx, 2019-05-26:
The following packages will be upgraded:
  libraspberrypi-bin
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
8 not fully installed or removed.
Need to get 332 kB of archives.
After this operation, 4,096 B of additional disk space will be used.
Get:1 https://archive.raspberrypi.org/debian stretch/main armhf libraspberrypi-bin armhf 1.20190517-1 [332 kB]
Fetched 332 kB in 1s (323 kB/s)        
(Reading database ... 35691 files and directories currently installed.)
Preparing to unpack .../libraspberrypi-bin_1.20190517-1_armhf.deb ...
Unpacking libraspberrypi-bin (1.20190517-1) over (1.20190401-1) ...
dpkg: error processing archive /var/cache/apt/archives/libraspberrypi-bin_1.20190517-1_armhf.deb (--unpack):
 unable to make backup link of './opt/vc/bin/containers_dump_pktfile' before installing new version: Operation not permitted
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libraspberrypi-bin_1.20190517-1_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

Kaddoxx, 2019-05-26:
root@DietPi:~# apt policy libraspberrypi-bin
libraspberrypi-bin:
  Installed: 1.20190401-1
  Candidate: 1.20190517-1
  Version table:
     1.20190517-1 500
        500 https://archive.raspberrypi.org/debian stretch/main armhf Packages
 *** 1.20190401-1 100
        100 /var/lib/dpkg/status

MichaIng, 2019-05-26:
@Kaddoxx 
Okay, so because `libraspberrypi-bin` fails to upgrade, the other firmware packages fail to install as well.

Did you check `dmesg` for any errors?

Please also paste: `ls -l /opt/vc/bin`
Since we anyway want to upgrade it, try to remove it (it is not required for any relevant system boot step, so don't worry): `rm -R /opt/vc/bin`
- Possibly this throws some I/O or permissions error then. Lets see.

Kaddoxx, 2019-05-26:
Thanks for your help

There's the dmesg errors in red : 

[    0.178122] bcm2708_fb soc:fb: Failed to allocate GPU framebuffer (-22)
[    0.178164] bcm2708_fb soc:fb: probe failed, err -22
[    0.641501] vc_vchi_sm_init: failed to open VCHI service (-1)
[    0.641509] [vc_sm_connected_init]: failed to initialize shared memory service[    0.744471] bcm2835-clk 3f101000.cprman: pllh: couldn't lock PLL
[    0.846211] bcm2835-clk 3f101000.cprman: pllh: couldn't lock PLL
[    1.048394] bcm2835-clk 3f101000.cprman: pllh: couldn't lock PLL
[    6.271906] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
[    6.581981] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    6.582572] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 9.10.105 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2018-03-09 18:56:28

MichaIng, 2019-05-26:
@Kaddoxx 
Okay nothing unexpected. Those fb/clk/vchi errors are normal when you run the RPi with 16M memory share or headless and can be safely ignored.

Kaddoxx, 2019-05-26:
root@DietPi:~# ls -l /opt/vc/bin
total 72
-rwxr-xr-x 1 root    root              29036 Feb 18 10:10 containers_check_frame_int
-rwxr-xr-x 1 root    root              13536 Feb 18 10:10 containers_datagram_receiver
-rwsr-xr-x 1     192     65536         13340 Feb 18 10:10 containers_datagram_sender
-rwxr-xr-x 1    8448 301989888             0 Mar 11  2020 containers_dump_pktfile
-rwxr-xr-x 1 4194368    524288             0 Feb 18 10:10 containers_stream_server
-r-xr-xr-x 1    8192 268435456 2267744872312 Mar 12  2020 containers_test
lrwxrwxrwx 1 root    root                  9 Apr  1 15:03 dtparam -> dtoverlay
-rwxr-xr-x 1 root    root               7824 Feb 18 10:10 vcmailbox
-rwxr-xr-x 1 root    root              15160 Feb 18 10:10 vcsmem
root@DietPi:~# rm -R /opt/vc/bin
rm: cannot remove '/opt/vc/bin/containers_dump_pktfile': Operation not permitted
rm: cannot remove '/opt/vc/bin/containers_test': Operation not permitted

MichaIng, 2019-05-26:
@Kaddoxx 
Jep corruption, so we found the evil.

Please do:
```
> /forcefsck
reboot
```
- So the root file system is checked for errors and repaired, at least attempted to.

After reboot, do:
```
chattr -R -i /opt/vc/bin
rm -R /opt/vc/bin
```

Then try to upgrade the APT packages: `apt upgrade`

Kaddoxx, 2019-05-26:
Unfortunately it didn't work

root@DietPi:~# chattr -R -i /opt/vc/bin
root@DietPi:~# rm -R /opt/vc/bin
rm: cannot remove '/opt/vc/bin/containers_dump_pktfile': Operation not permitted
rm: cannot remove '/opt/vc/bin/containers_test': Operation not permitted

root@DietPi:~# apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libraspberrypi-bin : Depends: libraspberrypi0 (= 1.20190401-1) but 1.20190517-1 is installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

MichaIng, 2019-05-26:
@Kaddoxx 
Hmm:
```
chattr -i /opt/vc/bin/containers_dump_pktfile
chown root:root /opt/vc/bin/containers_dump_pktfile
chmod 700 /opt/vc/bin/containers_dump_pktfile
rm /opt/vc/bin/containers_dump_pktfile
chattr -i /opt/vc/bin/containers_test
chown root:root /opt/vc/bin/containers_test
chmod 700 /opt/vc/bin/containers_test
rm /opt/vc/bin/containers_test
```

Any errors while trying this?
And you did:
```
> /forcefsck
reboot
```
?

Kaddoxx, 2019-05-26:
root@DietPi:~# chattr -i /opt/vc/bin/containers_dump_pktfile
root@DietPi:~# chown root:root /opt/vc/bin/containers_dump_pktfile
chown: changing ownership of '/opt/vc/bin/containers_dump_pktfile': Operation not permitted
root@DietPi:~# chmod 700 /opt/vc/bin/containers_dump_pktfile
chmod: changing permissions of '/opt/vc/bin/containers_dump_pktfile': Operation not permitted
root@DietPi:~# rm /opt/vc/bin/containers_dump_pktfile
rm: cannot remove '/opt/vc/bin/containers_dump_pktfile': Operation not permitted
root@DietPi:~# chattr -i /opt/vc/bin/containers_test
root@DietPi:~# chown root:root /opt/vc/bin/containers_test
chown: changing ownership of '/opt/vc/bin/containers_test': Operation not permitted
root@DietPi:~# chmod 700 /opt/vc/bin/containers_test
chmod: changing permissions of '/opt/vc/bin/containers_test': Operation not permitted
root@DietPi:~# rm /opt/vc/bin/containers_test
rm: cannot remove '/opt/vc/bin/containers_test': Operation not permitted

Kaddoxx, 2019-05-26:
> @Kaddoxx
> Hmm:
>
> ```
> chattr -i /opt/vc/bin/containers_dump_pktfile
> chown root:root /opt/vc/bin/containers_dump_pktfile
> chmod 700 /opt/vc/bin/containers_dump_pktfile
> rm [...]

Kaddoxx, 2019-05-26:
No errors

MichaIng, 2019-05-26:
@Kaddoxx 
There are errors, that still changing these files is "not permitted".
Another fsck on reboot did not help, when you retried the commands afterwards?
What does the following show:
```
lsattr /opt/vc/bin/containers_dump_pktfile
lsattr /opt/vc/bin/containers_test
```

Ahh try the following (another flag that prevents removal even as root):
```
chattr -a /opt/vc/bin/containers_dump_pktfile
chown root:root /opt/vc/bin/containers_dump_pktfile
chmod 700 /opt/vc/bin/containers_dump_pktfile
rm 
chattr -a /opt/vc/bin/containers_test
chown root:root /opt/vc/bin/containers_test
chmod 700 /opt/vc/bin/containers_test
rm /opt/vc/bin/containers_test
```

MichaIng, 2019-06-25:
I mark this as closed, feel free to reopen is required.

closed 2019-06-25 by MichaIng