MichaIng/DietPi #2859: Odroid C2 | I2C support

MichaIng, 2019-05-26:
Would be great if we could implement into DietPi-Config to toggle it. Requires investigation which kernel modules/packages are required.

Also SPI and general serial device control would be fine.

Same for other Odroids, if possible.

Ref: https://dietpi.com/phpbb/viewtopic.php?p=18174#p18174

eevan78, 2020-04-21:
Hi MichaIng,
There are some guides at C2 Wiki. They are for Ubuntu, but could be useful.
* [I2C](https://wiki.odroid.com/accessory/add-on_boards/rtc_shield) 
* [SPI](https://wiki.odroid.com/accessory/display/led_matrix_shield#tab__odroid-c2) 

Best regards,
Ivan

MichaIng, 2020-04-21:
@eevan78 
Many thanks for sharing. I'll see if I find time with v6.30 to implement this.