MichaIng/DietPi #2860: DietPi-Software | Enable Chromium on ARMv6 (RPi1/Zero)

MichaIng, 2019-05-26:
@Fourdee 
Do you know why we disabled it? Actually since all ARMv6 are RPi and we simply install the packages Raspberry Pi firmware repo, this should work well.

Ref:
- https://dietpi.com/phpbb/viewtopic.php?f=9&t=5972
- https://dietpi.com/phpbb/viewtopic.php?f=11&t=5675

symbios24, 2019-05-30:
chromium is very slow on rpi zero

MichaIng, 2019-05-30:
@symbios24 
That would be a good reason actually. Would apply for all ARMv6 RPis. I just wonder a bid since Raspbian comes with Chromium as default browser for a while.

Some time back when I used desktop and browser on my RPi2, I also found Chromium and Firefox to be much less responsive then the default browser that time: `netsurf`
- But AFAIK since hardware acceleration and such have been applied they switched to Chromium. But no experience about speed.

However if it generally works, I think it's okay to leave choice at user if it's too slow or not, as long as it works fine otherwise.

MichaIng, 2020-03-13:
Implemented for v6.29: https://github.com/MichaIng/DietPi/blob/dev/CHANGELOG.txt#L17

closed 2020-05-03 by MichaIng