MichaIng/DietPi #2864: Pinebook | New image required to fix firstrun update issue

MichaIng, 2019-05-27:
Ref: https://dietpi.com/phpbb/viewtopic.php?p=18226#p18226

The fix was patched with v6.20 but the image we offer for download is older and does not yet contain it. Since error appears during APT upgrade, patch cannot be applied.

We can move the fix to pre-patches (we should do this anyway), but a new image would not hurt as well.

Fourdee, 2019-05-28:
@MichaIng 

Leave the image update with me, will try and do it today.

MichaIng, 2019-05-28:
@Fourdee 
Many thanks!

Try out https://github.com/MichaIng/DietPi/blob/dietpi-image/.meta/dietpi-image to speed things up. Should be safe, only GPT I am not sure about.

Fourdee, 2019-05-28:
Notes:
- Unable to change brightness of LCD panel | ```/sys/class/backlight/backlight/brightness``` now has no effect.

Fourdee, 2019-05-29:
Notes:
- ```resizepart``` does not re-arrange contents: https://unix.stackexchange.com/questions/169395/how-do-i-resize-partitions-and-filesystems-on-them

MichaIng, 2019-05-29:
@Fourdee 
You mean `parted resizepart`, right? https://github.com/MichaIng/DietPi/pull/2693/files#diff-38702c6380ec8afa2acb3aff1616a6ccR114

Hmm we anyway can't resize the partition to a lower value then the size of the file system with `resize2fs`. Also gparted simply denies to do so. And since `resize2fs` does not rearrange content of course, `parted resizepart` should include possible gaps. AFAIK gparted as well does not rearrange content, in terms of closing gaps. It moves partitions around if you choose to do so but does not allow to shrink them any further than what resize2fs had as minimum before.

Did you run into the issue when running the above script?

Fourdee, 2019-06-04:
@MichaIng 

```dietpi-imager``` appears to be working now, nice work on the resize loop btw. Script will be a timer saver.

Redone the PB image, however, wont boot back up a second time after FS resize. more debugging to do.

Fourdee, 2019-06-09:
User reports truncated .7z, fails to extract. Will look into this once we fix the initial image.

For now, updated the URL for PB DL to this ticket.

MichaIng, 2019-06-09:
@Fourdee 
> truncated .7z

Since the 7z archive is not truncated of course, does the extraction indeed fail or flashing/booting the interned img file (which is truncated)?

Not sure if compression method could be an issue (LZMA2 ultra), but at with current Windows and Debian Stretch available versions it works. Also RAM usage on extraction is not high, just for the compression it is.

MichaIng, 2019-06-25:
New image from past week opens well, so I mark this as finished.

closed 2019-06-25 by MichaIng

canonex, 2019-09-21:
> #### PB image is currently being redone due to a previous update issue. We aim to have this image available again ASAP.
> 
> @MichaIng
> 
> `dietpi-imager` appears to be working now, nice work on the resize loop btw. Script will be a timer saver.
> 
> Redone the PB image, however, wont boot back up a second time after FS resize. more debugging to do.

Dear DiedPi,
the Pinebook download page redirects here:
is there any news for a fixed image?

Thank you
Riccardo

MichaIng, 2019-09-23:
@canonex 
Damn, the download link got reverted or was never updated. Just fixed it. Please try again, in case refresh browser cache by reloading the download page via CTRL+F5.

canonex, 2019-09-24:
I'm downloading the image right now, the link works fine.
I'll report any issue when installing.

Thank you,
Riccardo

canonex, 2019-09-25:
Diet is starting properly (even if no audio and don't know how to copy to eMMC).

MichaIng, 2019-09-26:
@canonex 
Many thanks for testing.

Selecting the default sound card via dietpi-config > Audio Options does not enable audio?

AFAIK it should be possible to flash the image to the eMMC directly, if you have some adapter. Or you download it on the now running Pinebook system and flash it onto internal eMMC via `dd`.

Finally it would be great if we would ship some DietPi installer that you flash on e.g. USB flash drive and which then installs/flashes DietPi to an internal drive. We have this for our EFI images, where it is required, and tools/methods (via Clonezilla) to create images for CSM/BIOS images as well. They only need some fine tuning and finally someone needs to find time to create those images, test them etc 😉.

canonex, 2019-09-26:
@MichaIng thank you, dd solved the copy.

Audio is not working, even after update. I should open a bug report (was working with Manjaro).
The configuration scripts are bash super-powered!