MichaIng/DietPi #2867: I-Sabre-K2M driver does not compile

moatazsalemVF, 2019-05-28:
Hello,

after installing and upgrading to I-Sabre-K2M driver does not compile with error

DietPi-Set_hardware: make CC=gcc-4.9                                                     
   - Exit code: 2                                                                          
   - DietPi version: v6.24.1 (MichaIng/master) | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4         
   - Image creator: DietPi Core Team                                                       
   - Pre-image: Raspbian Lite                                                              
                                                                                           
  Log file contents:                                                                       
                            ^                                                              
  /tmp/DietPi-Set_hardware/i-sabre-k2m_build/i-sabre-codec.c:200:16: error:                
  dereferencing pointer to incomplete type                                                 
     dev_err(codec->dev, "Bad frame size: %d                                               
  ",                                                                                       
                  ^                                                                        
  ./include/linux/device.h:1442:11: note: in definition of macro ‘dev_err’                 
    _dev_err(dev, dev_fmt(fmt), ##__VA_ARGS__) 

─────────────────────────────────────────────────────
 DietPi v6.24.1 : 20:43 - Tue 28/05/19
 ─────────────────────────────────────────────────────
 - Device model : RPi 3 Model B+ (armv7l)

MichaIng, 2019-05-28:
@moatazsalemVF 
I-Sabre-K2M has been implemented into the new kernel v4.19. The previously required custom dtoverlay build fails.

Simply run from console:
```
dtoverlay i-sabre-q2m
echo 'dtoverlay=i-sabre-q2m' >> /DietPi/config.txt
```
And you should be able to select this sound card from media players, respectively it is used automatically by ALSA.

Will be implemented correctly with v6.25. Until then, avoid using dietpi-config sound card selection for this DAC.

moatazsalemVF, 2019-05-29:
Many thanks @MichaIng 

i will be waiting...impatiently :)

closed 2019-05-29 by moatazsalemVF