MichaIng/DietPi #2868: add duckdns into remote access diet-pi software tab

mzramna, 2019-05-29:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- duckdns is an easy and free to use alternative to no-ip,that can be asily setup from terminal

#### Describe the solution you'd like:

- could be added to the software tab as an alternative to the already existent softwares

#### Additional context

- it dows exactaly the same as no-ip,so why not put it in there since it is free,and simple to setup? 
the own duckdns website already gives the bash script to setup in solutions like the dietpi-software (just change the variables by the given website name and the given token)

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2019-05-29:
@mzramna 
Many thanks for your request.

We'll have a look when we can. You might want to raise priority by adding the request to FeatHub and complete the link above to collect votes.

MichaIng, 2019-10-29:
Ref: https://dietpi.com/phpbb/viewtopic.php?p=18685#p18685

Joulinar, 2021-05-04:
@MichaIng 
I guess it can be closed as 7.1 has been released 😃

MichaIng, 2021-05-04:
Whoops, I forgot to add the milestone, so oversaw it, thanks.

closed 2021-05-04 by MichaIng