MichaIng/DietPi #2869: Dietpi Software: Wifi Hostspot

leealder, 2019-05-29:
-------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Wed 29 May 15:31:13 BST 2019
- Bug report     | N/A
- DietPi version | v6.24.1 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1219 SMP Tue May 14 21:20:58 BST 2019
- Distro         | buster (index=5)
- Command        | systemctl start hostapd
- Exit code      | 1
- Software title | DietPi-Config

#### Steps to reproduce:

1. ...Install item #60 "WiFi HostSpot"
2. ...

#### Expected behaviour:

- ... become HotSpot

#### Actual behaviour:

- ... hostapd.service ends in error

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
```
root@DietPi:~# systemctl status hostapd.service
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2019-05-29 15:59:20 BST; 1s ago
  Process: 2204 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=1/FAILURE)
root@DietPi:~#

MichaIng, 2019-05-29:
@leealder 
Many thanks for your report.

Please try it with the Raspberry Pi repo binary: `apt install --reinstall hostapd`

MichaIng, 2019-05-29:
I will change this to be the default on RPis, especially since this package/driver seems to support even RTL8188CU* WiFi chips.

leealder, 2019-05-29:
Sorry I kinda have to rely on you all for software. not quite as good as
you all!

MichaIng, 2019-05-29:
@leealder 
Jep, no worries. Simply copy paste the apt command I posted above. Afterwards you should be able to enable the WiFi Hotspot.

Fray2000, 2019-06-11:
Hi, the command apt install --reinstall hostapd does not work properly for me.
I am using a Netgear, Inc WPN111 RangeMax(TM) Wireless USB 2.0 Adapter with the same chip. Maybe add another

Fray2000, 2019-06-11:
Hi, the command apt install --reinstall hostapd does not work properly for me.
I am using a Netgear, Inc WPN111 RangeMax(TM) Wireless USB 2.0 Adapter with the same chip.

MichaIng, 2019-06-11:
@Fray2000 
This is on Raspberry Pi as well?

Please paste the output of: `hostapd /etc/hostapd/hostapd.conf` and `apt policy hostapd`

Fray2000, 2019-06-12:
Here is the output :

root@DietPi:~# echo $G_HW_MODEL
2
root@DietPi:~# hostapd /etc/hostapd/hostapd.conf
Configuration file: /etc/hostapd/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED
hostapd_free_hapd_data: Interface wlan0 wasn't started
root@DietPi:~# apt policy hostapd
hostapd:
  Installed: 2:2.6-21~bpo9~rpt1
  Candidate: 2:2.6-21~bpo9~rpt1
  Version table:
 *** 2:2.6-21~bpo9~rpt1 500
        500 https://archive.raspberrypi.org/debian stretch/main armhf Packages
        100 /var/lib/dpkg/status
     2:2.4-1+deb9u4 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
root@DietPi:~#

MichaIng, 2019-06-12:
@Fray2000 
Please paste `lsusb` to check WiFi chip. I could not reliable get it from web search.

> nl80211: Could not configure driver mode
> nl80211: deinit ifname=wlan0 disabled_11b_rates=0
> nl80211 driver initialization failed.

- Hmm at least worth to try to use the default "hostap" driver:
```
sed -i '/^driver=/d' /etc/hostapd/hostapd.conf
hostapd /etc/hostapd/hostapd.conf
```

Fray2000, 2019-06-21:
Sorry for the massive delay but i was having exams. 
Here is the output :
`root@DietPi:~# lsusb
Bus 001 Device 004: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 006: ID 1385:5f00 Netgear, Inc WPN111 RangeMax(TM) Wireless USB 2.0 Adapter
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`
Same Netgear product, the WPN111. It is an old wifi usb key and maybe have trouble with old drivers no more supported?

MichaIng, 2019-06-22:
@Fray2000 
> Sorry for the massive delay but i was having exams.

I hope everything went satisfying 😃.

Hmm no hint about RTL8188CU* chip.
Did you try:
```
sed -i '/^driver=/d' /etc/hostapd/hostapd.conf
hostapd /etc/hostapd/hostapd.conf
```

Last test with the very latest binaries:
[hostapd.zip](https://github.com/MichaIng/DietPi/files/3316797/hostapd.zip)
```
cd /tmp
wget https://github.com/MichaIng/DietPi/files/3316797/hostapd.zip
unzip https://github.com/MichaIng/DietPi/files/3316797/hostapd.zip
rm hostapd.zip
mv hostapd $(which hostapd)
mv hostapt_cli $(which hostapd_cli)
G_CONFIG_INJECT 'driver=' 'driver=nl80211' /etc/hostapd/hostapd.conf
hostapd /etc/hostapd/hostapd.conf
```

If this does not work then yeah, I am afraid either this dongle does not support AP mode or at least the hostapd drivers do not support it.

MichaIng, 2019-10-18:
@Fray2000 
If this is still an issue, since some time has passed, please retry with latest RPi firmware and latest hostapd package from RPi repo.

MichaIng, 2020-02-11:
I mark this issue as closed due to missing user reply. It could be resolved meanwhile kernel-wise, otherwise this WiFi adapter might simply not support AP/hotspot mode.

closed 2020-02-11 by MichaIng