MichaIng/DietPi #2870: Certbot installation disables HDMI on RPi Zero W

hyperionlevelq, 2019-05-29:
## Information:
- **DietPi version:** 6.24.1
- **Distro version:** Debian Stretch
- **Kernel version:** Linux DietPi 4.19.42+ #1219 Tue May 14 21:16:38 BST 2019 armv6l GNU/Linux
- **SBC device:** Raspberry Pi Zero W v1.1
- **Power supply used:** 5V 1A USB -> 120V (Generic Digipower brick with official-looking RPi USB cable)
- **SDcard used:** Kingston 16GB (Class 10)

- **Software title:** Certbot installation script

#### Steps to reproduce:
- Install Certbot from dietpi-software

#### Expected behaviour
- Reboot as normal with software installed

#### Actual behaviour
- Before rebooting, the process changed config.txt to disable HDMI, RPi took about 10 minutes after being plugged in to broadcast an IP address again. (I thought it wouldn't boot for the longest time, but an LED I have connected to an Arduino Nano connected to the Pi turned on after about 10 minutes of waiting.)

#### Extra details (the full log, see the bold part at bottom for unexpected behaviour)

dietpi@DietPi:~$ sudo dietpi-software
[  OK  ] DietPi-Software | Root access verified.
[  OK  ] DietPi-Software | RootFS R/W access verified.
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[ .... ] DietPi-Software | Testing connection to http://raspbian.raspberrypi.org/raspbian, please wait.[  OK  ] DietPi-Software | Connection test: http://raspbian.raspberrypi.org/raspbian
[  OK  ] NTPD: time sync | Completed
[  OK  ] DietPi-Software | Free space check: path=/ | available=9115 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[ .... ] DietPi-Software | Testing connection to http://raspbian.raspberrypi.org/raspbian, please wait.[  OK  ] DietPi-Software | Connection test: http://raspbian.raspberrypi.org/raspbian
[  OK  ] NTPD: time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask all: proftpd
[  OK  ] DietPi-Services | unmask all: mariadb
[  OK  ] DietPi-Services | unmask all: php7.3-fpm
[  OK  ] DietPi-Services | unmask all: lighttpd
[  OK  ] DietPi-Services | unmask all: gogs
[  OK  ] DietPi-Services | unmask all: mosquitto
[  OK  ] DietPi-Services | unmask all: node-red
[  OK  ] DietPi-Services | unmask all: webmin
[  OK  ] DietPi-Services | unmask all: docker
[  OK  ] DietPi-Services | unmask all: cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : docker
[  OK  ] DietPi-Services | stop : webmin
[  OK  ] DietPi-Services | stop : node-red
[  OK  ] DietPi-Services | stop : mosquitto
[  OK  ] DietPi-Services | stop : gogs
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : proftpd

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Update & upgrade APT

[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:3 https://download.docker.com/linux/raspbian stretch InRelease
Hit:4 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...
[  OK  ] DietPi-Software | G_AGUP
[ INFO ] DietPi-Software | Running apt simulation to check for errors, please wait...
[ INFO ] DietPi-Software | APT installation for: bash-doc -s, please wait...
Inst bash-doc (4.4-5 Raspbian:stable [all])
Conf bash-doc (4.4-5 Raspbian:stable [all])
[... 176 lines cut ...]
[  OK  ] DietPi-Services | dietpi_controlled : lighttpd
[  OK  ] DietPi-Services | dietpi_controlled : gogs
[  OK  ] DietPi-Services | dietpi_controlled : mosquitto
[  OK  ] DietPi-Services | dietpi_controlled : node-red
[  OK  ] DietPi-Services | dietpi_controlled : webmin
[  OK  ] DietPi-Services | dietpi_controlled : docker
[  OK  ] DietPi-Services | dietpi_controlled : cron
**[ SUB1 ] DietPi-Set_hardware > headless (disable)
[  OK  ] DietPi-Set_hardware | Comment in /DietPi/config.txt converted to setting: #hdmi_ignore_hotplug=0
[  OK  ] DietPi-Set_hardware | Comment in /DietPi/config.txt converted to setting: #hdmi_ignore_composite=0
[  OK  ] DietPi-Set_hardware | Comment in /DietPi/config.txt converted to setting: #hdmi_blanking=0
[  OK  ] headless disable | Completed
[  OK  ] DietPi-Survey | Connection test: ssh.dietpi.com**

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installation completed

[  OK  ] DietPi-Software | System rebooting in 3 seconds to finalise DietPi-Software installations. Please wait...
Connection to 192.168.2.10 closed by remote host.
Connection to 192.168.2.10 closed.
iMac:~ Bot-In-a-Box$ ssh dietpi@192.168.2.10
ssh: connect to host 192.168.2.10 port 22: Operation timed out

Like I said, I _was_ able to log back in and change back the HDMI settings, but I wasn't expecting this and figured it was worth sharing in case anybody else experienced this.

MichaIng, 2019-05-29:
@hyperionlevelq 
Many thanks for your report.

Actually the output is just misleading:
- `[ SUB1 ] DietPi-Set_hardware > headless (disable)`: It **disables** headless mode since you selected VNC server, which requires this.
- `[ OK ] DietPi-Set_hardware | Comment in /DietPi/config.txt converted to setting: #hdmi_ignore_hotplug=0`: This also states that that finally a commented entry is the result. If the setting would have been present in an active (not commented) way, the output would have been clearer to state that a setting was commented. The function we use is actually not really made to turn settings into comments, thus the misleading output. But the result is correct. Regardless I will change this to avoid confusion.

So actually you should have fount the three entries in a commented way, thus HDMI should have been still active.
__________
To check the long boot time, please paste: `dmesg`
And check `journalctl` the entries until networking has been started.

hyperionlevelq, 2019-05-30:
Right. OK, this makes a lot of sense. I can see now that the ```/DietPi/config.txt``` has the appropriate comments, as it should.

Although, the boot took around 7 minutes, if the ```journalctl``` is to be believed. 

```
A snippet from journalctl; note the time-gap at the end:
............
May 29 17:21:43 DietPi wpa_supplicant[429]: Successfully initialized wpa_supplicant
May 29 17:21:44 DietPi kernel: IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
May 29 17:21:44 DietPi kernel: brcmfmac: power management disabled
May 29 17:21:44 DietPi systemd[1]: Started Raise network interfaces.
May 29 17:21:44 DietPi systemd[1]: Reached target Network.
May 29 17:21:44 DietPi systemd[1]: Starting /etc/rc.local Compatibility...
May 29 17:21:44 DietPi systemd[1]: Starting DietPi-Boot...
May 29 17:21:44 DietPi systemd[1]: Starting Permit User Sessions...
May 29 17:21:44 DietPi systemd[1]: Started /etc/rc.local Compatibility.
May 29 17:21:44 DietPi systemd[1]: Started Permit User Sessions.
May 29 17:22:52 DietPi systemd[1]: dev-tty.device: Job dev-tty.device/start timed out.
May 29 17:22:52 DietPi systemd[1]: Timed out waiting for device dev-tty.device.
May 29 17:22:52 DietPi systemd[1]: Dependency failed for Serial Getty on tty.
May 29 17:22:52 DietPi systemd[1]: serial-getty@tty.service: Job serial-getty@tty.service/start failed with result 'dependency'.
May 29 17:22:52 DietPi systemd[1]: dev-tty.device: Job dev-tty.device/start failed with result 'timeout'.
May 29 17:28:11 DietPi kernel: random: crng init done
............
```
^ After this it connected to WiFi, and within another minute the boot was complete. 

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.42+ (dom@dom-XPS-13-9370) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1219 Tue May 14 21:16:38 BST 2019
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] On node 0 totalpages: 114688
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x49c with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=1024 bcm2708_fb.fbheight=768 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:90:72:E9 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=b8ef7a27-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 434484K/458752K available (6932K kernel code, 633K rwdata, 2068K rodata, 456K init, 797K bss, 16076K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
                   lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (6934 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   ( 456 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 634 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 798 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 25044 entries in 74 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000029] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000086] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000193] bcm2835: system timer (irq = 27)
[    0.000788] Console: colour dummy device 80x30
[    0.001421] console [tty1] enabled
[    0.001504] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060372] pid_max: default: 32768 minimum: 301
[    0.060962] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061021] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062490] CPU: Testing write buffer coherency: ok
[... 214 lines cut ...]
[   13.011913] bcm2835_audio soc:audio: card created with 8 channels
[   13.257729] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   13.610103] ax88179_178a 1-1.2:1.0 eth0: register 'ax88179_178a' at usb-20980000.usb-1.2, ASIX AX88179 USB 3.0 Gigabit Ethernet, 00:50:b6:29:a3:01
[   13.619746] usbcore: registered new interface driver ax88179_178a
[   13.898294] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   13.910138] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   13.910845] usbcore: registered new interface driver brcmfmac

[... cut ...]
Thanks for the clarification about the HDMI thing. It was definitely confusing to see that output before the Pi rebooted itself. I was afraid for a bit that I would have had to redo the entire installation since I couldn't log-in using either HDMI or SSH. 

However, I think the issues I experienced had something to do with the Pi's WiFi settings, since that doesn't seem to be working right now for some reason, and the boot process seemed to be waiting for WiFi, until it finally timed out. Though, I'm still not sure why HDMI didn't work (it still isn't working, though that might be because I'm using an HDMI to VGA converter?) but I'll make another update if I figure that out.

MichaIng, 2019-05-30:
@hyperionlevelq 
> May 29 17:28:11 DietPi kernel: random: crng init done

Indeed, known issue: https://github.com/MichaIng/DietPi/issues/2806
Please do: `G_AGI haveged`
If that resolves the issue as expected, if you don't mind, please send a `dietpi-bugreport` as I want to collect systems where this issue occurs to eventually help solving this Debian-side, or to know in which cases we should install the above package automatically via DietPi-Update. So far it looks like only PHP7.3-FPM users are affected. But not all of them, e.g. my RPi2 with PHP7.3-FPM runs fine for 1.5 years or such 🤔.

hyperionlevelq, 2019-05-30:
Wow. OK, yes that worked; boot time is back to normal . One ```G_SUDO G_AGI haveged``` later, I'm back in business. Never would've figured **low entropy** was the problem. However, unfortunately, now ```dietpi-bugreport``` isn't working...

```
[FAILED] DietPi-Bugreport | Failed to connect to SFTP server. 
Please try again later or report this to DietPi forum or GitHub repo in the first place.
```

MichaIng, 2019-05-30:
@hyperionlevelq 
Hmm perhaps a temporary overload. Please try again. Otherwise to test:
```
> testfile
curl -T "testfile" sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com/bugreport/
```

hyperionlevelq, 2019-05-30:
Output of that on my Pi:

```curl: (60) SSL peer certificate or SSH remote key was not OK
More details here: https://curl.haxx.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.```

MichaIng, 2019-05-30:
@hyperionlevelq 
Interestingly the testfile arrived.

Hmm perhaps the host key of ssh.dietpi.com is missing. Please try:
```
G_CONFIG_INJECT 'ssh.dietpi.com[[:blank:]]' 'ssh.dietpi.com ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQDE6aw3r6aOEqendNu376iiCHr9tGBIWPgfrLkzjXjEsHGyVSUFNnZt6pftrDeK7UX+qX4FxOwQlugG4fymOHbimRCFiv6cf7VpYg1Ednquq9TLb7/cIIbX8a6AuRmX4fjdGuqwmBq3OG7ZksFcYEFKt5U4mAJIaL8hXiM2iXjgY02LqiQY/QWATsHI4ie9ZOnwrQE+Rr6mASN1BVFuIgyHIbwX54jsFSnZ/7CdBMkuAd9B8JkxppWVYpYIFHE9oWNfjh/epdK8yv9Oo6r0w5Rb+4qaAc5g+RAaknHeV6Gp75d2lxBdCm5XknKKbGma2+/DfoE8WZTSgzXrYcRlStYN' /root/.ssh/known_hosts
```

hyperionlevelq, 2019-05-30:
I tried re-adding the key, but I got a warning that it already existed. So, I ran the curl test as suggested, and got this:
```
dietpi@DietPi:~$ sudo curl --connect-timeout 4 --retry 1 --retry-delay 4 -T testfile sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com/bugreport/
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0
Warning: Transient problem: timeout Will retry in 4 seconds. 1 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0
curl: (28) Operation timed out after 4476 milliseconds with 0 out of 0 bytes received
```

MichaIng, 2019-05-30:
@hyperionlevelq 
> curl: (28) Operation timed out after 4476 milliseconds with 0 out of 0 bytes received

Indeed a timeout 🤔.
Please do the following:
```
sed -i '/--connect-timeout 4 --retry 1 --retry-delay 4/--connect-timeout 10 --retry 1 --retry-delay 10/' /DietPi/dietpi/dietpi-backup
dietpi-bugreport # to send bug report
```

We should use the timeouts here that one can not set in via dietpi-config network options.

hyperionlevelq, 2019-05-30:
Sadly, another no-go...

MichaIng, 2019-05-30:
@hyperionlevelq 
Would be as well a no-go to let automated scripts run unlimited time since curl/wget by default do not apply any timeout. Having one is simply required!

The values we have in the code worked very well for several years. Not sure why lately some systems, even with fast internet bandwidth, have so slow connection times. **5s** for a simple sftp connection (upload not included) is the real no-go here. In my case it's **0.2s** for exactly the same operation 😉.

The reaction has already been done with v6.23 by allowing to adjust the timeouts. It has simply not yet been applied to bug report since there the usual connection test function is not in use.

Everything else is out of our control. It's core Linux/Debian network with default tools that are present on any default Raspbian/Debian image.

hyperionlevelq, 2019-05-30:
Yeah, that makes sense, and the connection really shouldn't be taking so long. Well, I guess I can't really make a bug report, then.

MichaIng, 2019-05-30:
@hyperionlevelq 
With the above it should work?
https://github.com/MichaIng/DietPi/issues/2870#issuecomment-497430303

hyperionlevelq, 2019-05-30:
```
root@DietPi:/home/dietpi# sed -i 's/--connect-timeout 4 --retry 1 --retry-delay 4/--connect-timeout 10 --retry 1 --retry-delay 10/' /DietPi/dietpi/dietpi-bugreport
root@DietPi:/home/dietpi# dietpi-bugreport
[  OK  ] DietPi-Bugreport | Root access verified.
[  OK  ] DietPi-Bugreport | RootFS R/W access verified.
[ INFO ] DietPi-Bugreport | Packing upload archive, please wait...
[  OK  ] DietPi-Bugreport | Connection test: ssh.dietpi.com
[FAILED] DietPi-Bugreport | Failed to connect to SFTP server. Please try again later or report this to DietPi forum or GitHub repo in the first place.
```
A check of ```/DietPi/dietpi/dietpi-bugreport``` confirms that the updated timeout is there, yet the error persists.

MichaIng, 2019-05-30:
@hyperionlevelq 
Hmm how long does it actually take?
`time curl -ILk https://ssh.dietpi.com`

In case:
```
sed -i 's/--connect-timeout 10 --retry 1 --retry-delay 10/--connect-timeout 20 --retry 1 --retry-delay 20/' /DietPi/dietpi/dietpi-bugreport
```

We could actually remove the timeout from dietpi-bugreport 🤔. It is never called non-interactively. So leave it running and print hint that ctrl+c can be used to cancel.

hyperionlevelq, 2019-05-30:
```
root@DietPi:/home/dietpi# time curl -ILk https://ssh.dietpi.com
HTTP/1.1 200 OK
Date: Thu, 30 May 2019 19:07:00 GMT
Server: Apache
Last-Modified: Sun, 19 May 2019 15:45:53 GMT
ETag: "18558-5893f82c528df"
Accept-Ranges: bytes
Content-Length: 99672
Vary: Accept-Encoding
Content-Type: text/html

real	0m3.476s
user	0m0.161s
sys	0m0.050s
```

MichaIng, 2019-05-30:
@hyperionlevelq 
Still long, but much less then 10s.

`time curl -IL sftp://ssh.dietpi.com` would be the more precise check. But this should only be done a single time, since fail2ban tracks and bans to much attempts (no user/password added here for the check).

hyperionlevelq, 2019-05-30:
```
root@DietPi:/home/dietpi# time curl -IL sftp://ssh.dietpi.com
curl: (67) Authentication failure

real	0m7.344s
user	0m0.964s
sys	0m0.097s
```

MichaIng, 2019-05-30:
@hyperionlevelq 
Ah there it is. Yeah so with the 20 above this should then work reliable. As mentioned I will remove the timeout and replace it with an info about how to exit the execution if it takes too long.

hyperionlevelq, 2019-05-30:
Yep, 20 will be sufficient (since that 7's from a computer that's low-power to begin with, pushed nearly to its limits, and using a satellite Internet connection).

MichaIng, 2019-05-30:
@hyperionlevelq 
Ah okay satellite internet is also a good reason.

I guess you already found the dietpi-config > Network Options: Misc > G_CHECK_URL_TIMEOUT setting?

I anyway plan to do a more precise error handling for the timeouts, so it a timeout has been detected that you will directly be offered to adjust them within the error prompt. As said they need to stay for all scripts that can be called non-interactively, which includes updates, software installs (automated first run installs), time sync, dietpi-survey and such. But for bug report indeed it can be removed completely.

hyperionlevelq, 2019-05-30:
Ah, the ```G_CHECK_URL_TIMEOUT```. I had forgotten about that actually, though I do remember changing that setting being useful when a clean installation failed a while back because of a timeout.

And, yes, being able to adjust timeouts directly within error prompts would be nice, for sure.

MichaIng, 2019-05-30:
I will reopen to now forget about the mentioned changes.

diveyez, 2019-06-12:
Uh. No it does not. I am sitting here, on my rpi0w with hdmi after using certbot to setup gitea. HDMI still works...

mtpiercey, 2019-06-12:
Like I noted above, it's possible that my monitor/HDMI-to-VGA converter caused one of the issues I experienced. However, I was suspecting the installation for doing it at first, since I was misled by one of the status messages it gave me before my Pi rebooted:

```sh
[ SUB1 ] DietPi-Set_hardware > headless (disable)
[  OK  ] DietPi-Set_hardware | Comment in /DietPi/config.txt converted to setting: #hdmi_ignore_hotplug=0
[  OK  ] DietPi-Set_hardware | Comment in /DietPi/config.txt converted to setting: #hdmi_ignore_composite=0
[  OK  ] DietPi-Set_hardware | Comment in /DietPi/config.txt converted to setting: #hdmi_blanking=0
[  OK  ] headless disable | Completed
```

However, I *was* experiencing the very real #2806, which greatly affected the time it took for my Pi to reboot.

Thanks for the confirmation that HDMI isn't actually affected.

MichaIng, 2019-06-12:
PR up: https://github.com/MichaIng/DietPi/pull/2907 (WIP)

closed 2019-07-03 by MichaIng