MichaIng/DietPi #2871: Pi-Hole Web GUI Failing After Upgrade to 6.24.1

5oundless, 2019-05-29:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.24.1
- Distro version | 9.9
- Kernel version | Linux RaspADB 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux
- SBC device | RPi 3 Model B+ (armv7l)
- Power supply used | CanaKit 2.5A USB Power Supply with Micro USB Cable
- SDcard used | Sandisk 32gb

#### Additional Information (if applicable)
- Software title | Pi-Hole
- Was the software title installed freshly or updated/migrated? Updated
- Can this issue be replicated on a fresh installation of DietPi? Unsure. Uninstall and reinstall of Lighttpd from dietpi-software did not resolve. 

- Bug report ID | 1f4f3e56-0781-4783-8262-567cc9b5e6df

#### Expected behaviour

- Pi-Hole webpage should load.

#### Actual behaviour

- 503 - Service Not Available

#### Extra details

- root@RaspADB:~# dietpi-services status pihole-FTL
[  OK  ] DietPi-Services | Root access verified.

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[FAILED] DietPi-Services | php7.3-fpm	failed (Result: timeout) since Tue 2019-05-28 20:28:52 CDT; 22h ago
● php7.3-fpm.service - The PHP 7.3 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.3-fpm.service; disabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Tue 2019-05-28 20:28:52 CDT; 22h ago
     Docs: man:php-fpm7.3(8)
  Process: 1008 ExecStart=/usr/sbin/php-fpm7.3 --nodaemonize --fpm-config /etc/php/7.3/fpm/php-fpm.conf (code=killed, signal=TERM)
 Main PID: 1008 (code=killed, signal=TERM)

May 28 20:27:22 RaspADB systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
May 28 20:28:52 RaspADB systemd[1]: php7.3-fpm.service: Start operation timed out. Terminating.
May 28 20:28:52 RaspADB systemd[1]: Failed to start The PHP 7.3 FastCGI Process Manager.
May 28 20:28:52 RaspADB systemd[1]: php7.3-fpm.service: Unit entered failed state.
May 28 20:28:52 RaspADB systemd[1]: php7.3-fpm.service: Failed with result 'timeout'.
[  OK  ] DietPi-Services | lighttpd	active (running) since Tue 2019-05-28 20:28:52 CDT; 22h ago
[  OK  ] DietPi-Services | plexmediaserver	active (running) since Tue 2019-05-28 20:28:53 CDT; 22h ago
[  OK  ] DietPi-Services | cron	active (running) since Tue 2019-05-28 20:28:53 CDT; 22h ago
[  OK  ] DietPi-Services | dropbear	active (running) since Tue 2019-05-28 20:27:20 CDT; 22h ago
[  OK  ] DietPi-Services | pihole-FTL	active (exited) since Wed 2019-05-29 00:17:03 CDT; 18h ago
[  OK  ] DietPi-Services | vncserver	active (exited) since Tue 2019-05-28 20:27:20 CDT; 22h ago
[  OK  ] DietPi-Services | xrdp	active (running) since Tue 2019-05-28 20:27:21 CDT; 22h ago
[  OK  ] DietPi-Services | dietpi-ramdisk	active (exited) since Tue 2019-05-28 20:26:58 CDT; 22h ago
[  OK  ] DietPi-Services | dietpi-ramlog	active (exited) since Tue 2019-05-28 20:26:58 CDT; 22h ago
[  OK  ] DietPi-Services | dietpi-preboot	active (exited) since Tue 2019-05-28 20:27:08 CDT; 22h ago
[  OK  ] DietPi-Services | dietpi-boot	active (exited) since Tue 2019-05-28 20:27:20 CDT; 22h ago
[  OK  ] DietPi-Services | dietpi-postboot	active (exited) since Tue 2019-05-28 20:27:20 CDT; 22h ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor	inactive (dead)
root@RaspADB:~#

MichaIng, 2019-05-29:
@5oundless 
Can you please paste: `dmesg | grep random`
And check whether `systemctl restart php7.3-fpm` finally succeeds but takes very long.

5oundless, 2019-05-29:
```
root@RaspADB:~# dmesg | grep random
[    0.000000] random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
[    1.183935] random: fast init done
[    2.021792] random: systemd: uninitialized urandom read (16 bytes read)
[    2.030315] random: systemd: uninitialized urandom read (16 bytes read)
[    2.149510] random: systemd-sysv-ge: uninitialized urandom read (16 bytes read)
[  164.463690] random: crng init done
[  164.463703] random: 7 urandom warning(s) missed due to ratelimiting
root@RaspADB:~# systemctl restart php7.3-fpm
root@RaspADB:~# systemctl status php7.3-fpm
● php7.3-fpm.service - The PHP 7.3 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.3-fpm.service; disabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-05-29 18:50:12 CDT; 8s ago
     Docs: man:php-fpm7.3(8)
 Main PID: 31032 (php-fpm7.3)
   Status: "Ready to handle connections"
   CGroup: /system.slice/php7.3-fpm.service
           ├─31032 php-fpm: master process (/etc/php/7.3/fpm/php-fpm.conf)
           ├─31033 php-fpm: pool www
           ├─31034 php-fpm: pool www
           ├─31035 php-fpm: pool www
           └─31036 php-fpm: pool www

May 29 18:50:11 RaspADB systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
May 29 18:50:12 RaspADB systemd[1]: Started The PHP 7.3 FastCGI Process Manager.
root@RaspADB:~# dietpi-services status pihole-FTL
[  OK  ] DietPi-Services | Root access verified.

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] DietPi-Services | php7.3-fpm	active (running) since Wed 2019-05-29 18:50:12 CDT; 15s ago
[  OK  ] DietPi-Services | lighttpd	active (running) since Tue 2019-05-28 20:28:52 CDT; 22h ago
[  OK  ] DietPi-Services | plexmediaserver	active (running) since Tue 2019-05-28 20:28:53 CDT; 22h ago
[  OK  ] DietPi-Services | cron	active (running) since Tue 2019-05-28 20:28:53 CDT; 22h ago
[  OK  ] DietPi-Services | dropbear	active (running) since Tue 2019-05-28 20:27:20 CDT; 22h ago
[  OK  ] DietPi-Services | pihole-FTL	active (exited) since Wed 2019-05-29 00:17:03 CDT; 18h ago
[  OK  ] DietPi-Services | vncserver	active (exited) since Tue 2019-05-28 20:27:20 CDT; 22h ago
[  OK  ] DietPi-Services | xrdp	active (running) since Tue 2019-05-28 20:27:21 CDT; 22h ago
[  OK  ] DietPi-Services | dietpi-ramdisk	active (exited) since Tue 2019-05-28 20:26:58 CDT; 22h ago
[  OK  ] DietPi-Services | dietpi-ramlog	active (exited) since Tue 2019-05-28 20:26:58 CDT; 22h ago
[  OK  ] DietPi-Services | dietpi-preboot	active (exited) since Tue 2019-05-28 20:27:08 CDT; 22h ago
[  OK  ] DietPi-Services | dietpi-boot	active (exited) since Tue 2019-05-28 20:27:20 CDT; 22h ago
[  OK  ] DietPi-Services | dietpi-postboot	active (exited) since Tue 2019-05-28 20:27:20 CDT; 22h ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor	inactive (dead)
root@RaspADB:~# 
```
@MichaIng thanks for the quick help!

systemctl restart php7.3-fpm worked immediately. However, after rebooting, the service was failed again. This time, running systemctl restart php7.3-fpm took 1-2 minutes.

With that said, this resolves the issue with the Pi-Hole GUI loading, but looks like it will require me to restart php7.3-fpm after each boot. Which is manageable.

MichaIng, 2019-05-30:
@5oundless 
Please try: `G_AGI haveged`
As of the linked issue, some users experience hanging boot and/or PHP-FPM start timing out due to empty entropy pool. An issue that seems to have become more popular with recent Debian and/or systemd changes, especially on systems without peripheral attached. The above package is a daemon that assures a filled entropy pool from early boot on.

5oundless, 2019-05-30:
It looks like haveged resolved the issue. Boot was faster and no issue with PHP this time.

`dietpi-bugreport 
[  OK  ] DietPi-Bugreport | Root access verified.
[  OK  ] DietPi-Bugreport | RootFS R/W access verified.
[ INFO ] DietPi-Bugreport | Packing upload archive, please wait...
[  OK  ] DietPi-Bugreport | Connection test: ssh.dietpi.com
[  OK  ] DietPi-Bugreport | Bug report sent, reference code: 1f4f3e56-0781-4783-8262-567cc9b5e6df
`

MichaIng, 2019-05-30:
@5oundless 
Okay bug report received and marked. Many thanks!

I mark this issue as closed now in favour of the other one to further investigate how to deal with this in general.

Feel free to reopen if required.

closed 2019-05-30 by MichaIng