MichaIng/DietPi #2873: Error after login / automatic distribution script

srGeo, 2019-05-30:
#### Details:
- Date           | Thu May 30 19:11:02 CEST 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi Zero W (armv6l) (index=1)
- Kernel version | #1219 Tue May 14 21:16:38 BST 2019
- Distro         | stretch (index=4)
- Command        | Connection test: http://raspbian.raspberrypi.org/raspbian
- Exit code      | 4
- Software title | DietPi-Login

#### Steps to reproduce:

1. This is an automatic script which starts after login. This seems to try testing the internet connection and stops with an error.
`[ .... ] DietPi-Login | (1/3) Testing connection to http://raspbian.raspberrypi.org/raspbian, please[FAILED] DietPi-Login | Connection test: http://raspbian.raspberrypi.org/raspbian`
2. After exiting this script the user shall open a ticket.

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:

```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2019-05-30 19:10:41--  http://raspbian.raspberrypi.org/raspbian
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... failed: Connection timed out.
wget: unable to resolve host address 'raspbian.raspberrypi.org'

--------------------------------------------------------------------

[FAILED] DietPi-Login | Unable to continue, DietPi-Login will now terminate.

```
I try to ping the URL manually an it work.

```
dietpi@DietPi-zero:~$ ping raspbian.raspberrypi.org
PING mirrordirector.raspbian.org (93.93.128.193) 56(84) bytes of data.
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=1 ttl=57 time=28.5 ms
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=2 ttl=57 time=29.2 ms
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=3 ttl=57 time=31.6 ms
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=4 ttl=57 time=33.3 ms
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=5 ttl=57 time=34.2 ms
^C
--- mirrordirector.raspbian.org ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 8068ms
rtt min/avg/max/mdev = 28.571/31.422/34.254/2.227 ms
```

I have no idea what is going wrong and why this script starts.

Thanks 
Sascha

MichaIng, 2019-05-30:
@srGeo 
A timeout. We added the possibility to adjust timeouts with v6.23 😄.

Ah luckily the code already takes an environment variable, so no manual code adjustment required. Please try:
```
export G_CHECK_URL_TIMEOUT=20
export timeout=20
/DietPi/dietpi/login
```

If this then works well, you can afterwards (after first run update) increase default timeouts. At best on dietpi-software menu prompt, enter dietpi-config > Network Options: Misc > G_CHECK_URL_TIMEOUT > Raise to 20 or such to be failsafe.

MichaIng, 2020-03-13:
I mark this issue as closed. Feel free to reopen if required.

closed 2020-03-13 by MichaIng