MichaIng/DietPi #2874: HQPlayer NAA daemon (networkaudiod) upgrade from 3.5.5 to 3.5.6

volpone, 2019-05-30:
HQPlayer NAA (networkaudiod) version is 3.5.5-39 in last DietPi released version.
NAA has been upgraded from 3.5.5-39 to 3.5.6-41
See: https://www.signalyst.eu/bins/naa/linux/stretch/
Thank's for your great work !

MichaIng, 2019-05-30:
@volpone 
Many thanks for your info. Will be implemented with v6.25!

MichaIng, 2019-05-31:
Okay, the next time NAA daemon gets an update, we will reinstall it with DietPi-Update to apply it to all users. I mark the issue as closed now. Manually reinstall to update the daemon as above.

closed 2019-05-31 by MichaIng

volpone, 2019-06-01:
> #### EDIT: Ah actually has already been done, not sure in which version between v6.20 and v6.24: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-software#L4457-L4472
> Note that we did not automated update. Perhaps we should have, especially since the install run is a simple trivial one. However if you need to update: `dietpi-software reinstall 124`

@MichaIng , sorry but not sure to understand correctly your instructions.
According changelog NAA has been updated to 3.5.5-39 in v6.20. I'm running v6.22.1 and my NAA version is actually 3.5.5-39.
```
root@USBridge:~# apt-cache policy networkaudiod
networkaudiod:
  Installé : 3.5.5-39
  Candidat : 3.5.5-39
 Table de version :
 *** 3.5.5-39 100
        100 /var/lib/dpkg/status
```
I've tried to reinstall using _dietpi-software reinstall 124_ but still NAA 3.5.5-39 installed.
Seems last NAA version is 3.5.6 according to https://www.signalyst.eu/bins/naa/linux/stretch/

**Do i have to upgrade first dietpi to 6.25** and then reinstall NAA with `dietpi-software reinstall 124` ? Thank's.

MichaIng, 2019-06-01:
@volpone 
> I've tried to reinstall using dietpi-software reinstall 124 but still NAA 3.5.5-39 installed.
> Do i have to update DietPi to 6.25 first and then reinstall NAA using dietpi-software reinstall 124 ? 

Strange, actually not, the current master code (v6.24.1) already contains v3.5.6 download and install: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L4451-L4466

> I'm running v6.22.1 and my NAA version is actually 3.5.5-39.

Ah that's it. Please update to current release (v6.24.1), other of course you can simply manually download and install the package.