MichaIng/DietPi #2877: APT update fails, fuzon.co.uk down?

benjaminpreiss, 2019-05-31:
#### Details:
- Date           | Sat  1 Jun 00:01:15 BST 2019
- Bug report     | N/A
- DietPi version | v6.24.1 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid N2 (aarch64) (index=15)
- Kernel version | #1 SMP PREEMPT Sat Mar 23 15:52:32 CET 2019
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:
1. log in to freshly installed dietpi via ssh
2. run dietpi-software
3. select open-ssh software as standard ssh server
4. wait for installation to complete

#### Expected behaviour:
ssh should just install

#### Actual behaviour:
installation aborts with the error from below:

#### Extra details:
I already encountere this behaviour 4 hours ago but it disappeared after trying several times.

#### Additional logs:
```
Log file contents:
Hit:1 https://download.docker.com/linux/debian stretch InRelease
Ign:2 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease
Hit:6 https://cdn-aws.deb.debian.org/debian stretch Release
Ign:8 http://fuzon.co.uk/meveric all InRelease
Ign:9 http://fuzon.co.uk/meveric stretch InRelease
Err:10 http://fuzon.co.uk/meveric all Release
  Unable to connect to fuzon.co.uk:http:
Err:11 http://fuzon.co.uk/meveric stretch Release
  Unable to connect to fuzon.co.uk:http:
Reading package lists...
E: The repository 'http://fuzon.co.uk/meveric all Release' does no longer have a Release file.
E: The repository 'http://fuzon.co.uk/meveric stretch Release' does no longer have a Release file.
```

MichaIng, 2019-06-01:
@benjaminpreiss 
Many thanks for reporting.

Jep, apparently our whole VPS host is down, even its website: https://myvirtualserver.com

Ref: https://github.com/MichaIng/DietPi/issues/2878

closed 2019-06-01 by MichaIng

crazyquark, 2019-07-16:
fuzon.co.uk is down again :(

ponasromas, 2019-07-16:
When this service be online? Is there any alternative to it?