MichaIng/DietPi #2878: https://dietpi.com/ website is down !!

cocoflan, 2019-05-31:


MichaIng, 2019-06-01:
@cocoflan 
My goodness, https://myvirtualserver.com itself is completely down. Damn, no chance to reach the VM.

cocoflan, 2019-06-01:
what is happening ???

MichaIng, 2019-06-01:
@cocoflan 
No idea. Since Fourdee is main contact, perhaps he got some info, not sure. Seem to be some serous issue if even their website is down + at least our VPS 😨.

cocoflan, 2019-06-01:
that is something that should be very exceptional, a host down, keep us informed.

MichaIng, 2019-06-01:
@cocoflan 
Jep, first time since 3 years 2 months. Hmm would bring my home server up to serve static downloads, but they're not up to date. Used this to keep static downloads up during server update and backup. Should keep it in sync regularly perhaps 🤔.

cocoflan, 2019-06-01:
yes i should do that, the more to know, always beter.

cocoflan, 2019-06-01:
Even https://status.myvirtualserver.com/ doen't work, not professional!!

cocoflan, 2019-06-01:
For users how need some manuals

https://web.archive.org/web/20171103051823/http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&sid=0559b1fac8c50dc0964322750da2cf1c

causefx, 2019-06-01:
:/ today was the day I was gonna try this out.... Now I can't download it :(

MichaIng, 2019-06-01:
@causefx 
Sorry for that. Tomorrow we will know more, it's late after midnight in Europe 😉.

causefx, 2019-06-01:
no worries :)

Hope it all works out for ya guys.

benjaminpreiss, 2019-06-01:
I just visited the status page. It says the downtime was planned and all systems should be Working again.

benjaminpreiss, 2019-06-01:
I just successfully ran sudo apt update and upgrade... server seems to be up and running.

Whatever you guys did, thanks!

MichaIng, 2019-06-01:
@benjaminpreiss 
Many thanks for the info. Jep, everything fine again. Hmm from the status page it sounds like if it was about the domain. But actually the VPS' were down as well, and they were not shut down gracefully but it was a kill. We can just hope there there is no database corruption or such, will do some checkup.

MichaIng, 2019-06-01:
Okay everything indeed seems to be fine, although I am still not happy with the fact that it was no ordered shutdown and that obviously there was not information given e.g. 1 week before. If one relies on the web services, the information about planned downtimes needs to be given way earlier so that one can prepare alternative servers and such.

However I mark this issue as closed.

closed 2019-06-01 by MichaIng