MichaIng/DietPi #2879: How to Disable MOD?

Gill-Bates, 2019-06-01:
After the login the message of the day "MOTD" annoys me very much. It's distracting me from important messages. I miss an option under dietpi-config to disable it.

MichaIng, 2019-06-01:
@Gill-Bates 
It's `dietpi-banner` where you can adjust banner options.

Which kind of info would you like to see in the MOTD?

Gill-Bates, 2019-06-03:
@michalng: Thank you. That menu was what I am looking for. Confusing that there are so many dedicated menus for the settings. My first logical touchpoint was Dietpi-config -> Display

MichaIng, 2019-06-03:
@Gill-Bates 
I plan another "appearance" option, to allow adjusting the output colour of DietPi scripts, e.g. banner and INFO/OK/FAILURE output and such. I would create a new "Appearance" menu to dietpi-config for this. "Display" usually means the display hardware you attached.

I will switch the MOTD to inform about the "dietpi-banner" menu 😉.

closed 2019-06-03 by MichaIng

candeira, 2020-01-03:
Would you accept a PR with an option way to completely disable the banner? I use Byobu as a tty "desktop", and would like to skip the wait for the banner on every bash split.

MichaIng, 2020-01-03:
@candeira 
Hmm, we thought about this but decided against it to keep credits visible in all cases. However when you disable all entries via `dietpi-banner`, the print should be immediate without any wait. You can even leave everything, but only disable MOTD and WAN IP, else those are the two which take up to two seconds for network connection. Speeding this up by pre-loading via cron job is planned, so the actual banner print will then always be immediate.