MichaIng/DietPi #2880: Show Uptime and last boot after Login

Gill-Bates, 2019-06-01:
After logging in via SSH, a note about uptime and the last boot process would be very helpful. This helps me with the error analysis.

I can solve it with the command `uptime`.

But the command `who -b` shows crap:

[image]

MichaIng, 2019-06-01:
@Gill-Bates 
You can enable it via `dietpi-banner` in the login banner. We use `uptime` as well.

I never used `who -b`, but it shows crap on my RPi2 as well. Possibly it reads the value before `fake-hwclock` applies the stored one, not sure. So then this would not work on any device without real RTC.

Gill-Bates, 2019-06-03:
That you! Dietpi-Banner works as expected now. Please go ahead and close this issue.

Maybe an improvement to force who -b to show the correct time?

MichaIng, 2019-06-03:
@Gill-Bates 
I think it is impossible to have a correct [who -b](https://manpages.debian.org/stretch/manpages-de/who.1.de.html) output on system that do not have a real RTC.

You can check `hwclock` command which should fail like that:
```
2019-06-03 02:50:49 root@micha:/tmp# hwclock
hwclock: Cannot access the Hardware Clock via any known method.
hwclock: Use the --verbose option to see the details of our search for an access method.
```

We could investigate deeper but I think `uptime` is a sufficient alternative.

closed 2019-06-03 by MichaIng