MichaIng/DietPi #2882: Image | Odroid C2 unable to resolve host address ‘raw.githubusercontent.com’

LexiconCode, 2019-06-01:
### Creating a bug report/issue

#### Required Information
DietPi version: v6.24.1 (MichaIng/master) | HW_MODEL:12 | HW_ARCH:3 | DISTRO:4
Image creator: DietPi Core Team
Pre-image: Meveric

#### Steps to reproduce

1. Flash Odroid C2 and start dietpi

#### Expected behaviour

- Dietpi setup should complete 

#### Actual behaviour

 unable to resolve host address ‘raw.githubusercontent.com’

#### Extra details

─
Exit code: 4
DietPi version: v6.24.1 (MichaIng/master) | HW_MODEL:12 | HW_ARCH:3 | DISTRO:4
Image creator: DietPi Core Team
Pre-image: Meveric

 file contents:
der mode enabled. Check if remote file exists.
019-06-01 16:00:33--
ps://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
olving raw.githubusercontent.com (raw.githubusercontent.com)... failed: Connecti
ed out.
t: unable to resolve host address ‘raw.githubusercontent.com’

ps://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt

Works
https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt

MichaIng, 2019-06-01:
@LexiconCode 
> failed: Connection timed out.

The next slow DNS resolving 🤔.
Please estimate the time required:
```
time wget --spider https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
```

Then apply an enough high timeout and redo the step that failed during update, e.g.:
```
G_CONFIG_INJECT 'CONFIG_G_CHECK_URL_TIMEOUT=' 'CONFIG_G_CHECK_URL_TIMEOUT=20' /DietPi/dietpi.txt
/DietPi/dietpi/func/dietpi-set_software verify_dietpi.txt
```

diveyez, 2019-06-12:
Disable IPv6 and that should do it.

closed 2019-06-12 by LexiconCode

MichaIng, 2019-06-12:
@LexiconCode @diveyez 
Interesting, IPv6 was indeed to issue for you?

I observed this on my VMs. I have no actively enabled IPv6/DHCPv6 in my router, however SLAAC is done by the local devices automatically. But on my VMs, connected via WiFi bridged network (VirtualBox) hang  nearly forever when trying to access an IPv6 address. Disabling IPv6 solves this as well. However I thought it is some unique issue of my network setup.

This is also the reason why DietPi images are shipped by default with IPv4 prefer active (for APT and wget) while IPv6 is enabled. This should be the most compatible default setup. However e.g. curl does not have a static config file (AFAIK) where one can force it to use/prefer IPv4 over IPv6. However `wget` was what times out here and prefer IPv4 should be enabled by default in /etc/wgetrc.